OctoPrint/OctoPrint #4746: [Request]

Obicom, 2023-03-05:
### Is your feature request related to a problem? Please describe.

If I connect Octoprimnt to the new Snapmaker Artisan no bed temperature is shown.

### Describe the solution you'd like

The Snapmaker Artisan delivers two bed temperatures. One for the inner zone and one for the outer zone. It would be great if Octoprint could handle this new feature with two temperture zones.

foosel, 2023-03-06:
I don't see how anyone could do anything productive here unless you share in which way the Snapmaker's temperature reports differ from the standard - so how about you share some logs? 😉 Ideally: enable serial.log, connect to the printer, create a system info bundle, share it here.

In any case it would be something that would only be added through a plugin - OctoPrint core can't support the whims of every printer vendor under the sun 😐

foosel, 2023-08-29:
Closing due to lack of response.

closed 2023-08-29 by foosel

pmzandbergen, 2023-11-30:
```
Recv:  T0:210.90 /210.00 B0:59.95 /60.00 B1:36.43 /0.00 M:44.26 /0.00 T0:210.90 /210.00 T1:37.70 /0.00 @:0 B0@:127 B1@:0 @0:0 @1:0 @B_ZONE: CENTER
Recv:  T0:210.40 /210.00 B0:59.92 /60.00 B1:36.53 /0.00 M:44.27 /0.00 T0:210.40 /210.00 T1:37.70 /0.00 @:127 B0@:127 B1@:0 @0:127 @1:0 @B_ZONE: CENTER
Recv:  T0:209.60 /210.00 B0:60.00 /60.00 B1:36.39 /0.00 M:44.43 /0.00 T0:209.60 /210.00 T1:37.70 /0.00 @:127 B0@:0 B1@:0 @0:127 @1:0 @B_ZONE: CENTER
Recv:  T0:209.10 /210.00 B0:59.90 /60.00 B1:36.43 /0.00 M:44.38 /0.00 T0:209.10 /210.00 T1:37.70 /0.00 @:127 B0@:127 B1@:0 @0:127 @1:0 @B_ZONE: CENTER
Recv:  T0:210.10 /210.00 B0:60.00 /60.00 B1:36.43 /0.00 M:44.36 /0.00 T0:210.10 /210.00 T1:37.70 /0.00 @:0 B0@:0 B1@:0 @0:0 @1:0 @B_ZONE: CENTER
```

The interesting part is: `B0:59.95 /60.00 B1:36.43 /0.00`

B0 = inner bed (in this example set to heat to 60 degree), B1 = outer bed (in this example not set to heat)

pmzandbergen, 2023-11-30:
Re-opened in https://github.com/OctoPrint/OctoPrint/issues/4913