OctoPrint/OctoPrint #4747: Temperature graph legend is "minutes since round number" ( -3 -8 -13  changes to -4 -9 -14...)

disconn3ct, 2023-03-06:
### The problem

The temperature graph legend is "since a round number". At 15:43, the legend starts with "-3" (15:40) but has no other relation to wall time.

This graph should have a static legend of 0, -5, -10... with the preheat change moving left. The legend itself should not change:
[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.7

### Operating system running OctoPrint

Octopi 1.0.0

### Printer model & used firmware incl. version

MKS3+

### Browser and version of browser, operating system running browser

Chromium, PopOs

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

foosel, 2023-11-30:
This is now implemented by the above commit:

[image]

Will be part of 1.10.0

foosel, 2024-04-25:
1.10.0 has been released.

closed 2024-04-25 by foosel

disconn3ct, 2024-04-25:
A quick look says this is fixed. Thanks!