OctoPrint/OctoPrint #4750: [RC Feedback] Feedback on the 1.9.0rc2 Release Candidate

foosel, 2023-03-07:
Please provide *general* feedback on your experience with the 1.9.0rc2 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### ❗👇 IMPORTANT 👇❗
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports *with system info bundles*, not just a statement that something doesn't work!
### ❗👆 IMPORTANT 👆❗

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

b-morgan, 2023-03-07:
RPi 3B with RPi camera V2. Upgraded from 1.8.7 to 1.9.0rc2 on image 3328. All is well so far 😄

NewsGuyTor, 2023-03-07:
My camera doesn't work on the Octopi Dashboard, it just blinks "Loading webcam…" or is a black square. 

https://octopi.local/webcam/?action=stream still works fine, as before.

r3Fuze, 2023-03-07:
After updating I got the screen that says "The OctoPrint server is not currently running". No process running, no obvious errors in `octoprint.log`, and restarting the service and the pi did not help.

I saw [this forum post](https://community.octoprint.org/t/the-octoprint-server-is-currently-not-running-error/24440/6) and running the `--force-reinstall` command reverted OctoPrint to version 1.8.7, and after that I could update to 1.9.0rc2 without problems. I suspect the issue was just what that forum post mentioned (`we also see it with updates to OctoPrint (<0.1%)`) but I thought I'd report here just in case.

I uninstalled/reinstalled a plugin after fixing the issue so there is some noise in the logs, but here are some relevant timestamps from `octoprint.log`:

- `2023-03-07 18:42:40` OctoPrint shuts down after first install of 1.9.0rc2
- `2023-03-07 18:53:58` OctoPrint starts after reverting to 1.8.7

[octoprint-systeminfo-20230307190602.zip](https://github.com/OctoPrint/OctoPrint/files/10913003/octoprint-systeminfo-20230307190602.zip)

N0YHR, 2023-03-08:
> My camera doesn't work on the Octopi Dashboard, it just blinks "Loading webcam…" or is a black square.
> 
> https://octopi.local/webcam/?action=stream still works fine, as before.

I have the identical issue when attempting to view the camera feed in "Control".  When I go to the stream setup as the original author indicates, I get the stream.

Rebooting in Safe Mode does allow me to see the video stream.  Again rebooting, the video stream disappears.

Hardware is Raspberry Pi 4B+ 8GB running PandaPi's old Buster based PandaPi OS.  (I can't get him to update it, and I'm not a programmer for compiling on my own).

foosel, 2023-03-08:
> Rebooting in Safe Mode does allow me to see the video stream.

That sounds like a third party plugin might be causing this. @NewsGuyTor how do things look in safe mode for you?

Both of you, please share a system info bundle.

Silverstar, 2023-03-08:
I have the camera issue as well. In _my_ case, it seems to be UI customizer :/
As soon as I disable it, the camera in the octoprint UI shows again.
If I disable the "webcam widget" in the UI customizer -> Layout tab, I can have the webcam stream again, but I have to reload the web interface every time I leave the settings widget, otherwise the box is black again.

[octoprint-systeminfo-20230308091914.zip](https://github.com/OctoPrint/OctoPrint/files/10919148/octoprint-systeminfo-20230308091914.zip)

LazeMSS, 2023-03-08:
I will probally removing the entire webcam stuff from my plugin :)

LazeMSS, 2023-03-08:
This can be closed here - I have a separate issue now
https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/281

crysxd, 2023-03-08:
So I guess my input is no longer needed? @LazeMSS feel free to tag me in the other issue if needed

NewsGuyTor, 2023-03-08:
> I will probally removing the entire webcam stuff from my plugin :)

Please don't do it, it is absolutely vital to have the webcam visible all the time over the "Main tabs", like so:

[image]

LazeMSS, 2023-03-08:
Yes but the webcam stuff is really hard to "modify" without breaking stuff or causing mem leaks and I only have a limited time working on these things

LazeMSS, 2023-03-08:
@jneilliii  i think you have the same problem in https://github.com/jneilliii/OctoPrint-ConsolidateTempControl/blob/839bf6178cf48e367b544eb92dd3aaa3de33cc03/octoprint_consolidate_temp_control/static/js/consolidate_temp_control.js#L119

NewsGuyTor, 2023-03-08:
> Yes but the webcam stuff is really hard to "modify" without breaking stuff or causing mem leaks and I only have a limited time working on these things

Wouldn't just iframing(/framing with JS) https://octopi.local/webcam/?action=stream work?

LazeMSS, 2023-03-08:
> > Yes but the webcam stuff is really hard to "modify" without breaking stuff or causing mem leaks and I only have a limited time working on these things
> 
> Wouldn't just iframing(/framing with JS) https://octopi.local/webcam/?action=stream work?

No

jneilliii, 2023-03-08:
> @jneilliii i think you have the same problem in https://github.com/jneilliii/OctoPrint-ConsolidateTempControl/blob/839bf6178cf48e367b544eb92dd3aaa3de33cc03/octoprint_consolidate_temp_control/static/js/consolidate_temp_control.js#L119

Yep, I've created issues in all my plugins that may be effected. Was going to start looking at them this weekend.

https://github.com/jneilliii/OctoPrint-ConsolidateTempControl/issues/28

NewsGuyTor, 2023-03-08:
> > > Yes but the webcam stuff is really hard to "modify" without breaking stuff or causing mem leaks and I only have a limited time working on these things
> > 
> > Wouldn't just iframing(/framing with JS) https://octopi.local/webcam/?action=stream work?
> 
> No

Are you sure? Is it for compatibility reasons?

I tried replacing the HTML with the image HTML, and it worked great:

1.  [image]

2. [image]

3. [image]

cp2004, 2023-03-08:
This is not the issue for discussing the plugin features. Webcam handling is more complicated than 'just' this or that, and it is LazeMSS's plugin so he can do what he likes with it.

LazeMSS, 2023-03-09:
Im closing this - if @cp2004 or @jneilliii  have the time have a look here:
https://github.com/LazeMSS/OctoPrint-UICustomizer/pull/282

LazeMSS, 2023-03-09:
Sorry this is generic issue - so not closing but please no more discussion about problems with webcam in UI Customizer - look here instead: https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/281

foosel, 2023-03-09:
And I'm *very* confused right now that you even *could* close it 🤨

dan-and, 2023-03-09:
Overall 1.9.0rc2 looks great, but I try to understand why sometimes I don't see the temperature graphs. 

The developer console in Chrome give me a hint: 
`Error calling fromCurrentData on view model TemperatureViewModel : TypeError: Cannot read properties of undefined (reading 'css')
    at https://e3octopi.danand.de/static/webassets/packed_core.js?1f6b9a20:423:20
    at Array.map (<anonymous>)
    at TemperatureViewModel.self._drawMarkings (https://e3octopi.danand.de/static/webassets/packed_core.js?1f6b9a20:417:123)
    at drawGrid (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:1013:299)
    at Plot.draw (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:1003:49)
    at TemperatureViewModel.self.updatePlot (https://e3octopi.danand.de/static/webassets/packed_core.js?1f6b9a20:415:505)
    at TemperatureViewModel.self._processTemperatureUpdateData (https://e3octopi.danand.de/static/webassets/packed_core.js?1f6b9a20:407:124)
    at TemperatureViewModel.self.fromCurrentData (https://e3octopi.danand.de/static/webassets/packed_core.js?1f6b9a20:402:1211)
    at callViewModelIf (https://e3octopi.danand.de/static/webassets/packed_core.js?1f6b9a20:1795:75)
    at https://e3octopi.danand.de/static/webassets/packed_core.js?1f6b9a20:1789:138
    at Pn (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:30:530)
    at Function.<anonymous> (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:51:66)
    at callViewModelsIf (https://e3octopi.danand.de/static/webassets/packed_core.js?1f6b9a20:1789:95)
    at callViewModels (https://e3octopi.danand.de/static/webassets/packed_core.js?1f6b9a20:1788:56)
    at https://e3octopi.danand.de/static/webassets/packed_core.js?1f6b9a20:1669:252
    at DataUpdater.self._ifInitialized (https://e3octopi.danand.de/static/webassets/packed_core.js?1f6b9a20:1683:1002)
    at DataUpdater.self._onCurrentData (https://e3octopi.danand.de/static/webassets/packed_core.js?1f6b9a20:1669:226)
    at https://e3octopi.danand.de/static/webassets/packed_client.js?3339ed72:24:109
    at Pn (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:30:530)
    at Function.<anonymous> (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:51:66)
    at OctoPrintSocketClient.propagateMessage (https://e3octopi.danand.de/static/webassets/packed_client.js?3339ed72:24:77)
    at https://e3octopi.danand.de/static/webassets/packed_client.js?3339ed72:31:316
    at https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:47:23
    at _t (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:35:392)
    at https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:46:369
    at Function.<anonymous> (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:51:73)
    at w.onMessage [as onmessage] (https://e3octopi.danand.de/static/webassets/packed_client.js?3339ed72:31:278)
    at r.dispatchEvent (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:1585:2864)
    at https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:1585:13481
    at Array.forEach (<anonymous>)
    at w._transportMessage (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:1585:13432)
    at i.emit (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:1585:1745)
    at s.<anonymous> (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:1585:29210)
    at i.emit (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:1585:1745)
    at s.<anonymous> (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:1585:28544)
    at i.emit (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:1585:1745)
    at s._chunkHandler (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:1585:35350)
    at i.emit (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:1585:1745)
    at xhr.onreadystatechange (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:1585:20804)
    at XMLHttpRequest.sentryWrapped (https://e3octopi.danand.de/static/webassets/packed_core.js?1f6b9a20:703:52003)
callViewModelIf @ packed_core.js?1f6b9a20:1796
(anonymous) @ packed_core.js?1f6b9a20:1789
Pn @ packed_libs.js?6bee1dee:30
(anonymous) @ packed_libs.js?6bee1dee:51
callViewModelsIf @ packed_core.js?1f6b9a20:1789
callViewModels @ packed_core.js?1f6b9a20:1788
(anonymous) @ packed_core.js?1f6b9a20:1669
DataUpdater.self._ifInitialized @ packed_core.js?1f6b9a20:1683
DataUpdater.self._onCurrentData @ packed_core.js?1f6b9a20:1669
(anonymous) @ packed_client.js?3339ed72:24
Pn @ packed_libs.js?6bee1dee:30
(anonymous) @ packed_libs.js?6bee1dee:51
OctoPrintSocketClient.propagateMessage @ packed_client.js?3339ed72:24
(anonymous) @ packed_client.js?3339ed72:31
(anonymous) @ packed_libs.js?6bee1dee:47
_t @ packed_libs.js?6bee1dee:35
(anonymous) @ packed_libs.js?6bee1dee:46
(anonymous) @ packed_libs.js?6bee1dee:51
onMessage @ packed_client.js?3339ed72:31
r.dispatchEvent @ packed_libs.js?6bee1dee:1585
(anonymous) @ packed_libs.js?6bee1dee:1585
w._transportMessage @ packed_libs.js?6bee1dee:1585
i.emit @ packed_libs.js?6bee1dee:1585
(anonymous) @ packed_libs.js?6bee1dee:1585
i.emit @ packed_libs.js?6bee1dee:1585
(anonymous) @ packed_libs.js?6bee1dee:1585
[... 105 lines cut ...]
XMLHttpRequest.send (async)
(anonymous) @ packed_core.js?1f6b9a20:703
(anonymous) @ packed_core.js?1f6b9a20:703
[... cut ...]
i @ packed_libs.js?6bee1dee:1585
s @ packed_libs.js?6bee1dee:1585
s._scheduleReceiver @ packed_libs.js?6bee1dee:1585
(anonymous) @ packed_libs.js?6bee1dee:1585
e @ packed_libs.js?6bee1dee:1585
i.emit @ packed_libs.js?6bee1dee:1585
(anonymous) @ packed_libs.js?6bee1dee:1585
e @ packed_libs.js?6bee1dee:1585
i.emit @ packed_libs.js?6bee1dee:1585
xhr.onreadystatechange @ packed_libs.js?6bee1dee:1585
sentryWrapped @ packed_core.js?1f6b9a20:703
`

[image]

I will make a complete bug report, if I figure it out more in-depth.

cp2004, 2023-03-10:
> TypeError: Cannot read properties of undefined (reading 'css')

That sounds a bit like it might be related to the changes in the way event markings are processed to allow plugins to extend it, added in PR #4658 potentially? Not sure. @dan-and please do make a full bug report regardless of figuring it out so it can be tracked as an issue in the RC easily.

dan-and, 2023-03-10:
> That sounds a bit like it might be related to the changes in the way event markings are processed to allow plugins to extend it, added in PR #4658 potentially? Not sure. @dan-and please do make a [...]

Sounds like it, I'll do a complete report on the weekend.

The-EG, 2023-03-10:
This breaks the firmware updater plugin. `/plugin/firmwareupdater/flash` results in a 400 response now.

Filed on the plugin repo: https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/341

Additionally, I'm getting the following error in the browser console now and again when clicking the 'Save' button in the settings window. After this error appears, clicking the 'Close' button does not close the settings window:
> Uncaught TypeError: Cannot read properties of undefined (reading 'length')

and the following error when changing to the 'Control' tab:
> Error calling onTabChange on view model ControlViewModel : TypeError: orgTabChange is not a function

I'll see if I can whittle down where those errors are coming from.

jneilliii, 2023-03-10:
The orgTabChange error is from UI Customizer. We're still trying to figure out the best way to deal with the webcam changes.

Thisismydigitalself, 2023-03-11:
> Overall 1.9.0rc2 looks great, but I try to understand why while printing I don't see the temperature graphs. Before and after a print job, the graphs are visible again and do have history data. Well even mouse-over while printing shows the actual values in the bottom left of the graph. I tested it on Chrome and Firefox.
> 

Same here, on firefox latest ver, running on Wintel64bit.
first version ever to have this issue.

Thisismydigitalself, 2023-03-11:
just installed 1.9.0.rc2. temp graph is not working as usual. no realtime graphs, 
first version ever to introduce such issue.
[image]

foosel, 2023-03-13:
@dan-and & @Thisismydigitalself, the temperature graph issue is now tracked in #4761.

But: *please* next time file a proper bug report. Y'all don't need to analyse the bugs yourself, you don't need to find the solution yourself, but for the love of OctoPrint's cute mascot please create proper bug reports instead of making me dig through comments, Sentry, and worst case having to beg for it 😅

dan-and, 2023-03-13:
> @dan-and & @Thisismydigitalself, the temperature graph issue is now tracked in #4761.
>
> But: *please* next time file a proper bug report. Y'all don't need to analyse the bugs yourself, you don't need [...]

Sorry, I planned on doing a proper bug report but just got side tracked. I planned on preparing it today, but will have a look at #4761.

Thanks and sorry for long delay

benlye, 2023-03-18:
> This breaks the firmware updater plugin. `/plugin/firmwareupdater/flash` results in a 400 response now.
> 
> Filed on the plugin repo: [OctoPrint/OctoPrint-FirmwareUpdater#341](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/341)
> 

I believe I have a fix for this.  The new version of werkzeug was causing the problem, but fortunately a contributor had already provided the solution and I just needed to merge it (https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/pull/324)

I'll get a new release of the plugin out as soon as I can.

The-EG, 2023-03-18:
> I believe I have a fix for this.

I can confirm that the latest dev version of FirmwareUpdater resolves the issue I mentioned above.

closed 2023-03-22 by foosel