OctoPrint/OctoPrint #4751: [Request] Ability to use firmware pause command (M125)

Simonlebucheron, 2023-03-07:
### Is your feature request related to a problem? Please describe.

We have two needs with our printer, I am sure we are not alone:
- _Need 1_: we often pause the print and park the nozzle during print **with command** added in the gcode to insert nut or structural reinforcement into our designs. 
- _Need 2_: Sometimes we pause print **with octoprint** to clean something, so the nozzle should be parked during this time.

But we are not allowed to use the specific firmware function `M125` to manage the pause with octoprint.

### Describe the solution you'd like

We would like to be able to change the pause method used by octoprint with a [`M125`](https://marlinfw.org/docs/gcode/M125.html) command to use the _PARK_HEAD_ON_PAUSE_ marlin function to pause the print and park in the defined spot.

### Describe alternatives you've considered

- First alternative:
We currently use "Pause at height" post-script in our slicer (Cura) to easily generate gcode but it only allows us to insert `M0` or `@pause` into gcode for our machine type.
We can manually use a custom script or add a `M125` command into the gcode but it is not convenient and it would counterproductive to use these commands since octoprint `@pause` action already manage all related events/trigger very well.
By the way, this alternative does not fit our _need 2_
- Second alternative:
We could use the default print `@pause` action with our slicer or with octoprint and we could configure the [afterPrintPaused script](https://docs.octoprint.org/en/master/features/gcode_scripts.html?highlight=pause#id4)
But we would prefer a dedicated firmware machine function to handle the pause feature to avoid other possible issues (like time-out on steppers for example). 
And using this method make the pause process procedure unnecessary redundant with firmware function.

### Additional context

Alongside this, we will certainly ask the developer team of Cura Post Processing Scripts to let us choose the `M125` command the post-processing script.

CmdrCody51, 2023-03-07:
Simonlebucheron
When I hit the Pause, my script does this:

M400 ; complete moves
M114 ; report position
OCTO203 ; turn on light
G60 ; Save position requires firmware update
; relative XYZE
G91
M83
; move Z upwards and retract
G1 Z+15 E-5 F500
; absolute XYZE
M82
G90
; move to a safe rest position, adjust as necessary
G1 X5 Y210 F4500
M300 P500 S777 ; beep
M84 S0 ; disable stepper time out don't forget to set to S30 in after pause

Of course, your M125 saves a lot of steps...

M400
M114
M84 S0
M125

Wouldn't that work?

Simonlebucheron, 2023-03-08:
> Simonlebucheron When I hit the Pause, my script does this:
>
> M400 ; complete moves M114 ; report position OCTO203 ; turn on light G60 ; Save position requires firmware update ; relative XYZE G91 M83 [...]

That is what I describe in my second alternative of my original post, it works but It is unnecessary redundant with firmware function. And we could let the machine manage his pause to avoid other possible issue. 
I think it is a safe way to handle the machine pause from octoprint point of view.

cp2004, 2023-03-08:
Are you talking about when printing from the SD card here, or when printing from OctoPrint?

If it is printing from OctoPrint, then there is no need to tell the machine to pause the print because it is not in control of said print. If you would like to send M125 when you pause in OctoPrint, then this is what the gcode scripts are for, you can make it do what you would like to do rather than having everyone's use cases written in OctoPrint's code, this is why it is made configurable.

If you are printing from the printer's SD card, the case is different, and the M25 command is sent, to pause the SD print. Are you asking for this command to be made configurable so you can change it to M125 not M25? From https://marlinfw.org/docs/gcode/M025.html it looks like this will park the nozzle as well, with the parking configuration in the firmware.

foosel, 2023-03-08:
> this is why it is made configurable.

**Exactly**!

@Simonlebucheron 

> * _Need 1_: we often pause the print and park the nozzle during print **with command** added in the gcode to insert nut or structural reinforcement into our designs.

Two options: 

1. Instead of inserting `M125`, [insert `@pause`](https://docs.octoprint.org/en/master/features/atcommands.html). 
2. Configure `M125` as a pause trigger (Settings > Serial Connection > Firmware & Protocol > Protocol Fine Tuning > Pausing commands)

> * _Need 2_: Sometimes we pause print **with octoprint** to clean something, so the nozzle should be parked during this time.

Throw M125 into your pausing code in Settings > GCODE Scripts > After print job is paused.

> It is unnecessary redundant with firmware function.

`M125` is not documented on https://reprap.org/wiki/G-code, so with that being the case for now I must assume this is a Marlin only feature, and quite recent to boot, so OctoPrint cannot just assume it will a) exist and b) do what it does on recent Marlin and thus cannot blindly send it to a printer, especially since at least in Marlin's docs I can't even find whether it should generate a capability report or not.

CmdrCody51, 2023-03-08:
Think I figured it out...
If the printer gets the M125, the flow goes to the printer and when you are done doing what ever, you push the button.
OctoPrint is unaware that the printer is ready.
I'm thinking host action commands. ACTION_ON_RESUME maybe?

jneilliii, 2023-03-09:
As the option is already configurable here for what command to use for pause and both needs are met, closing issue.

closed 2023-03-09 by jneilliii