OctoPrint/OctoPrint #4753: Setting log levels to WARN causes python error - AttributeError: 'PlaceHolder' object has no attribute 'setLevel'

sammcj, 2023-03-08:
### The problem

I've been trying to turn down the log level from the default which seems to be DEBUG for some things and INFO for others to reduce the size of OctoPrints log files (which can easily be 10GB+ a day!).

I was wondering why it wasn't making any difference despite "successfully" applying the log levels in the UI - turns out when looking through the log files buried in them is the following Python error:

```
2023-03-08 12:29:52,239 - octoprint - ERROR - Exception on /plugin/logging/setup/levels [PUT]
Traceback (most recent call last):
  File "/home/octo/OctoPrint/lib/python3.11/site-packages/flask/app.py", line 2073, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/octo/OctoPrint/lib/python3.11/site-packages/flask/app.py", line 1519, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/octo/OctoPrint/lib/python3.11/site-packages/flask/app.py", line 1517, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/octo/OctoPrint/lib/python3.11/site-packages/flask/app.py", line 1503, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/octo/OctoPrint/lib/python3.11/site-packages/octoprint/server/util/flask.py", line 1598, in decorated_view
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/octo/OctoPrint/lib/python3.11/site-packages/octoprint/access/permissions.py", line 175, in _decorated
    rv = f(*args, **kw)
         ^^^^^^^^^^^^^^
  File "/home/octo/OctoPrint/lib/python3.11/site-packages/octoprint/plugins/logging/__init__.py", line 124, in set_logging_levels_api
    self._set_logging_levels(json_data)
  File "/home/octo/OctoPrint/lib/python3.11/site-packages/octoprint/plugins/logging/__init__.py", line 211, in _set_logging_levels
    self._logger.manager.loggerDict[component].setLevel(logging.INFO)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'PlaceHolder' object has no attribute 'setLevel'
```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.7

### Operating system running OctoPrint

Armbian (latest stable)

### Printer model & used firmware incl. version

N/A

### Browser and version of browser, operating system running browser

Firefox 111, macOS

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

- Couldn't include SystemInfo bundle as was too large to download and caused OctoPrint to lag up (I gave up at 120MB) and `OctoPrint/bin/octoprint systeminfo .` doesn't seem to generate any files.

[image]

The octoprint.log file is pretty much unreadable but it looks like it contains raw GCODE outputs etc.., for example here's a tail of my current [octoprint.log](https://github.com/OctoPrint/OctoPrint/files/10915574/octoprint.log)

sammcj, 2023-03-08:
@GitIssueBot not possible - it looks like that file is too big for you.

foosel, 2023-03-08:
If your logs grow this much in size, something must be off because that is *not* normal. 

> The octoprint.log file is pretty much unreadable but it looks like it contains raw GCODE outputs etc..,

Yeah, that is *definitely* not normal. Can you please share an excerpt with such readings? I might be able to tell you what's up there.

sammcj, 2023-03-09:
That might be really helpful thanks @foosel!

I've taken an 8GB log file and chopped it at 2MB then compressed it:

[octoprint.log.2023-03-03-reduced-size.zip](https://github.com/OctoPrint/OctoPrint/files/10936511/octoprint.log.2023-03-03-reduced-size.zip)

The main error I see in it is `ModuleNotFoundError: No module named '_sqlite3'` - but that can't be using that much disk space.

cp2004, 2023-03-10:
The main issue seems to be the Slicer Thumbnails plugin logging your file metadata, which might not be intentional and is putting out a lot of data into the log file.

Perhaps the below line in the plugin is the culprit:

https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/blob/15631376dfdfead0579ccfa4dabbae3c1ae9db68/octoprint_prusaslicerthumbnails/__init__.py#LL261C9-L261C9

You also look like you have quite a *lot* of plugins, running OctoPrint in safe mode to see if you can get a useable log to debug if you actually have a different issue (than the huge logs) would help.

cp2004, 2023-03-10:
It looks like you have a `sync_on_refresh` option that might trigger a rescan every time the file list is fetched?

jneilliii, 2023-03-10:
Yeah, that's probably the case. That option was added later to call the crawl and didn't think about the impact.

sammcj, 2023-03-11:
Interesting, good spotting on that sync_on_refresh.

For now I've disabled the slicer thumbnails - I assume that's a bug then @jneilliii ?

jneilliii, 2023-03-11:
Technically not a bug, but an oversight. I'll have an update out for the plugin soon to change that to a debug log line instead of info.

sammcj, 2023-03-11:
Cool, I'll close this for now as logging does seem to have calmed down a lot, still not perfect but as you said I've quite a few plug-ins running.

cheer folks, I appreciate your time.

jneilliii, 2023-03-11:
new version 1.0.2 out now you may want to look into the plugin that was erroring, SpoolManager, SlicerEstimator, arc_welder, PrintJobHistory and octoeverywhere also looks to be info logging a lot as well @QuinnDamerell.

foosel, 2023-03-11:
Let's keep this open - the Python error you saw still needs to be investigated as it certailny shouldn't be happening ;)

sammcj, 2023-03-11:
Ok cool - I'm in the process of migrating from marlin to klipper today - but that gives me a chance to run up octo in safe mode etc... if not today, tomorrow I think.

QuinnDamerell, 2023-03-11:
Thanks for the tag @jneilliii. If I'm reading this correctly, the issue here is that when the log system is used a lot, the result might be a bug?

OE does log a decent amount, but I do that because in the past its been helpful to give context to issues that users can't repo after the change in the log level. But I can definitely reduce the logging some.

cp2004, 2023-03-11:
No, the point is that over-logging fills up the disk space and can make it hard to work with the log.

QuinnDamerell, 2023-03-11:
Oh, I c. That's also a good reason to reduce it. Are there no rolling log limits or anything in OctoPrint?

cp2004, 2023-03-11:
The logs rotate every 24 hours as far as I can remember, not based on size, although I haven't checked that just going from memory.

sammcj, 2023-03-12:
Fills up the disk - but also causes much higher I/O than needed - which is especially bad on low powered devices like raspberry pis etc...

foosel, 2023-07-03:
The reason for the originally reported AttributeError has now been identified and fixed as of the above commit. Will be part of 1.10.0

commit fba5f907 (2023-07-03): 🐛 logging: Fix runtime setting of level

`loggerDict[logger]` might be a placeholder. `getLogger` takes care of
replacing said placeholder with a full logger upon first access, before
returning it.

Closes #4753
closed 2024-04-24 by foosel