OctoPrint/OctoPrint #4754: Number of Extruders on Printer Profile is Not Saved

FlintRCTech, 2023-03-08:
### The problem

Version 1.9.0rc2
1. Go to Settings -> Printer -> Printer Profiles -> Edit Profile -> Hotbed and Extruder
2. Change the number of extruders
3. Click Confirm
4. Click Edit Profile again
5. Go to Hotbed and Extruder 
6. Notice number of extruders is back to 1
7. Change the number of extruders again
8. Click Confirm
9. Click Save
10. Go to Settings -> Printer -> Printer Profiles -> Edit Profile -> Hotbed and Extruder
11. Notice number of extruders is back to 1

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.0rc2

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Geeetech A10T - Marlin 2.0.8.2

### Browser and version of browser, operating system running browser

Edge, Windows 11

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230307210557.zip](https://github.com/OctoPrint/OctoPrint/files/10916823/octoprint-systeminfo-20230307210557.zip)
[BrowserConsole.log](https://github.com/OctoPrint/OctoPrint/files/10916830/BrowserConsole.log)

foosel, 2023-03-08:
Confirmed, looks like a simple copy-paste error on the newly introduced sanity checks.

foosel, 2023-03-08:
Fix ready for 1.9.0rc3

foosel, 2023-03-22:
1.9.0rc3 is out.

commit f5e90a1a (2023-03-08): 🐛 Fix saving of extruder count in printer profiles

Simple copy-paste error.

Closes #4754
closed 2023-03-22 by foosel