OctoPrint/OctoPrint #4756: Including Variables in GCODE-Scripts: Not possible for more than one plugin at a time

kaenguruhs, 2023-03-09:
### The problem

In the documentation it is stated that plugins are able to provide variables used in GCODE-Scripts. That was included with PR 2338: https://github.com/OctoPrint/OctoPrint/pull/2338/commits/6545f9a81668cf5e6a19faf5ee4e41a16aa64944#diff-46e37c69234bdecf15747b361173b913596cfd2ab85bb6ac09de9ec6809e66bd

I noticed that it is only possible to have one plugin at a time providing variables, due to the way the variables are handled in https://github.com/OctoPrint/OctoPrint/blob/devel/src/octoprint/util/comm.py#L1380:
1. First of all, no variables from plugins are set. The `context` is in the form `{'foo': 'bar*}`.
2. With the first plugin providing variables, the `context` changes to `{'foo': 'bar', 'plugins': {'PluginA': {'VariableA': 'ValueA'}}}`
3. If another plugin provides variables to be included in the script, the `context` gets updated. But, instead of only adding the Variables from `PluginB`, the value of the `context[plugins]` is overwritten: `{'foo': 'bar', 'plugins': {'PluginB': {'VariableB': 'ValueB'}}}`.

I did a small Python-fiddle to make it clear:
```
# 1. context before adding any plugin-related variable
context = {'foo': 'bar'}

# 2. name and variables from PluginA
name = "PluginA"
variables = {'VariableA': 'ValueA'}

# 3. update context
context.update({"plugins": {name: variables}})
print("After adding pluginA: " + str(context))

# 4. Add name and variables from PluginB
name = "PluginB"
variables = {'VariableB': 'ValueB'}

# 5. update context
context.update({"plugins": {name: variables}})
print("After adding pluginB: " + str(context))
```

For adding the variables correctly I suppose to change https://github.com/OctoPrint/OctoPrint/blob/devel/src/octoprint/util/comm.py#L1380:
from: 
`context.update({"plugins": {name: variables}})`
to: 
```
if 'plugins' in context:
    context["plugins"].update({name: variables})
else:
    context.update({"plugins": {name: variables}})
```

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Because it is a problem with the way the variables from plugins are handled.

### Version of OctoPrint

1.8.7

### Operating system running OctoPrint

RaspiOS with docker

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230309143846.zip](https://github.com/OctoPrint/OctoPrint/files/10932140/octoprint-systeminfo-20230309143846.zip)

cp2004, 2023-03-10:
Would you like to submit a PR to fix this issue? Seeing as you did most of the work to figure out the problem already, go and take the credit 🙂

foosel, 2023-06-20:
PR merged (thanks!), will be in 1.10.0

commit 388e45ae (2023-06-20): 🐛 Fix context variables in GCODE scripts from multiple plugins (#4757)

* provide variables from multiple plugins

* 🎨 Fix linter issues

---------

Closes #4756
closed 2024-04-24 by foosel