OctoPrint/OctoPrint #4760: Compatibility with Python 3.12

cp2004, 2023-03-11:
### Is your feature request related to a problem? Please describe.

## Compatibility Issues
* #4685 
* Issue with psutil - see trace below

```
2023-03-11 15:15:45,756 - octoprint - ERROR - Exception on /api/files [GET]
Traceback (most recent call last):
  File "C:\Users\charl\Development\venv312\Lib\site-packages\flask\app.py", line 2528, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\charl\Development\venv312\Lib\site-packages\flask\app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\charl\Development\venv312\Lib\site-packages\flask\app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\charl\Development\venv312\Lib\site-packages\flask\app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\charl\Development\OctoPrint Core\OctoPrint\src\octoprint\vendor\flask_principal.py", line 196, in _decorated
    rv = f(*args, **kw)
         ^^^^^^^^^^^^^^
  File "C:\Users\charl\Development\OctoPrint Core\OctoPrint\src\octoprint\server\util\flask.py", line 1345, in decorated_function
    response = f(*args, **kwargs)
               ^^^^^^^^^^^^^^^^^^
  File "C:\Users\charl\Development\OctoPrint Core\OctoPrint\src\octoprint\server\api\files.py", line 161, in readGcodeFiles
    usage = psutil.disk_usage(settings().getBaseFolder("uploads", check_writable=False))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\charl\Development\venv312\Lib\site-packages\psutil\__init__.py", line 2000, in disk_usage
    return _psplatform.disk_usage(path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\charl\Development\venv312\Lib\site-packages\psutil\_pswindows.py", line 274, in disk_usage
    total, free = cext.disk_usage(path)
                  ^^^^^^^^^^^^^^^^^^^^^
SystemError: argument 1 (impossible<bad format char>)
2023-03-11 15:15:45,761 - tornado.access - ERROR - 500 GET /api/files?recursive=true (::1) 101.69ms
```
As a result any endpoints that need disk usage fail.

### Describe the solution you'd like

Fix the issues - Python 3.12 is still in alpha so it could still change (still alpha until May), but the earlier we mark it supported the more chance that there will be a release before October when it is finally released that OctoPrint can support it straight away. https://peps.python.org/pep-0693/ for release schedule.

### Describe alternatives you've considered

none

foosel, 2023-10-04:
Looked into this today and solved the versioneer issue, but found a new issue with the regex dependency, mrabarnett/mrab-regex#509, that affects our vendored in `awesome-slugify` dependency.

foosel, 2023-10-04:
False alarm on that one, I overlooked that I was running a 3.12 *alpha* version, with released 3.12.0 all is fine on that front.

foosel, 2023-10-04:
Implemented by

- a922ff8973a08f9c5f4617dbdaf91cfb04b387d9
- 0718dfb68eaae5d9547690458c6f33b3243d9d44
- 1e6a915e3f4f856c5d3029624506df0ff663a882
- aba8d21e75e27fc966e57a744335affeca37a547

(yes, I forgot to tag this issue in the commits once again, hooray)

So, 1.10 will be 3.12 compatible.

foosel, 2024-04-25:
1.10.0 has been released.

closed 2024-04-25 by foosel