OctoPrint/OctoPrint #4761: [1.9.0rc2] `label` can be unset in chart markings, causing the temperature graph not to render

foosel, 2023-03-13:
### Problem

In 1.9.0rc2, the code was changed to fetch the color of the mark from the background color of the label (which is determine by css through a custom class). The problem with that is that a label element is only created when it would still fit on the graph, leading to an exception in the code when attempting to read the background color from an unrendered label, which in turn leaves the temperature graph blank.

Case in point:

https://github.com/OctoPrint/OctoPrint/issues/4750#issuecomment-1461633959
```
Error calling fromCurrentData on view model TemperatureViewModel : TypeError: Cannot read properties of undefined (reading 'css')
    at https://e3octopi.danand.de/static/webassets/packed_core.js?1f6b9a20:423:20
    at Array.map (<anonymous>)
    at TemperatureViewModel.self._drawMarkings (https://e3octopi.danand.de/static/webassets/packed_core.js?1f6b9a20:417:123)
    at drawGrid (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:1013:299)
    at Plot.draw (https://e3octopi.danand.de/static/webassets/packed_libs.js?6bee1dee:1003:49)
[...]
```

also: https://github.com/OctoPrint/OctoPrint/issues/4750#issuecomment-1464913531

Sentry Issue: [COREUI-528](https://octoprint.sentry.io/issues/3983454381/?referrer=github_integration)

```
TypeError: Cannot read properties of undefined (reading 'css')
  at None (/static/webassets/packed_core.js:423:20)
  at Array.map (<anonymous>)
  at TemperatureViewModel.self._drawMarkings (/static/webassets/packed_core.js:417:123)
  at drawGrid (/static/webassets/packed_libs.js:1013:299)
  at Plot.draw (/static/webassets/packed_libs.js:1003:49)
...
(28 additional frame(s) were not displayed)
```

### Solution

Either create and color the `label` div in any case, or simply don't draw grid lines that can't fit their label on the graph. Or maybe I'll come up with another solution yet.

foosel, 2023-03-13:
Fix ready for 1.9.0rc3

rfinnie, 2023-03-13:
Oops, sorry, I actually had a WIP where I had moved label creation out of `if (o.left > yAxisLabelWidth) {` when adding the new label functionality, but looks like it got lost by the time it hit the final PR.

foosel, 2023-03-22:
1.9.0rc3 is out.

commit ba54bd30 (2023-03-13): 🐛 Make sure label is always defined to read background color from

Fixes #4761