OctoPrint/OctoPrint #4763: 1.9.0-rc2 500 Internal Server Error

JoGrob, 2023-03-14:
### The problem

What is the problem?
Octoprint does not properly start up. octopi.local gives "500 Internal Server Error"
with redirected url
http://octopi.local/login/?redirect=%2F%3F&permissions=STATUS,SETTINGS_READ

Maybe this is causes by deleting the cookies in webbrowser, so autologin can't remeber webbrowser???

What did you already try to solve it?
i've tried booting octopi several times.
I've tried several webbrowsers Edge and Firefox.

Have you tried running in [safe mode](http://docs.octoprint.org/en/master/features/safemode.html)?
can't get in save mode.

Did running in safe mode solve the problem?
n/a

Systeminfo Bundle
can't download

Additional information about your setup
octoprint.log shows some ERROR Exception on /login/ [GET]

``
2023-03-14 10:34:39,749 - octoprint.startup - INFO - ******************************************************************************
2023-03-14 10:34:39,754 - octoprint.startup - INFO - Starting OctoPrint 1.9.0rc2
2023-03-14 10:34:39,762 - octoprint.startup - INFO - ******************************************************************************

2023-03-14 10:34:52,420 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2023-03-14 10:34:54,250 - octoprint.util.pip - INFO - Version of pip is 22.3
2023-03-14 10:34:54,262 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python3.9/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2023-03-14 10:34:54,263 - octoprint.util.pip - INFO - ==> pip ok -> yes
2023-03-14 10:34:54,521 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2023-03-14 10:34:54,553 - octoprint.plugins.softwareupdate - INFO - Loaded update log from disk
2023-03-14 10:34:54,699 - octoprint.plugins.firmwareupdater - INFO - Python binproto2 package installed: False
2023-03-14 10:34:54,750 - octoprint.plugin.core - INFO - Initialized 22 plugin implementation(s)
2023-03-14 10:34:54,822 - octoprint - INFO - Installing webcam compat overlay for configured default webcam <octoprint.webcams.ProvidedWebcam object at 0x750a32e0>
2023-03-14 10:34:54,825 - octoprint.plugin.core - INFO - 31 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/tracking
| !Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/appkeys
| !Arc-Welder (1.0.0+u.bb71e8f) = /home/pi/oprint/lib/python3.9/site-packages/octoprint_arc_welder
|  Autoselect Plugin (0.3.0) = /home/pi/oprint/lib/python3.9/site-packages/octoprint_autoselect
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/backup
|  Classic Webcam (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/classicwebcam
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/corewizard
|  Cost Estimation (3.5.0) = /home/pi/oprint/lib/python3.9/site-packages/octoprint_costestimation
| !Discovery (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/discovery
|  DisplayProgress (0.1.3) = /home/pi/oprint/lib/python3.9/site-packages/octoprint_displayprogress
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/errortracking
| !Event Manager (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/eventmanager
|  File Check (2021.2.23) (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint_file_check
| !FileManager (0.1.6) = /home/pi/oprint/lib/python3.9/site-packages/octoprint_filemanager
| !Firmware Check (2021.10.11) (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint_firmware_check
|  Firmware Updater (1.13.3) = /home/pi/oprint/lib/python3.9/site-packages/octoprint_firmwareupdater
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/gcodeviewer
|  Logging (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/logging
|  Octoprint Display ETA (2.2.0) = /home/pi/oprint/lib/python3.9/site-packages/octoprint_display_eta
|  Pi Support Plugin (2022.6.13) (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint_pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/pluginmanager
| !PortLister (0.1.10) = /home/pi/oprint/lib/python3.9/site-packages/octoprint_portlister
|  PrintTimeGenius Plugin (2.3.0) = /home/pi/oprint/lib/python3.9/site-packages/octoprint_PrintTimeGenius
|  RequestSpinner (0.2.0) = /home/pi/oprint/lib/python3.9/site-packages/octoprint_requestspinner
|  Resource Monitor (0.3.12) = /home/pi/oprint/lib/python3.9/site-packages/octoprint_resource_monitor
|  Software Update (bundled) = /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/softwareupdate
| !Tab Order (0.5.12) = /home/pi/oprint/lib/python3.9/site-packages/octoprint_taborder
[... 266 lines cut ...]
werkzeug.routing.exceptions.BuildError: Could not build url for endpoint 'plugin.appkeys.handle_auth_dialog' with values ['app_token']. Did you mean 'plugin.backup.delete_backup' instead?
``

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

can't enter gui

### Version of OctoPrint

1.9.0-rc2

### Operating system running OctoPrint

octopi 1.0.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

foosel, 2023-03-20:
Fix ready for 1.9.0rc3

foosel, 2023-03-22:
1.9.0rc3 is out.

commit bb519ae8 (2023-03-23): 🐛 Fix login dialog when appkeys plugin is disabled

Closes #4763
closed 2023-03-22 by foosel