OctoPrint/OctoPrint #4764: Rest API fails when issuing multiple requests in too quick succession (Connection reset by peer)

markusressel, 2023-03-15:
### The problem

As seen through the Home Assistant integration for OctoPrint, the sensors become unavailable sometimes, especially often when OctoPrint is actively printing.

While researching a bit in this issue: https://github.com/home-assistant/core/issues/83624
I have found that the cause of this is very likely to be that the Home Assistant OctoPrint integration tries to request multiple endpoints at once (or at least in quick succession), which seems to cause `[Errno 104] Connection reset by peer` exceptions within the python OctoPrint API client libraries (both `pyoctoprintapi` and `OctoRest`), causing the sensor to become unavailable.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.8.7 Python 3.10.9

### Operating system running OctoPrint

Linux 3DPrinter 5.15.92-1-MANJARO-ARM-RPI #1 SMP PREEMPT Mon Feb 6 22:02:34 UTC 2023 aarch64 GNU/Linux

### Printer model & used firmware incl. version

Virtual Printer

### Browser and version of browser, operating system running browser

None, only API client libs

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230315061101.zip](https://github.com/OctoPrint/OctoPrint/files/10976229/octoprint-systeminfo-20230315061101.zip)

JavaScript browser console not applicable

`.octoprint/logs/octoprint.log` does not contain any errors

[image]

GCode not relevant (happens with any)

**Please have a look at the other issue, where I have posted a lot more info and logs about my research so far.**
https://github.com/home-assistant/core/issues/83624

pyoctoprintapi logs while having OctoPrint running in Safe Mode (and virtual printer):

```
/home/markus/programming/PycharmProjects/pyoctoprintapi/venv/bin/python /home/markus/programming/PycharmProjects/pyoctoprintapi/pyoctoprintapi/__main__.py 
2023-03-15 06:17:42.871966: Operational
2023-03-15 06:17:43.930038: Operational
2023-03-15 06:17:45.004080: Operational
2023-03-15 06:17:46.073795: Operational
2023-03-15 06:17:47.137507: Operational
2023-03-15 06:17:48.204104: Operational
2023-03-15 06:17:49.240188: Operational
2023-03-15 06:17:50.304449: Operational
2023-03-15 06:17:51.377577: Operational
2023-03-15 06:17:52.442461: Operational
2023-03-15 06:17:53.512279: Operational
2023-03-15 06:17:54.584815: Operational
2023-03-15 06:17:55.644649: Operational
2023-03-15 06:17:56.709416: Operational
2023-03-15 06:17:57.777073: Operational
2023-03-15 06:17:58.814365: Printing
2023-03-15 06:17:59.879374: Printing
2023-03-15 06:18:00.941072: Printing
2023-03-15 06:18:02.005913: Printing
2023-03-15 06:18:03.060200: Printing
2023-03-15 06:18:04.109465: Printing
2023-03-15 06:18:05.176736: Printing
2023-03-15 06:18:06.238048: Printing
2023-03-15 06:18:07.303923: Printing
2023-03-15 06:18:08.368112: Printing
2023-03-15 06:18:09.431400: Printing
2023-03-15 06:18:10.496582: Printing
2023-03-15 06:18:11.561063: Printing
2023-03-15 06:18:12.620439: Printing
2023-03-15 06:18:13.680874: Printing
2023-03-15 06:18:14.719975: Printing
2023-03-15 06:18:15.781751: Printing
2023-03-15 06:18:16.843066: Printing
2023-03-15 06:18:17.904322: Printing
2023-03-15 06:18:18.952980: Printing
2023-03-15 06:18:20.018243: Printing
2023-03-15 06:18:21.080457: Printing
Error: [Errno 104] Connection reset by peer
2023-03-15 06:18:23.178556: Printing
2023-03-15 06:18:24.241051: Printing
2023-03-15 06:18:25.310895: Printing
2023-03-15 06:18:26.372014: Printing
2023-03-15 06:18:27.436914: Printing
Error: Server disconnected
2023-03-15 06:18:29.581242: Printing
Error: Server disconnected
2023-03-15 06:18:31.686184: Printing
Error: Server disconnected
2023-03-15 06:18:33.798404: Printing
Error: [Errno 104] Connection reset by peer
Error: Server disconnected
Error: [Errno 104] Connection reset by peer
2023-03-15 06:18:38.002935: Printing
2023-03-15 06:18:39.068669: Printing
Error: [Errno 104] Connection reset by peer
2023-03-15 06:18:41.188568: Printing
Error: [Errno 104] Connection reset by peer
Error: [Errno 104] Connection reset by peer
2023-03-15 06:18:44.365129: Printing
2023-03-15 06:18:45.429469: Pausing
Error: Server disconnected
2023-03-15 06:18:47.529224: Paused
2023-03-15 06:18:48.593871: Paused
2023-03-15 06:18:49.655448: Paused
2023-03-15 06:18:50.719292: Paused
2023-03-15 06:18:51.788281: Paused
2023-03-15 06:18:52.853820: Paused
2023-03-15 06:18:53.895486: Paused
```

Problems seem to be there, but minimal, during (virtual) heating up, and start to become more aparent when the GCode is processed (again, on a virtual printer, no real USB communication going on here).

cp2004, 2023-03-16:
I'm not sure where this issue *actually* lies, but it is interesting. In general I can say that OctoPrint's rest API doesn't have any issues responding to multiple requests in quick succession - every time you load the web UI, there are lots of requests sent in. On my install I can see 16 requests within one second.

So something must be different between the browser accessing OctoPrint and the clients you're using, and that is triggering this issue. The error message would seem to indicate that it is a sever issue rather than client I guess?

Do you have anything in front of OctoPrint like a reverse proxy? Or are you accessing OctoPrint directly?

markusressel, 2023-03-16:
Thx for the quick response.

> I'm not sure where this issue _actually_ lies, but it is interesting. In general I can say that OctoPrint's rest API doesn't have any issues responding to multiple requests in quick succession - [...]

Definitely interesting. As mentioned in the linked issue, the client libs (at least one of them) already seem to include delays to work around some kind of session validity issue. It might be worthwhile to ask the maintainers about the exact reason.

> So something must be different between the browser accessing OctoPrint and the clients you're using, and that is triggering this issue. The error message would seem to indicate that it is a sever issue rather than client I guess?

From the error message itself it would seem to indicate a server issue, yes. Also the fact that two different and independent libraries cause a very similar error scenario does. However, both libraries are using asyncio, so the issue might be related to that as well. However, this would also require both library maintainers to have made the exact same mistake, or an issue in the underlying http libraries used by both of them (I have not looked deeply into that yet, so this is pure speculation at this point).

> Do you have anything in front of OctoPrint like a reverse proxy? Or are you accessing OctoPrint directly?

There is no proxy involved. OctoPrint is running in a virtualenv "natively" on a Raspberry Pi 4 running Manjaro for ARM (as posted above). The client libraries are accessing it using the IP address of the Pi, so no DNS is involved either.

markusressel, 2023-03-19:
I would like to investigate this further, but I don't really know where to go from here. Could you give me some advice?
Maybe I can debug the webserver of OctoPrint? How easy is it to locally build and debug OctoPrint? Can I run (and debug) it in PyCharm somehow?

markusressel, 2023-03-20:
It seems like there are some specific actions that cause this issue to get exaggerated for quite a while, f.ex. simply uploading a file to octoprint from a browser, while running the test script, results in a pretty much unusable API for while.

I tried pinning this down further by running multiple clients in parallel, so the output isn't the prettiest thing, but I hope it is at least somewhat understandable:

* Dots in the output represent successful requests, they are printed without newlines
* Error prints are printed for each error
* the first number in the error line is the client ID (x in 0..9)
* the second number in the error line is the error count this specific client has reached in the current testing run (not that insightful without the total duration and other infos, but it shows that the issue is not only present once a year 😄 )

```
...
Client 0: Error (61): ClientOSError [Errno 104] Connection reset by peer
Client 1: Error (66): ClientOSError [Errno 104] Connection reset by peer
Client 4: Error (62): ClientOSError [Errno 104] Connection reset by peer
Client 5: Error (76): ServerDisconnectedError Server disconnected
Client 7: Error (65): ClientOSError [Errno 104] Connection reset by peer
Client 8: Error (76): ClientOSError [Errno 104] Connection reset by peer
Client 0: Error (62): ClientOSError [Errno 104] Connection reset by peer
.
Client 1: Error (67): ClientOSError [Errno 104] Connection reset by peer
Client 2: Error (68): ClientOSError [Errno 104] Connection reset by peer
..
Client 5: Error (77): ClientOSError [Errno 104] Connection reset by peer
Client 7: Error (66): ClientOSError [Errno 104] Connection reset by peer
Client 8: Error (77): ClientOSError [Errno 104] Connection reset by peer
Client 0: Error (63): ClientOSError [Errno 104] Connection reset by peer
Client 6: Error (76): ClientOSError [Errno 104] Connection reset by peer
Client 2: Error (69): ClientOSError [Errno 104] Connection reset by peer
Client 9: Error (69): ClientOSError [Errno 104] Connection reset by peer
.
Client 3: Error (72): ClientOSError [Errno 104] Connection reset by peer
```

This output represents the errors right after and during a file upload to the web frontend.

After a while the system seems to mostly recover:

```
...............................................................................................
Client 4: Error (77): ClientOSError [Errno 104] Connection reset by peer
.........................................................................................................................................................................
Client 4: Error (78): ClientOSError [Errno 104] Connection reset by peer
...............................................................................................................................................
Client 6: Error (98): ClientOSError [Errno 104] Connection reset by peer
................................................................................................................................................................................................
Client 0: Error (80): ClientOSError [Errno 104] Connection reset by peer
................................................................................................................................................
Client 7: Error (86): ClientOSError [Errno 104] Connection reset by peer
............................................................................................................................................................................................
Client 5: Error (94): ClientOSError [Errno 104] Connection reset by peer
.........................................................................................................................................................................
Client 5: Error (95): ClientOSError [Errno 104] Connection reset by peer
......................................................................................................................................................................
Client 0: Error (81): ClientOSError [Errno 104] Connection reset by peer
................................................................................................................................................................................................................................................................................................................................................
Client 3: Error (97): ClientOSError [Errno 104] Connection reset by peer
.....................................................................................
```

markusressel, 2023-03-20:
After some googling I stumbled upon this post:

https://stackoverflow.com/questions/51248714/aiohttp-client-exception-serverdisconnectederror-is-this-the-api-servers-issu

Utilizing the second suggestion of [this answer](https://stackoverflow.com/a/13929101/1941623) seems to have mitigated the issue entirely.

Using this:
```python
conn = aiohttp.TCPConnector(force_close=True)
    async with aiohttp.ClientSession(connector=conn) as websession:
```

instead of this (which is the default):
```python
conn = aiohttp.TCPConnector(force_close=False)
    async with aiohttp.ClientSession(connector=conn) as websession:
```

Results in a perfect success rate (dots are success):
```
............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
```

As I see it there are two ways to mitigate this issue:

1. Add some documentation about this to let API clients know that they cannot reuse a TCP connection for longer durations
2. Change the configuration of the Flask app to allow TCP connection reuse

I don't know much about flask, so I have no idea how feasible the second option is. However, I would prefer this option, since it would not require API clients to change. If it is not feasible, I will try to go ahead and open a PR on `pyoctoprintapi` for this.

markusressel, 2023-03-20:
When looking at the client libs again I realized that the session is not created by the client libraries, but the code using the library. It looks like the session used by the Home Assistant OctoPrint integration builds on the integrated methods by Home Assistant Core itself. The line creating the `TCPConnector` is this:

https://github.com/home-assistant/core/blob/d7de23fa6506b9e0f21c914966e2f6d3d639f25b/homeassistant/helpers/aiohttp_client.py#L278

I will try to investigate if using a custom connector would fix the "unavailable" issue within home assistant.

markusressel, 2023-03-20:
Reproducing the issue within Home Assistant was successful. The moment the state becomes "unavailable", the error log contains this:

```
2023-03-20 03:36:48.261 ERROR (MainThread) [homeassistant.components.octoprint] Error requesting octoprint-594e6f06e683623a8074787fe6c1b129 data: [Errno 104] Connection reset by peer
```

Adding the `force_close=True` parameter to the TCPConnector seems to fix the issue:

[image]

```python
connector = aiohttp.TCPConnector(
    force_close=False,
    enable_cleanup_closed=True,
    ssl=ssl_context,
    limit=MAXIMUM_CONNECTIONS,
    limit_per_host=MAXIMUM_CONNECTIONS_PER_HOST,
)
```

[image]

Changing this within Home Assistant is probably undesirable though, so the question of whether to modify the OctoPrint integration to use a separate session, or to update the OctoPrint webserver to fix this issue remains.

markusressel, 2023-03-21:
Reading [this](https://stackoverflow.com/questions/10523879/how-to-make-flask-keep-ajax-http-connection-alive), the issue might actually only come up because of the way I (and apparently others too) are running OctoPrint using the built in Webserver, which is not running in production mode?

cp2004, 2023-03-25:
Thanks for all the investigation into this!

I can understand the issue with the TCP connections and keep-alive, and it does make sense. I'm still not sure what could change on the OctoPrint side to mitigate the issue - there are lots of API clients out there making requests to the API lots of times without issues, so perhaps they already do this or it is something specific to aiohttp's implementation. I know I have seen code using `requests.get` lots to access OctoPrint's API and have not heard of issues there, but that doesn't mean the same thing doesn't exist. Maybe this behaviour is default there, not sure.

Commenting on your last comment,

> Reading [this](https://stackoverflow.com/questions/10523879/how-to-make-flask-keep-ajax-http-connection-alive), the issue might actually only come up because of the way I (and apparently others too) are running OctoPrint using the built in Webserver, which is not running in production mode?

OctoPrint has a Flask/Tornado webserver setup that should be suitable for running in 'production' - it doesn't use the built in Flask development server.

markusressel, 2023-03-25:
I also do think that this issue is specific to the use of `aiohttp`, because of its focus on performance rather than ease of use. `requests` probably doesn't even try to reuse the connection by default, because its not supported by all Webservers (as can be seen here) and adds "unnecessary" complexity for a library designed to make http easy to use.

I don't know if OctoPrint should be modified to support TCP connection reuse (or whatever is going on exactly, I honestly have no idea how this works). It depends on whether the performance gain is important in practice and is worth the time investment. Based on the fact that this has not been an issue for years, the answer to this is probably "no". However, both of the rest clients for OctoPrint that come up in search first have this issue, because both of them rely on aiohttp's default behavior. So as I see it the value of not having to change client implementations should be taken into account, even though I already submitted a mitigation PR to `pyoctoprintapi`. But again I have no idea about flask or tornado and cannot evaluate how much effort this would take.

foosel, 2023-03-27:
From my understanding of Tornado, the HTTP server should be supporting `keep-alive` out of the box, if the corresponding headers are set by the client. What I could imagine happening here is either that the `Connection` header isn't set properly, or that we have some weird issue due to the API being implemented in Flask, accessed through WSGI from Tornado, or due the custom connection class, that primarily just copies the original implementation, but also checks max body sizes.

foosel, 2023-11-22:
@markusressel I finally had a chance to look closer into this, but so far I haven't been able to reproduce the issue.

Do you still happen to have your test code that you [used above](https://github.com/OctoPrint/OctoPrint/issues/4764#issuecomment-1475471342) and could share it here? I'm trying to get a minimal example of triggering this issue with enabled keep alive going so I can see if I can maybe do something on OctoPrint's side after all, even though I fear for 1.10.0 it'll be a warning in the docs.

markusressel, 2023-11-22:
@foosel you can see most of the test code in the comments of this issue: https://github.com/home-assistant/core/issues/83624

This is a link to the exact comment: https://github.com/home-assistant/core/issues/83624#issuecomment-1469334565

While these are not full examples (the event loop setup is missing) it should be enough to reproduce the issue.
Keep in mind that on the home assistant side the issue has been fixed already, so you will need to use an older commit or come up with something else yourself if you want to reproduce the issue. You can **not** use the latest develop of home assistant.

foosel, 2023-11-23:
After looking at the examples I was actually able to reproduce it with this code:

``` python
import datetime
import time

import requests

KEEP_ALIVE = True
PRINTER_INFO = (
    "http://127.0.0.1:5000/api/printer?apikey=72C26A2DA6D14576B02C9690B694C19F"
)
JOB_INFO = "http://127.0.0.1:5000/api/job?apikey=72C26A2DA6D14576B02C9690B694C19F"
SLEEP = 0.1

def main():
    session = requests.Session()
    while True:
        try:
            printer_info = session.get(PRINTER_INFO)
            _ = printer_info.json()

            job_info = session.get(JOB_INFO)
            _ = job_info.json()

            print(".", end="", flush=True)
        except Exception as ex:
            print("!", end="", flush=True)
            print(f"Error: {ex}")
        finally:
            time.sleep(SLEEP)

if __name__ == "__main__":
    main()
```

And after the above changes I was no longer able to reproduce it at all, no matter how often I paused/resumed, connected/disconnected or uploaded files.

It seems to indeed have been caused by connection reuse, which for some reason caused issues before, but with the new code as copied from tornado's `WSGIContainer`* things seem to work.

A welcome side effect of that change is that long running flask requests now no longer should be able to completely block Tornado.

So I'm carefully marking this as done, we'll see during the RC phase if this has unexpected side effects, but so far things look fine.

---

* We need a special version of that to be able to rewrite the body of incoming upload requests, and for some header additions and removals that otherwise weren't possible to do globally.

foosel, 2023-11-23:
Well, ok, that broke the E2E tests, investigating.

foosel, 2023-11-23:
Reverted for now, as somethings is definitely completely broken in E2E test now. Getting server hang up messages when attempting to log in via the API in prep for a test, works fine through curl, no clue wtf is going on. But calling it a day, it's been a long an intense one.

foosel, 2023-11-27:
Managed to solve the issues with the change, tests are now all green. So this will be in 1.10.0.

Commit is 703ec2da4cc2f34dcdaf35a36805ff6bedd1230c (forgot to tag this issue)

foosel, 2024-04-25:
1.10.0 has been released.