OctoPrint/OctoPrint #4765: Octoprint 1.9.0rc2 does not start on Debian 12

guillaume-pais, 2023-03-16:
### The problem

We tried Octoprint 1.9.0rc2 and it does not seem to run on Debian 12:

```
guillaume@debian:~$ cd /tmp/
guillaume@debian:/tmp$ mkdir octo
guillaume@debian:/tmp$ cd octo/
git clone https://github.com/OctoPrint/OctoPrint.git
Cloning into 'OctoPrint'...
remote: Enumerating objects: 85567, done.
remote: Counting objects: 100% (584/584), done.
remote: Compressing objects: 100% (350/350), done.
remote: Total 85567 (delta 264), reused 502 (delta 223), pack-reused 84983
Receiving objects: 100% (85567/85567), 51.33 MiB | 13.47 MiB/s, done.
Resolving deltas: 100% (55873/55873), done.
guillaume@debian:/tmp/octo$ cd OctoPrint/
guillaume@debian:/tmp/octo/OctoPrint$ git checkout 1.9.0rc2
Note: switching to '1.9.0rc2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 95e87eb19 ⏪ Bring back `octoprint_setuptools`
guillaume@debian:/tmp/octo/OctoPrint$ cd ..
guillaume@debian:/tmp/octo$ virtualenv --python=python3 venv
created virtual environment CPython3.11.2.final.0-64 in 207ms
  creator CPython3Posix(dest=/tmp/octo/venv, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/guillaume/.local/share/virtualenv)
    added seed packages: pip==23.0.1, setuptools==66.1.1, wheel==0.38.4
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
guillaume@debian:/tmp/octo$ . venv/bin/activate
(venv) guillaume@debian:/tmp/octo$ pip3 install ./OctoPrint/
Processing ./OctoPrint
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting OctoPrint-FileCheck>=2021.2.23
  Using cached OctoPrint_FileCheck-2021.2.23-py2.py3-none-any.whl (19 kB)
Collecting OctoPrint-FirmwareCheck>=2021.10.11
  Using cached OctoPrint_FirmwareCheck-2021.10.11-py2.py3-none-any.whl (33 kB)
Collecting OctoPrint-PiSupport>=2022.6.13
  Using cached OctoPrint_PiSupport-2022.6.13-py2.py3-none-any.whl (33 kB)
Collecting argon2-cffi<22,>=21.3.0
  Using cached argon2_cffi-21.3.0-py3-none-any.whl (14 kB)
Collecting Babel<2.13,>=2.12.1
  Using cached Babel-2.12.1-py3-none-any.whl (10.1 MB)
Collecting cachelib<0.11,>=0.10.2
  Using cached cachelib-0.10.2-py3-none-any.whl (18 kB)
Collecting Click<9,>=8.1.3
  Using cached click-8.1.3-py3-none-any.whl (96 kB)
Collecting colorlog<7,>=6.7.0
  Using cached colorlog-6.7.0-py2.py3-none-any.whl (11 kB)
Collecting emoji<3,>=2.2.0
  Using cached emoji-2.2.0-py3-none-any.whl
Collecting feedparser<7,>=6.0.10
  Using cached feedparser-6.0.10-py3-none-any.whl (81 kB)
Collecting filetype<2,>=1.2.0
[... 178 lines cut ...]

Yes, it did persist

### Version of OctoPrint

1.9.0rc2

### Operating system running OctoPrint

Debian 12

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

We cannot get any bundle. It is empty:
```
(venv) guillaume@debian:/tmp/octo$ octoprint systeminfo
(venv) guillaume@debian:/tmp/octo$ 
```

guillaume-pais, 2023-03-16:
Hi @foosel, 

we cannot get any system info bundle from the command line. We added the ```octoprint.log``` file.

Thanks

guillaume-pais, 2023-03-20:
Thanks for your answer.

We had a conflict with ```snapd```. After uninstall everything works fine.

closed 2023-03-20 by guillaume-pais

foosel, 2023-03-20:
You could also just have told OctoPrint to use a different port with `--port` or via [`config.yaml`](https://docs.octoprint.org/en/master/configuration/config_yaml.html#server).

jneilliii, 2024-01-09:
have you tried the octoprint_deploy script?

DonCanDo, 2024-01-09:
Sure it fails too founding my printer not

Grüße Steven