OctoPrint/OctoPrint #4766: autologin sometimes logs in with wrong user after power cycle

timur-tabi, 2023-03-17:
### The problem

I don't know if this is a problem with the OctoPrint-AutoLoginConfig plugin or with Octoprint itself, since the plugin author says that his plugin is only a configuration tool.  I apologize if I don't provide enough information up front.

I have two users configured, "timur" and "prusaslicer".  The second user is used by Prusa Slicer to upload files to Octoprint.  The first user is my primary user and what I use when I log into the web interface.

Sometimes, after a power cycle of my RPi, when I first load the web interface, the autologin plugin with log me in with the "prusaslicer" user.  Since this user is limited to only uploading files, the web interface is degraded.  I have to manually log out and log in with "timur" in order to fix this.  After that, I cannot repro the problem until (maybe) after the next power cycle.

Looking at the log, it appears that maybe Octoprint is logging itself in with the prusaslicer account before the autologin plugin has a chance to run.  That's just a guess.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

It happens infrequently, so I notice it only when I start the server normally.

### Version of OctoPrint

1.8.7

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Prusa MK3s

### Browser and version of browser, operating system running browser

Firefox for Linux 111.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230317013622.zip](https://github.com/OctoPrint/OctoPrint/files/10997048/octoprint-systeminfo-20230317013622.zip)

cp2004, 2023-03-19:
Can you share a screenshot (or the specific section of the config.yaml file) showing how you have configured autologin? I can see that it is logging in the `prusaslicer` user via autologin, but this should only happen if that is how it is configured. The autologin plugin just configures OctoPrint's built in autologin feature through the UI, it doesn't actually do any actions itself so I would say this is an OctoPrint-related issue.

timur-tabi, 2023-03-19:
[image]

I think the problem may be that the plugin forgets what I set it up.  The screenshot shows the default login is prusaslicer, but last time I loaded this page, I set it to 'timur' and saved it.

I changed this to 'timur' and hit save.  If I close and reopen the config page, it says 'timur'.  But at some point in the future, it's going to reset to prusaslicer again.

jneilliii, 2023-03-19:
@cp2004 I wonder if your select list might need `valueAllowUnset: true` added to the data-bind?

jneilliii, 2023-03-19:
@timur-tabi the settings this plugin sets is built into core OctoPrint. If you set it the way you like and then disable the plugin does it revert after time then? This would help identify if it's a plugin issue or not.

cp2004, 2023-03-19:
Yeah it does seem to be an issue with the plugin if you choose a user that is not at the top of the list. I'll get that sorted ASAP.

closed 2023-03-19 by cp2004

timur-tabi, 2023-03-19:
I'll try it and let you know.  However, I just opened the config panel again, and it once again says prusaslicer, despite the fact that I it's only been a few hours and all I've been doing is printing. 

So right now it says prusaslicer.  I rebooted my Pi, and it logged me in as 'timur'. I open the plugin config panel, and it says prusaslicer.  So now I'm confused.

I changed it to 'timur' again, and then disabled the plug.  I'm power-cycling my Pi now, but I'm not sure what's going on now.

cp2004, 2023-03-19:
That strangeness you are seeing is likely because the plugin was effectively showing the wrong user in the UI. It would have only been set into OctoPrint's configuration if you hit save after noticing that it was showing wrong, so autologin would remain with the older configuration.