OctoPrint/OctoPrint #4768: Downloading a file often prompts me to reload the web interface

timur-tabi, 2023-03-17:
### The problem

This started happening recently, even though I haven't updated Octoprint in a while.  Most of the time, when I download a file (either a gcode or a log file), I am immediately prompted to reload the web interface because "There is a new version of the server active now".

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why

With safe mode enabled, instead of being prompted to restart the server, the "Safe mode is active" yellow pop-up reappears every time I try to download something.

### Version of OctoPrint

1.8.7

### Operating system running OctoPrint

OctoPi 0.18.0

### Browser and version of browser, operating system running browser

Firefox for Linux 111.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230317034316.zip](https://github.com/OctoPrint/OctoPrint/files/10997629/octoprint-systeminfo-20230317034316.zip)

Javascript console log (safe mode enabled):
Navigated to https://octopi.local/downloads/files/local/Dominion_Carousel_Bottom_Wedge_031423_0.3mm_PETG_MK3S_3h38m.gcode
Connected to the server [packed_core.js:1136:456](https://octopi.local/static/webassets/packed_core.js?389582e5)
Login mechanism: http [packed_core.js:234:35](https://octopi.local/static/webassets/packed_core.js?389582e5)
❗ Safe mode is active. Third party plugins and language packs are disabled and cannot be enabled. [packed_core.js:1137:110](https://octopi.local/static/webassets/packed_core.js?389582e5)
❗ Reason for safe mode: settings [packed_core.js:1137:220](https://octopi.local/static/webassets/packed_core.js?389582e5)
Triggering connect on all view models [packed_core.js:1141:240](https://octopi.local/static/webassets/packed_core.js?389582e5)
Server (re)connect processed [packed_core.js:1143:5](https://octopi.local/static/webassets/packed_core.js?389582e5)

​

cp2004, 2023-03-19:
Are you able to try it with a different browser or from a different device?

timur-tabi, 2023-03-19:
Well, I can't repro the problem any more, on either Firefox or Chrome.  I'll keep running tests, but maybe it was a fluke.

timur-tabi, 2023-03-24:
I got it to repro once again with Firefox, but then it stopped.  I've been unable to repro it at all with Chrome.  I can only assume that it's the result of some Firefox plugin.  If I can repro it consistently, I will re-open the bug.

closed 2023-03-24 by timur-tabi