OctoPrint/OctoPrint #4770: [RC Feedback] Feedback on the 1.9.0rc3 Release Candidate

foosel, 2023-03-22:
Please provide *general* feedback on your experience with the 1.9.0rc3 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### ❗👇 IMPORTANT 👇❗
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports *with system info bundles*, not just a statement that something doesn't work!
### ❗👆 IMPORTANT 👆❗

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

- UiCustomizer plugin: https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/281 - fix ready, as of yet unreleased
- FirmwareUpdater plugin: https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/341 - fix ready, as of yet unreleased

### Issues needing analysis

jneilliii, 2023-03-22:
sync with job progress not working: https://github.com/OctoPrint/OctoPrint/issues/4772

The-EG, 2023-03-22:
The Firmware Updater fix is in release: <https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/releases/tag/1.14.0>

jneilliii, 2023-03-24:
is anyone else getting 404 not found for an image file on page load? 

[image]

The css below seems to be pathed incorrectly? no third-party plugins installed.

```
#temperature-graph {
    height: 350px;
    width: 100%;
    background: url(../../plugin/classicwebcam/static/img/graph-background.png) center no-repeat
}
```

TheMedGeek, 2023-03-25:
> Updated test RPi 4B system (with camera V3) to 1.9.0rc3 from 1.8.7. I'm not sure what version of OctoPi 1.0.0 I started with. The camera doesn't work at the moment but I'll wait for the next new camera stack image.

Same issue

TheMedGeek, 2023-03-25:
> Updated test RPi 4B system (with camera V3) to 1.9.0rc3 from 1.8.7. I'm not sure what version of OctoPi 1.0.0 I started with. The camera doesn't work at the moment but I'll wait for the next new camera stack image.

OK I got it working again - I did two moves so not sure which one worked but I'm gonna keep it since its working ha ha

edited /boot/config.txt 
commented out the line "dtoverlay=imx708" to now read "#dtoverlay=imx708"

edited /boot/octopi.txt
noticed it was now showing 

### EXPERIMENTAL
# Support for different streamer types.
#
# Available options:
#   mjpeg [default] - camera-streamer
#   hls - FFMPEG HLS streamer
#camera_streamer=mjpeg
#camera_streamer=libcamera
camera="libcamera"
camera_libcamera_options=""

#enable libcamera
camera="libcamera"
camera_libcamera_options="-r 1280x720"

.... .so I commented out the first to so it now shows

### EXPERIMENTAL
# Support for different streamer types.
#
# Available options:
#   mjpeg [default] - camera-streamer
#   hls - FFMPEG HLS streamer
#camera_streamer=mjpeg
#camera_streamer=libcamera
#camera="libcamera"
#camera_libcamera_options=""

#enable libcamera
camera="libcamera"
camera_libcamera_options="-r 1280x720"

foosel, 2023-03-27:
> I'm not sure what version of OctoPi 1.0.0 I started with

Neither am I, because those options you quoted there don't even remotely ring a bell with me (and I've been all up in libcamera support with the OctoPi-UpToDate image through a switch to camera-streamer the past few weeks, when I was not focused on 1.9.0). In any case, that's webcam server specific stuff and has nothing to do with 1.9.0rc ;)

ChrisHeerschap, 2023-04-01:
Just thought "oh wow I should install that RC" and went to do it and realized that I had already installed it on all three of my Raspis, and I've been using two of them heavily and the other one occasionally. I guess the fact that I forgot I installed it is proof enough that this one is working _just fine_.

If it matters:

Raspi 4 with octopi 0.18.0 (although I've kept the OS up to date) with a Prusa MK3S
Raspi 4 with octopi 1.0.0 with a Prusa MK3S+
Raspi 3 with octopi 0.17.0 (OS kept updated) on a Prusa Mini

All three running 1.9.3RC3 with zero complaints.

foosel, 2023-04-12:
1.9.0rc4 is out, new ticket here: #4790

closed 2023-04-12 by foosel