OctoPrint/OctoPrint #4772: Sync With Progress Broken in 1.9.0rc3

jneilliii, 2023-03-22:
### The problem

Start a print and go to gcode viewer tab with the option to sync with job progress and it remains blank. 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.0rc3

### Operating system running OctoPrint

OctoPi 1.0.0

### Printer model & used firmware incl. version

Virtual Printer

### Browser and version of browser, operating system running browser

Chrome Version 111.0.5563.66 (Official Build) (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

clicking on center viewport on model and zoom in on model will flash the full layer for a second and then gets cleared again. navigating layers manually show the layers as expected. 

[octoprint-systeminfo-20230322100233.zip](https://github.com/OctoPrint/OctoPrint/files/11040797/octoprint-systeminfo-20230322100233.zip)

the below javascript console error only happened with BLV and PTG plugins installed. no errors with all plugins disabled.
 
[192.168.126.109-1679492740516.log](https://github.com/OctoPrint/OctoPrint/files/11040817/192.168.126.109-1679492740516.log)

[CCR10_3DBenchy.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/11040830/CCR10_3DBenchy.gcode.txt)

foosel, 2023-03-23:
Will take a look at that today, but would be good if you could also take a look at it @JoveToo, my current guess would be that whatever is up here is caused by some of the GCode Viewer changes.

foosel, 2023-03-23:
Hmmm.... @jneilliii I can't seem to be able to reproduce this here 🤔

First I tested in Firefox (110.0 (64-bit)):

[Screencast from 2023-03-23 12-51-07.webm](https://user-images.githubusercontent.com/83657/227195636-9a0a6dff-9898-4e43-994a-104e33f75f77.webm)

Then in Chromium (Version 111.0.5563.64 (Official Build, ungoogled-chromium) (64-bit)):

[Screencast from 2023-03-23 12-56-03.webm](https://user-images.githubusercontent.com/83657/227196327-36fc0dc9-bc13-4772-ad47-a7e6a59209ac.webm)

Also works centered and zoomed in:

[Screencast from 2023-03-23 12-56-45.webm](https://user-images.githubusercontent.com/83657/227196519-77874939-0319-4f68-b266-eba0489b5b44.webm)

Am I misunderstanding the issue? Tested this on Fedora 37 first, then also tried on Windows 10. Everything seems to work.

jneilliii, 2023-03-23:
I'll try to get a video, one thing I notice on your captures is that the prime line isn't drawn (assuming you're using the same gcode file). For me the prime line would render in the gcode viewer tab, then freeze from there and not progress further.

foosel, 2023-03-23:
I saw a prime line at the start being drawn (along the y axis on the left), but my captured started later.

I think I've got an idea where in the code it failed in your stack trace, so I could harden it against that, but I'd also like to figure out what's happening.

jneilliii, 2023-03-23:
sure, I'll do additional testing when I can. I was able to reproduce the same thing on two separate instances.

foosel, 2023-03-23:
@JoGrob can you also share a gcode file with which to reproduce?

Also, under Advanced options, is "Hide empty layers" checked or unchecked?

jneilliii, 2023-03-23:
> Do you also see it loading twice (sometimes) when switching to the viewer tab? I'm wondering if there might be a race condition to blame here.

Can't say I remember it doing that.

> Also, under Advanced options, is "Hide empty layers" checked or unchecked?

Checked on both instances. Tried unchecking and seems to not make a difference, issue persists.

JoGrob, 2023-03-23:
> @JoGrob can you also share a gcode file with which to reproduce?
[Chick_100.zip](https://github.com/OctoPrint/OctoPrint/files/11052790/Chick_100.zip)
 
> Also, under Advanced options, is "Hide empty layers" checked or unchecked?
I have tested "Hide empty layers" checkund and unchecked, both dont care.

JoGrob, 2023-03-23:
> Do you also see it loading twice (sometimes) when switching to the viewer tab?
> I'm wondering if there might be a race condition to blame here.

I can confirm loading twice sometimes.

foosel, 2023-03-23:
Figured out the loading twice issue, so that's fixed. Also made sure the exception in the stack trace from @jneilliii should no longer be able to happen. But that doesn't fix the core cause here, only (hopefully) renders the viewer functional again. Sadly can't test because I still can't reproduce.

JoveToo, 2023-03-23:
@jneilliii does the issue still happen with those fixes from foosel?

jneilliii, 2023-03-23:
I won't know until much later tonight but we'll give it a test as soon as possible.

konstantin-kelemen, 2023-03-23:
I'm also having the same issue. I've applied the changes locally to `/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer/static/js/viewer/reader.js`, rebooted the pi, cleared the browser cache but still can reproduce the issue.

JoveToo, 2023-03-23:
> I'm also having the same issue. 

@konstantin-kelemen  Please share a gcode file.

In the video above, shared by @JoGrob, the display did not go blank. It just stopped rendering after the initial purge line.

I can reproduce the issue.

If I look in the model, all percentages are set to infinity. And any percentage is smaller than infinity, resulting in every render entry returning the first command...

I traced the issue to the download of the gcode file from OctoPrint server. The parser determines the length of the file to download by getting the header "content-length" (worker.js:356) but this length is null. The result is that all percentage calculations are divisions by 0.... resulting in infinity.

JoveToo, 2023-03-24:
I am also running Firefox 111.0

jneilliii, 2023-03-24:
When selecting the gcode file without printing while on the GCODE Viewer tab I do see (infinity) show up for a short period in the progress and then it shows percentage numbers and it eventually shows the first layer. Once I start the print I get the prime line like others have mentioned and then nothing.

mioux, 2023-03-24:
Same here, it's not a problem getting informations from the printer as Pretty Gcode continue to be synchronized.

https://user-images.githubusercontent.com/6904170/227459540-a08e2125-1af7-4b39-bd89-614498dcb087.mp4

foosel, 2023-03-24:
To summarize some discussion from Discord, the content-length is not set on the file download, causing the progress calculation to break. The reason for that is as of yet unknown, but might be down to Tornado, which could also explain why some people see this while others (me) don't, thanks to version differences.

Will look into this further on Monday.

mioux, 2023-03-25:
By the way, I was looking at HAProxy, gzip compression is not recommended on pi-like computers. 

With 1GB RAM/1 core, the expected traffic is "Very low SSL / GZip". (Pi 3 has this configuration)
https://www.haproxy.com/documentation/hapee/latest/getting-started/hardware/

As Octopi is not designed to be used over Internet (a reverse proxy is recommended in documentation), why compressing at pi level ?
X-Original-Content-Length is still needed if there is another reverse proxy between Octopi and client

cp2004, 2023-03-25:
Somewhere I have seen some graphs of performance tests related to compression - I'll see if I can dig them out, but I suspect haproxy compression was enabled as a result of tests that showed it performed better.

foosel, 2023-03-27:
> I suspect haproxy compression was enabled as a result of tests that showed it performed better.

I have no memory. Could just as well have been copied from an example somewhere. Not really in the scope of this report (or repository for that matter) but these days I'd also like to rather get away from haproxy for what we use it for. 10 years ago haproxy was chosen because back then websocket support with the other options on the market was spotty at best. These days things are entirely different, and we don't really need high availability features.

foosel, 2023-03-27:
Fix merged (thanks @JoveToo!) and ready for 1.9.0rc4

luizbgomide, 2023-04-06:
I'm also affected by this issue. Just I just make the changes from #4775? Where in the octopi should I change the files?

jhoff80, 2023-04-07:
> I'm also affected by this issue. Just I just make the changes from #4775? Where in the octopi should I change the files?

ssh into the pi:

`cd /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/gcodeviewer/static/js/viewer
nano worker.js`

make the edits from the first part of the commit.

`cd /home/pi/oprint/lib/python3.9/site-packages/octoprint/server/util
nano tornado.py`

make the edits from the second part of the commit.  Or just wait for rc4.

foosel, 2023-04-12:
1.9.0rc4 has been released.

commit a4b0d340 (2023-04-12): 🐛 Add X-Original-Content-Length to file downloads (#4775)

Avoids issues with chunked transfers that remove the "Content-Length" header, which the GCODE viewer requires for calculating percentages for progress monitoring.

Closes #4772
closed 2023-04-12 by foosel