OctoPrint/OctoPrint #4773: Gcode viewer not syncing with print job

romses, 2023-03-24:
### The problem

Upgrade to Octoprint 1.9.0rc3
Start a Print job
Open the Gcode viewer
Check, if "Job mit Fortschritt synchronisieren" is checked

You should see the current progress of the print job

The current progress is not shown

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

I am currently running a print job.

### Version of OctoPrint

1.9.0rc3

### Operating system running OctoPrint

0.18.0

### Printer model & used firmware incl. version

Prusa MK3S+

### Browser and version of browser, operating system running browser

Google Chrome Version 111.0.5563.110 (Offizieller Build) (64-Bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

romses, 2023-03-24:
I attached the system info bundle

foosel, 2023-03-24:
Duplicate of #4772 😉

closed 2023-03-24 by foosel