OctoPrint/OctoPrint #4777: [Request] On the "control" tab add an option for changing feed rate on next layer

MDReptile, 2023-03-25:
### Describe the solution you'd like

It would be very handy to be able to start a print and set a low speed for the feed rate, then have octopi / octoprint allow me to set a different (faster) feed rate once it begins the second layer.

This would be good for a very accurate first layer without having to monitor the print and wait for the second layer to manually change the feed rate.

jneilliii, 2023-03-25:
This seems more appropriate to do in your slicer. An example showing where it's done in PrusaSlicer's speed settings.

[image]

MDReptile, 2023-03-25:
I suppose that is possible in most if not all slicers - but even so it would be useful at print time too!

foosel, 2023-03-27:
This is definitely outside of the scope for OctoPrint and something that should be done in the slicer. Alternatively, this could be done through a plugin.

closed 2023-03-27 by foosel