OctoPrint/OctoPrint #4778: [1.9.0rc3] 'Default Webcam' setting doesn't seem to apply to the control tab

cp2004, 2023-03-26:
### The problem

I'm not completely 'sure' what this setting is supposed to control, but my intuition said that it would mean that the webcam I selected under 'Webcam & Timelapse' > 'Default webcam' would be the one that shows up when I first go to open the control tab on page load. It doesn't do this, and I can't see a reference to some code that might be trying but failing, just that it doesn't exist.

In my opinion, it would make sense for this to be the case.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.0rc3

### Operating system running OctoPrint

OctoPi 1.0.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

jneilliii, 2023-03-27:
I agree that would be the intuitive thing for it to do. 

> It doesn't do this,

Does it just select the classic webcam or not display anything at all or display all registered webcam templates? If you disable the classic webcam plugin does your replacement then take over?

cp2004, 2023-03-27:
> Does it just select the classic webcam or not display anything at all or display all registered webcam templates? If you disable the classic webcam plugin does your replacement then take over?

The tabbed interface works fine, but regardless of whichever webcam I set as default, it doesn't change which webcam is active when you first load the page. You can change it manually, which works fine, but would have to do that every time you load the page.

foosel, 2023-03-27:
That definitely sounds like an oversight and I agree, that should be corrected.

jneilliii, 2023-03-27:
Winner if it's another case of allowUnset  being needed in select list possibly, haven't looked at the code yet in depth to see if the ninja is setting the active class on tab or not.

foosel, 2023-03-27:
It's jinja doing this, and it's simply selecting the first entry in the template list.

There's one problem I see here. A plugin can potentially provide more than one template for embedding a webcam. The default webcam is selected from a list of webcam configurations returned from plugins. And there's no link between the two, so in theory, you could have one webcam configuration that you can select as default to use to populate the backwards compatible parts of the API (which this default is currently used for) but which provides two or more webcam templates.

And tbh I'm not sure if I want to change that, since it makes things very much flexible (e.g. think about multiple view points from some camera implementation). 

Some wording changes could help here, to make clear that this "default" is only used to populate the backwards compatible API settings (suggestions welcome!), but then we'd still need another option to set a default for the selected stream here... or just move that into local storage? 🤔

foosel, 2023-03-27:
Now synced to local storage, leaving only the wording change. Any suggestions how to make clear this is only used for the stream URL shared on the backwards compatible API?

Currently it says

> Selects the default webcam. If this webcam provides a MJPEG video stream, it will be available for old clients that do not make use of the webcam plugin system introduced with OctoPrint 1.9.0

foosel, 2023-04-11:
No problem, happens to me all the time ^^

foosel, 2023-04-11:
After looking a bit more into the code, it turns out that this was still not accurate. So it's now this:

[image]

foosel, 2023-04-12:
1.9.0rc4 has been released.

closed 2023-04-12 by foosel

b-morgan, 2023-04-12:
I think "Select the webcam to fall back to to populate..." might read better if it was "Select the webcam to fall back on to populate..."

foosel, 2023-04-13:
Good point, will be in 1.9.0rc4

commit 0f922283 (2023-04-11): 🚸 Improve documentation of use of Default Webcam

Make clear it's only used on the webcam compat layer on the settings API

Also fix some HTML semantics.

Closes #4778