OctoPrint/OctoPrint #4779: Login not possible on Rock64 (aarch64)

cperrin88, 2023-03-28:
### The problem

When logging in there is an Authentication error even with a newly created account through the assistant. 

I have narrowed down the problem to the hashing of the password with argon2. Somehow the hashes are not valid. I have written a simple testcase that I have added to the `UserManagerTest`:
```python
    def test_check_password(self):
        password = "test1234"
        password_hash = octoprint.access.users.UserManager.create_password_hash(password)
        user = octoprint.access.users.User(
            "username",
            password_hash,
            True,
            permissions=[],
            apikey="apikey",
            settings={"key": "value"},
        )
        
        self.assertTrue(user.check_password(password))
```

On x86_64 this testcase is succesful on my Rock64 this fails. If I comment out the argon2 import the test is also successful. When I tested the hash that was produced during the run I couldn't verify the password. 

This was the hash of `test1234`:
```
$argon2id$v=19$m=65536,t=3,p=4$EOJ8b805JyQE4HwPIWTs3Q$2pnkpbDdKmuGMb+C5LnUUK5a14vRhace/joHZiKAyYc
```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

master

### Operating system running OctoPrint

Armbian (23.02.2) aarch64

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230328161258.zip](https://github.com/OctoPrint/OctoPrint/files/11090468/octoprint-systeminfo-20230328161258.zip)

foosel, 2023-03-29:
Patch merged (thanks!), will be in 1.10.0 (or maybe made part of a bugfix release in the 1.9.x cycle beforehand, if this turns out to be a more common issue).

closed 2023-07-18 by foosel

foosel, 2023-07-18:
The fix for this has been backported to 1.9.x and just been released as part of 1.9.2.

commit df60defc (2023-03-29): 🥅 Workaround for faulty argon2 hashes on aarch64 (#4780)

Ensure we can successfully verify an argon2 hash before using it.

Closes #4779
commit cf86f08d (2023-07-18): 🥅 Workaround for faulty argon2 hashes on aarch64 (#4780)

Ensure we can successfully verify an argon2 hash before using it.

Closes #4779