OctoPrint/OctoPrint #4781: octoprint not upgradable from 1.7.3 to 1.8.7 because of wrapt

MasterDimmy, 2023-03-30:
### The problem

cant update octoprint from 1.7.3 to 1.8.7

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

orange pi (ubuntu)

### Printer model & used firmware incl. version

tronxy x5sa

### Browser and version of browser, operating system running browser

-

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

```
t# pip install --upgrade octoprint
Requirement already satisfied: octoprint in /usr/local/lib/python3.7/dist-packages (1.7.3)
Collecting octoprint
  Downloading OctoPrint-1.8.7-py2.py3-none-any.whl (3.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.9/3.9 MB 5.1 MB/s eta 0:00:00
Requirement already satisfied: tornado<7,>=6.0.4 in /usr/local/lib/python3.7/dist-packages (from octoprint) (6.2)
Collecting passlib<2,>=1.7.4
  Downloading passlib-1.7.4-py2.py3-none-any.whl (525 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 525.6/525.6 kB 7.2 MB/s eta 0:00:00
Collecting Flask-Babel<3,>=2.0
  Downloading Flask_Babel-2.0.0-py3-none-any.whl (9.3 kB)
Collecting colorlog<7,>=6
  Downloading colorlog-6.7.0-py2.py3-none-any.whl (11 kB)
Collecting argon2-cffi<22,>=21.3.0
  Downloading argon2_cffi-21.3.0-py3-none-any.whl (14 kB)
Requirement already satisfied: future<1,>=0.18.2 in /usr/local/lib/python3.7/dist-packages (from octoprint) (0.18.3)
Requirement already satisfied: emoji<2,>=1.4.2 in /usr/local/lib/python3.7/dist-packages (from octoprint) (1.7.0)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /usr/local/lib/python3.7/dist-packages (from octoprint) (2.0)
Requirement already satisfied: pathvalidate<3,>=2.4.1 in /usr/local/lib/python3.7/dist-packages (from octoprint) (2.5.2)
Requirement already satisfied: regex in /usr/local/lib/python3.7/dist-packages (from octoprint) (2022.10.31)
Requirement already satisfied: PyYAML<6,>=5.4.1 in /usr/local/lib/python3.7/dist-packages (from octoprint) (5.4.1)
Requirement already satisfied: pyserial<4,>=3.4 in /usr/local/lib/python3.7/dist-packages (from octoprint) (3.5)
Requirement already satisfied: requests<3,>=2.26.0 in /usr/local/lib/python3.7/dist-packages (from octoprint) (2.28.2)
Requirement already satisfied: sarge==0.1.6 in /usr/local/lib/python3.7/dist-packages (from octoprint) (0.1.6)
Requirement already satisfied: filetype<2,>=1.0.7 in /usr/local/lib/python3.7/dist-packages (from octoprint) (1.2.0)
Collecting markdown<4,>=3.2.2
  Downloading Markdown-3.4.3-py3-none-any.whl (93 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 93.9/93.9 kB 9.4 MB/s eta 0:00:00
Collecting Click<9,>=8.0.3
  Downloading click-8.1.3-py3-none-any.whl (96 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 96.6/96.6 kB 9.4 MB/s eta 0:00:00
Collecting watchdog<2,>=1
  Downloading watchdog-1.0.2-py3-none-manylinux2014_armv7l.whl (72 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 72.6/72.6 kB 9.3 MB/s eta 0:00:00
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /usr/local/lib/python3.7/dist-packages (from octoprint) (2021.10.11)
Requirement already satisfied: unidecode in /usr/local/lib/python3.7/dist-packages (from octoprint) (1.3.6)
Collecting Flask-Limiter<3,>=2.6
  Downloading Flask_Limiter-2.9.2-py3-none-any.whl (27 kB)
Collecting cachelib<0.3,>=0.2
  Downloading cachelib-0.2.0-py3-none-any.whl (12 kB)
Requirement already satisfied: netaddr<0.9,>=0.8 in /usr/local/lib/python3.7/dist-packages (from octoprint) (0.8.0)
Requirement already satisfied: blinker<2,>=1.4 in /usr/local/lib/python3.7/dist-packages (from octoprint) (1.5)
Requirement already satisfied: feedparser<7,>=6.0.8 in /usr/local/lib/python3.7/dist-packages (from octoprint) (6.0.10)
Collecting frozendict<3,>=2.0
  Downloading frozendict-2.3.6.tar.gz (35 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: pkginfo<2,>=1.7.1 in /usr/local/lib/python3.7/dist-packages (from octoprint) (1.9.6)
Collecting websocket-client<2,>=1.2.1
  Downloading websocket_client-1.5.1-py3-none-any.whl (55 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 55.9/55.9 kB 9.2 MB/s eta 0:00:00
Requirement already satisfied: netifaces<1,>=0.11 in /usr/local/lib/python3.7/dist-packages (from octoprint) (0.11.0)
Collecting wrapt<1.15,>=1.14
  Downloading wrapt-1.14.1.tar.gz (50 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 50.9/50.9 kB 9.3 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
  WARNING: Requested wrapt<1.15,>=1.14 from https://files.pythonhosted.org/packages/11/eb/e06e77394d6cf09977d92bff310cb0392930c08a338f99af6066a5a98f92/wrapt-1.14.1.tar.gz (from octoprint), but installing version 1.12.1
Discarding https://files.pythonhosted.org/packages/11/eb/e06e77394d6cf09977d92bff310cb0392930c08a338f99af6066a5a98f92/wrapt-1.14.1.tar.gz (from https://pypi.org/simple/wrapt/) (requires-python:!=3.0.*,!=3.1.*,!=3.2.*,!=3.3.*,!=3.4.*,>=2.7): Requested wrapt<1.15,>=1.14 from https://files.pythonhosted.org/packages/11/eb/e06e77394d6cf09977d92bff310cb0392930c08a338f99af6066a5a98f92/wrapt-1.14.1.tar.gz (from octoprint) has inconsistent version: expected '1.14.1', but metadata has '1.12.1'
  Downloading wrapt-1.14.0.tar.gz (50 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 50.8/50.8 kB 9.2 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
  WARNING: Requested wrapt<1.15,>=1.14 from https://files.pythonhosted.org/packages/c7/b4/3a937c7f8ee4751b38274c8542e02f42ebf3e080f1344c4a2aff6416630e/wrapt-1.14.0.tar.gz (from octoprint), but installing version 1.12.1
Discarding https://files.pythonhosted.org/packages/c7/b4/3a937c7f8ee4751b38274c8542e02f42ebf3e080f1344c4a2aff6416630e/wrapt-1.14.0.tar.gz (from https://pypi.org/simple/wrapt/) (requires-python:!=3.0.*,!=3.1.*,!=3.2.*,!=3.3.*,!=3.4.*,>=2.7): Requested wrapt<1.15,>=1.14 from https://files.pythonhosted.org/packages/c7/b4/3a937c7f8ee4751b38274c8542e02f42ebf3e080f1344c4a2aff6416630e/wrapt-1.14.0.tar.gz (from octoprint) has inconsistent version: expected '1.14.0', but metadata has '1.12.1'
Collecting octoprint
  Downloading OctoPrint-1.8.6-py2.py3-none-any.whl (3.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╺━━━━ 3.4/3.9 MB 1.7 MB/s eta 0:00:01
ERROR: Operation cancelled by user
```

wrapt module is not upgratable to 1.14 from 1.12.1
it can be installed only 1.15 , but octoprint doesnt accept it.

MasterDimmy, 2023-03-30:
[octoprint-systeminfo-20230330234905.zip](https://github.com/OctoPrint/OctoPrint/files/11116975/octoprint-systeminfo-20230330234905.zip)

jneilliii, 2023-03-30:
looking at the log it seems you are upgrading from 1.7.2.post0.dev0+gd60c5a631 -> 1.8.7. Maybe SSH'ing to the pi and running the following command will get you there.

```
/home/pi/oprint/bin/pip install --force-reinstall --no-cache-dir octoprint==1.8.7
```

jneilliii, 2023-03-31:
looking more at the log, that command probably won't work. sorry, wasn't paying attention as it looks like you have a manual install of OctoPrint. It almost looks like OctoPrint is installed in the system python rather than a venv like what is recommended.

MasterDimmy, 2023-03-31:
i run 2 copies of octoprint simultaneously to serve 2 printers at the same time (tronxy and laser).
This works great (i changed web ports via netns, etc.), but octoprint cant upgrade from 1.7.3

MasterDimmy, 2023-03-31:
> looking at the log it seems you are upgrading from 1.7.2.post0.dev0+gd60c5a631 -> 1.8.7. Maybe SSH'ing to the pi and running the following command will get you there.
> 
> ```
> /home/pi/oprint/bin/pip install --force-reinstall --no-cache-dir octoprint==1.8.7
> ```

```
# pip install --force-reinstall --no-cache-dir octoprint==1.8.7
Collecting octoprint==1.8.7
  Downloading OctoPrint-1.8.7-py2.py3-none-any.whl (3.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.9/3.9 MB 4.8 MB/s eta 0:00:00
Collecting cachelib<0.3,>=0.2
  Downloading cachelib-0.2.0-py3-none-any.whl (12 kB)
Collecting pyserial<4,>=3.4
  Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.6/90.6 kB 9.5 MB/s eta 0:00:00
Collecting Flask-Assets<3,>=2.0
  Downloading Flask_Assets-2.0-py3-none-any.whl (8.5 kB)
Collecting OctoPrint-PiSupport>=2022.6.13
  Downloading OctoPrint_PiSupport-2022.6.13-py2.py3-none-any.whl (33 kB)
Collecting feedparser<7,>=6.0.8
  Downloading feedparser-6.0.10-py3-none-any.whl (81 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 81.1/81.1 kB 9.2 MB/s eta 0:00:00
Collecting netifaces<1,>=0.11
  Downloading netifaces-0.11.0.tar.gz (30 kB)
  Preparing metadata (setup.py) ... done
Collecting semantic-version<3,>=2.8.5
  Downloading semantic_version-2.10.0-py2.py3-none-any.whl (15 kB)
Collecting sentry-sdk<2,>=1.5.7
  Downloading sentry_sdk-1.18.0-py2.py3-none-any.whl (194 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 194.8/194.8 kB 9.6 MB/s eta 0:00:00
Collecting wrapt<1.15,>=1.14
  Downloading wrapt-1.14.1.tar.gz (50 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 50.9/50.9 kB 9.4 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
  WARNING: Requested wrapt<1.15,>=1.14 from https://files.pythonhosted.org/packages/11/eb/e06e77394d6cf09977d92bff310cb0392930c08a338f99af6066a5a98f92/wrapt-1.14.1.tar.gz (from octoprint==1.8.7), but installing version 1.12.1
Discarding https://files.pythonhosted.org/packages/11/eb/e06e77394d6cf09977d92bff310cb0392930c08a338f99af6066a5a98f92/wrapt-1.14.1.tar.gz (from https://pypi.org/simple/wrapt/) (requires-python:!=3.0.*,!=3.1.*,!=3.2.*,!=3.3.*,!=3.4.*,>=2.7): Requested wrapt<1.15,>=1.14 from https://files.pythonhosted.org/packages/11/eb/e06e77394d6cf09977d92bff310cb0392930c08a338f99af6066a5a98f92/wrapt-1.14.1.tar.gz (from octoprint==1.8.7) has inconsistent version: expected '1.14.1', but metadata has '1.12.1'
  Downloading wrapt-1.14.0.tar.gz (50 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 50.8/50.8 kB 9.2 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
  WARNING: Requested wrapt<1.15,>=1.14 from https://files.pythonhosted.org/packages/c7/b4/3a937c7f8ee4751b38274c8542e02f42ebf3e080f1344c4a2aff6416630e/wrapt-1.14.0.tar.gz (from octoprint==1.8.7), but installing version 1.12.1
Discarding https://files.pythonhosted.org/packages/c7/b4/3a937c7f8ee4751b38274c8542e02f42ebf3e080f1344c4a2aff6416630e/wrapt-1.14.0.tar.gz (from https://pypi.org/simple/wrapt/) (requires-python:!=3.0.*,!=3.1.*,!=3.2.*,!=3.3.*,!=3.4.*,>=2.7): Requested wrapt<1.15,>=1.14 from https://files.pythonhosted.org/packages/c7/b4/3a937c7f8ee4751b38274c8542e02f42ebf3e080f1344c4a2aff6416630e/wrapt-1.14.0.tar.gz (from octoprint==1.8.7) has inconsistent version: expected '1.14.0', but metadata has '1.12.1'
ERROR: Ignored the following versions that require a different python version: 1.3.11 Requires-Python >=2.7.9,<3; 1.3.12 Requires-Python >=2.7.9,<3; 1.3.12rc1 Requires-Python >=2.7.9,<3; 1.3.12rc3 Requires-Python >=2.7.9,<3
ERROR: Could not find a version that satisfies the requirement wrapt<1.15,>=1.14 (from octoprint) (from versions: 1.0.0, 1.1.0, 1.1.1, 1.1.2, 1.1.3, 1.1.4, 1.2.0, 1.2.1, 1.3.0, 1.3.1, 1.4.0, 1.4.1, 1.4.2, 1.5.0, 1.5.1, 1.6.0, 1.7.0, 1.8.0, 1.9.0, 1.10.0, 1.10.1, 1.10.2, 1.10.4, 1.10.5, 1.10.6, 1.10.7, 1.10.8, 1.10.9, 1.10.10, 1.10.11, 1.11.0, 1.11.1, 1.11.2, 1.12.0, 1.12.1, 1.13.0rc1, 1.13.0rc2, 1.13.0rc3, 1.13.0, 1.13.1, 1.13.2rc1, 1.13.2, 1.13.3rc1, 1.13.3rc2, 1.13.3, 1.14.0rc1, 1.14.0, 1.14.1, 1.15.0rc1, 1.15.0)
ERROR: No matching distribution found for wrapt<1.15,>=1.14
root@orangepizero:/home#
```

cp2004, 2023-04-06:
It seems that it's had some kind of issue downloading wrapt, and pip got confused. I haven't been able to reproduce this issue, downloading & installing `wrapt-1.14.1.tar.gz` or `wrapt-1.14.0.tar.gz` works just fine for me, which is where it goes wrong on your install. The only thing I can suggest is to try installing the dependency manually first to see if that helps.

This kind of issue is probably best placed on the [community forums](https://community.octoprint.org/) or discord(https://discord.octoprint.org/) server, where there are a lot more people who might be able to help you. There are only limited eyes on the issue tracker & so for help fixing this error and getting OctoPrint updated, please post there.