OctoPrint/OctoPrint #4782: [1.9.0rc3] Typo in classic webcam snapshot hook

cp2004, 2023-04-02:
### The problem

The below setting looks a bit... mixed up

https://github.com/OctoPrint/OctoPrint/blob/ed798e4c6f473ed15bfd8ff583f3913b18ab716c/src/octoprint/plugins/classicwebcam/__init__.py#L137

`rosnapshotSslValidationtate90` ➡️  `snapshotSslValidation`

I'll send in a PR

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.0rc3

### Operating system running OctoPrint

All

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

foosel, 2023-04-11:
Fixed in #4785

foosel, 2023-04-12:
1.9.0rc4 has been released.

closed 2023-04-12 by foosel