OctoPrint/OctoPrint #4783: [1.9.0rc3] Unable to save changes to 'Snapshot Webcam'

cp2004, 2023-04-02:
### The problem

Saving the snapshot webcam does not get written to config.yaml.

I'm still working out why this might be, so I thought I'd report it as a bug in case I don't find it by the end of the evening...

1. Install a WebcamProviderPlugin ([this is my one I've used](https://github.com/cp2004/OctoPrint-CameraStreamer-Control))
2. Try and change the snapshot in the settings
3. It doesn't save.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.0rc3

### Operating system running OctoPrint

Windows 11

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

cp2004, 2023-04-02:
Another 5 mins of looking at this one and I found that issue, but it still doesn't seem to be *actually* used for the timelapse yet

cp2004, 2023-04-02:
OK so changing the timelapse webcam requires a restart of OctoPrint - this should either be fixed/documented if that is a requirement of how it works.

foosel, 2023-04-11:
> OK so changing the timelapse webcam requires a restart of OctoPrint - this should either be fixed/documented if that is a requirement of how it works.

I've extracted that into its own ticket at #4789

This one here is fixed by the merge of #4784 (Thanks!)

foosel, 2023-04-12:
1.9.0rc4 has been released.

PR #4784 (2023-04-11): 🐛  Fix `webcam.snapshotWebcam` not saving to config
It had no default so OctoPrint didn't want to save it

Closes #4783
commit a6de25e8 (2023-04-11): :bug: Fix `webcam.snapshotWebcam` not saving to config (#4784)

It had no default so OctoPrint didn't want to save it

Closes #4783