OctoPrint/OctoPrint #4792: [RC Feedback] Feedback on the 1.9.0rc5 Release Candidate

foosel, 2023-04-13:
# 1.9.0rc5 isn't released yet, this a placeholder

Please provide *general* feedback on your experience with the 1.9.0rc5 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### ❗👇 IMPORTANT 👇❗
###
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports *with system info bundles*, not just a statement that something doesn't work!
###
### ❗👆 IMPORTANT 👆❗

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

Ruhel786, 2023-04-13:
Installed on Rasberry Pi 4 4GB (with HyperPixel 4 rectangle touchscreen)
The update setting is a nice touch straight away, before the restore window.
OctoPrint settings were restored from the backup with no issues. 
Webcam Stream not working, just stuck loading.

HyperPixel 4 screen driver installation was simple via one line legacy installer.
`curl -sSL get.pimoroni.com/hyperpixel4-legacy | bash`
In `/boot/config.txt` add 
`dtoverlay=vc4-kms-dpi-hyperpixel4`
`dtparam=touchscreen-inverted-x`
`display_lcd_rotate=3`
The screen and touchscreen work perfectly. 

Using one line installer for OctoDash would fail and throw this error when it is tried to be run.
`bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)`
[image]

I did a manual install using
`wget -O octodash.deb https://github.com/UnchartedBull/OctoDash/releases/download/v2.3.1/octodash_2.3.1_armv7l.deb`
OctoDash now opens but immediately crashes when it detects any input at all. 
[image]

Other than these issues OctoPrint works as intended (other than webcam stream)
Ruhel™

foosel, 2023-04-17:
@Ruhel786 Please don't confuse OctoPrint (the print server) with OctoPi (the image that comes with the print server, and on which you also install HyperPixel drivers and additional stuff like OctoDash). This RC *only* touches Octo*Print* and anything HyperPixel or OctoDash related is completely out of scope.

> Webcam Stream not working, just stuck loading.

What webcam is that and does it work with OctoPrint 1.8.7? The webcam integration saw some heavy touches in 1.9.0, so if this works on the same image and the same webcam with 1.8.7, but doesn't on 1.9.0rc5, there might still be something up with the webcam integration changes. However, in that case I'll need you to open up a [full bug report](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report) with logs and everything else required, as requested in the first post of this ticket.

foosel, 2023-04-17:
If that is a regular OctoPi image, and not the current `camera-streamer` preview one, then the Cam v3 not working is to be expected and no bug report for that is needed.

LightningShark25, 2023-04-18:
Webcam is now showing in the Command screen (reported broken in rc4)  Thanks
Performance of webcam is very laggy, but haven't looked into it yet - might be some framerate settings I need to adjust

ChrisHeerschap, 2023-04-18:
TL;DR: noticed an issue with gcode viewer in rc3 after I reported everything was good but that is now working perfectly.

I had previously reported on 1.9.0rc3 that everything was working well, but after reporting that noticed that the gcode viewer wasn't updating to follow the job. Was going to report that but was away, and saw rc5 was available before I reported it. Now running on rc5 on the three systems previously reported (two pi4 one pi3, all running various base versions of octopi but all kept up to date on the OS) and the gcode viewer is following the print as one would expect.

JohnOCFII, 2023-04-18:
OctoPrint 1.9.0rc5 on OctoPi 0.18.0 seems to be working well for me:

- Uploaded new GCODE to OctoPrint
- Printed GCODE on Prusa MK3S+ w/MMU2S+ in single extruder mode
- GCODE viewer worked well
- Terminal worked well
- Timelapse worked well
- The Prusa MMU2 Select Filament plugin worked well
- OctoSlack plugin worked well
- Print Job History plugin worked well
- Announcements/Notifications in title bar worked well
- Classic Webcam and Camera settings all worked well

b-morgan, 2023-05-04:
Installed on an RPi 4B with an RPi camera V3 running OctoPi 20230504122418. The camera image on the Control tab is flipped horizontally. I installed the [WebcamExtras Plugin](https://github.com/poulou0/OctoPrint-Webcamextras) before installing rc5 but disabling it didn't fix the flip. I need to start over and carefully document each step before I create a bug report.

b-morgan, 2023-05-04:
With too many browser tabs open, I accidentally installed rc5 on my main system, RPi 3B, RPi camera V2, LulzBot TAZ 6. This system is OctoPi 0.18.0 and was running OctoPrint 1.8.7. While the update claimed success, the OctoPrint server is not running.
The octoprint.log doesn't show anything obvious. Neither a system reboot nor a "sudo service octoprint restart" adds anything to octoprint.log.

b-morgan, 2023-05-05:
The camera image problem on my RPi 4B system has been resolved and may have been operator error.

b-morgan, 2023-05-05:
The upgrade to rc5 on my RPi 3B system completed successfully the second time.

kj5wi, 2023-05-07:
First, a major thank you to you & all the others working on this project.  It is amazing how often a website copies out the same install steps you have documented.
 
My system runs under VERSION_CODENAME=lunar ID=ubuntu. Actually Ubuntu Studio latest on an atopnucma90 using venv and tilix . Tilix has advanced paste turned on to clean up commands before running them.  The venv for rc5 has python 3.12 installed.  All seems fine using virtual serial but cannot find USB0. Dmesg sees it as 

new full-speed USB device number 9 using ehci-pci
New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
New USB device string
[octoprint-systeminfo-20230507090322.zip](https://github.com/OctoPrint/OctoPrint/files/11415366/octoprint-systeminfo-20230507090322.zip)
s: Mfr=0, Product=2, SerialNumber=0
Product: USB Serial
ch341 1-1.1.1.3:1.0: 
ch341-uart converter now
One or both of the next two comm
[octoprint-systeminfo-20230507090322.zip](https://github.com/OctoPrint/OctoPrint/files/11415363/octoprint-systeminfo-20230507090322.zip)
ands made RC5 happy
    sudo chmod a+rw /dev/ttyUSB0
    sudo usermod -a -G dialout kj5wi
the system bundles are before and after these commands.
The plugins install & work as expected. 
I plan to use this as my main system now with continued updates.

foosel, 2023-05-08:
> One or both of the next two comm
> ands made RC5 happy
> sudo chmod a+rw /dev/ttyUSB0
> sudo usermod -a -G dialout kj5wi

That was in all likelihood adding the user to the `dialout` group, as also recommended in the [setup instructions for RPi](https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspberry-pi-os-debian/2337) that are applicable more or less directly to manual installs as well:

[image]

AFAIK the install script by Paul Paukstelis also would have taken care of that.

All in all this initial problem looks less like an RC5 issue and more a case of missing permissions on the user you are running OctoPrint under there. In any case, glad things are now working.

cp2004, 2023-05-19:
➡️ 1.9.0rc6 #4805

closed 2023-05-19 by cp2004