OctoPrint/OctoPrint #4800: The "Sort by date of last print" isn't applied to folders

DominikPalo, 2023-04-23:
### The problem

1. In the files explorer, set the sorting method to "Sort by name (ascending)
2. Observe, that both files and folders are sorted by their name (OK)
3. Change the sorting method to "Sort by date of last print"
4. Observe, that files are sorted based on their date of last print, while folders remain sorted by name (I'd expect to have also folders sorted by their date of last print *)
5. Change the sorting method to "Sort by file size (descending)"
6. Observe, that both files and folders are sorted by their file size (OK)
7. Change the sorting again to "Sort by date of last print"
8. Observe, that files are sorted based on their date of last print, while folders remain sorted by file size (I'd expect to have also folders sorted by their date of last print *)

Not only this seems like a bug (sorting of folders is a bit indeterministic and confusing after switching to "Sort by date of last print" - the order of folders stays set from the previously selected order method), but since I have organized the majority of my gcodes into folders, the "Sort by date of last print" feature is almost useless for me. It would be more practical if the folders could utilize the date of the most recently printed gcode within that specific folder for sorting purposes

I understand that implementing this approach would require a recursive traversal since there could be nested folders, but I don't think this would be an issue, as the "Sort by file size (descending)" method works in a similar way: When using the "Sort by file size (descending)" method, it iterates through all of the contained files, adds up their sizes, and displays the total as the size of the parent folder. Similarly, when implementing the "Sort by date of last print" method, the date of the most recently printed file (or nested folder) should be set as the date for the parent folder.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.7

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

N/A

### Browser and version of browser, operating system running browser

N/A

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230423110522.zip](https://github.com/OctoPrint/OctoPrint/files/11303393/octoprint-systeminfo-20230423110522.zip)

foosel, 2023-05-15:
I agree, that needed some love. Fix is ready for 1.10 (1.9.0 is frozen). I'll have to think about whether a backport of this to any potential 1.9.x bugfix releases can be risked without a RC run, it was a bit more invasive of a change than I normally feel comfortable for that, but straight forward...

foosel, 2023-07-18:
This fix feels too invasive for a bugfix release, so it's not going to see a backport to 1.9.x

commit 0c966efe (2023-05-15): 🐛 Fix folder sorting

It was completely non deterministic due to no info about last print
and upload date having been collected in the backend.

This is fixed now, making storing by last upload date, last print
etc also apply to folders.

The corresponding data has also been made available in the frontend.

Closes #4800
closed 2024-04-24 by foosel