OctoPrint/OctoPrint #4803: "Hotend Idle Timeout" or "cold extrusion prevented" should pause/ stop the print

BjoernSch, 2023-05-03:
### Is your feature request related to a problem? Please describe.

1. Starting a print with "M109 S240" in the gcode
2. Temperature is not stabilizing due to bad PID-Tuning
3. Printer issues "Hotend Idle Timeout" on serial interface after the configured time (HOTEND_IDLE_TIMEOUT_SEC)
4. Hotend set temperature and bed temperature goes to 0 (configurable) by firmware.
5. Octoprint now starts sending the next commands?!
6. Some lines are extruded, then the nozzle clogs due to cold filament and/ or after the temperature falls below 170°C (EXTRUDE_MINTEMP) the firmware issues "cold extrusion prevented" on every command.
8. Print continues to the end without stopping.

### Describe the solution you'd like

Pause or Stop the print at "cold extrusion prevented" or "Hotend Idle Timeout" if the system is currently printing. Something is clearly not going as planned at this point.
"Hotend Idle Timeout" will also cause the bed heater to power off, so bed adhesion will be lost at this point.
Most hard errors during print will fail to "Heating failed", but not unstable temperature issues after M109.

### Describe alternatives you've considered

Issue is caused by bad PID tuning/ lost settings in my case.
But Octoprint starting to extrude after "Hotend Idle Timeout" or continuing to send commands after "cold extrusion prevented" makes no sense to me.
Something has clearly gone wrong with the printer at this point.

cp2004, 2023-05-03:
_If_ the printer returned a line that looked like a normal printer error message, then OctoPrint would stop the print and disconnect. OctoPrint currently triggers error handling if the line starts with `Error:` or `!!`. In my opinion, it would be better for the printer to report this as an error, than add yet another special case to OctoPrint's received line handling. Parsing communication is annoying, and the messages 'cold extrusion prevented' or 'hotend idle timeout' might change at any point - so if the firmware could stick to a standard of prefixing errors with `error:` then it makes life a lot easier.

So my argument would be to make this a request/PR to Marlin, maybe to make it configurable - I can see people complaining they *don't* want OctoPrint to disconnect & shut down their printer, maybe if they press extrude accidentally before preheating for example. An option that makes cold extrusion an actual 'error', which would then trigger OctoPrint's normal error handling.

jneilliii, 2023-05-04:
it would be easy enough to create a plugin that does this as well, which could be configurable as to what things it's looking for. I had created something similar for not doing anything on specific errors coming back from your printer to prevent disconnecting. It would take some minor tweaking to extend it past "error" messages. 

https://github.com/jneilliii/OctoPrint-FirmwareErrorHandler

foosel, 2023-05-04:
Agreed. This is something that should be solved in firmware, which should raise a proper error here instead of a frankly half-hearted informal message. Alternatively someone could write a plugin to turn this into a proper error to then be further consumed by OctoPrint proper. But I really don't want to add more and more and more special handling for firmware quirks to the comm layer. I should have stopped doing this years ago.

closed 2023-05-04 by foosel