OctoPrint/OctoPrint #4804: Unclear application and installation instructions [Request]

schoenid, 2023-05-04:
### Is your feature request related to a problem? Please describe.

Well, I wanted to use OctoPrint, to control my 3D-Printer (Anycubic Kossel).
The descriptions leads to the assumption, that I could install it on the local PC, which I use to control my printer or that I could install it directly on the printer itself. It's not clear, that I hve to use a third system between PC and Printer, which controls the printer and which will be controlled by the PC over network.

### Describe the solution you'd like

In the description (Homepage and manual) should be stated clearly and at the beginning, that the subject of the description and the target of the software is, to configure a third system (1st is the PC, 2nd is the 3D-Printer), to be able to print jobs independently from the state of the first system.

Then as a hint or to clarify, the usable systems should shortly be listed by names, to make sure, a potential new user can understand, what are the requirements in hardware, software and knowledge.

This will prevent from any wrong suggestions right at the beginning and helps to guide new potential users straight to the right path.

### Describe alternatives you've considered

Actually new users like me are frustrated, not to find a possibility, how to install the software on their PC or their 3D-Printer. So I've used Repetier-Host to directly control my 3D-Printer over USB from the PC. It worked.
Now, because I've changed my system from Windows to Linux, this is no more working, because of an incompatibility of Repetier-Host (wine system) with SuperSlicer on Linux, which I used on Windows before.
(Fortunately SuperSlicer is able to communicate with OctoPrint directly ...)
So I have to change.

jneilliii, 2023-05-04:
Technically it is possible to install this software on your PC and control your 3D printer, but only with a USB connection between them.  Directly on the downloads page are links to a windows installer I created.

foosel, 2023-05-04:
This is at the very top of the download page:

[image]

These links link to specific instructions, including the Windows installer that @jneilliii mentioned, and a Linux-specific deploy script maintained by Paul Paukstelis that should have had you covered.

Was this not visible enough? Or did you never end up on that page but somewhere else? If so, where? I'm not trying to alienate you here, I'm trying to understand why you consider the currently available information to be insufficient.

schoenid, 2023-05-04:
@foosel and @jneilliii  Maybe this was not clear enough for me, because english is not my native language and
`... this is for installing the OctoPrint server under Windows, which is probably a rather unusual setup ...`
sounds hear-raising to me.

So basically I found out, that this is to set up a server on windows, which then could be accessed over the localhost network. I agree, this is **rather unusual**, to control any application on the own system. And that is with what I mean "I have to change...". I'm installing it on a Raspberry Pi, located nearly the printer and access it by LAN (cable). This makes the printing process totally independent from my PC, which is great. Also the power for the printer, including the Raspberry Pi, could be provided by an UPS ...

Before I've printed out directly over USB, which in some special cases, has been interrupted, because of some incautious handlings with the system, like connecting a new unknown device to a free USB port ...

What I meant with my post: I was looking for a native desktop application, to directly install on Windows or Linux (now I'm working on Ubuntu Studio 22.04 LTS). I didn't find it. Then I was looking for an installation, directly on the printer ... before I realised, that the capacity on most printers does not fit the requirements. (Sometimes I should turn on my mind in advance ...)

I mean, if there would have been a short introduction or statement on the main page, to state, that the goal of the software is, to have an independent controller between PC and 3D printer, it would have been more straightforward for me. Then, with the information, that it is a network server and that it could be installed on an additional system, like a single board computer (Raspberry Pi, Tinker Board), mini PCs, some NAS, or even locally, I would have known it, right from the start und would have give me more transparency.

Well, the informations on the web page are great and it looks good. Altough I had some hours to find out, how it really works and wether I wanted it or not.

Best Regards

closed 2026-01-20 by foosel