OctoPrint/OctoPrint #4805: [RC Feedback] Feedback on the 1.9.0rc6 Release Candidate

foosel, 2023-05-08:
# 1.9.0rc6 isn't released yet, this a placeholder

Please provide *general* feedback on your experience with the 1.9.0rc6 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### ❗👇 IMPORTANT 👇❗
###
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports *with system info bundles*, not just a statement that something doesn't work!
###
### ❗👆 IMPORTANT 👆❗

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

b-morgan, 2023-05-08:
RPi 3B, RPi camera V2.1, OctoPi 0.18.0 upgraded from 1.9.0rc5 to 1.9.0rc6.
RPi 4B, RPi camera V3, OctoPi 20230504122418 upgraded from 1.9.0rc5 to 1.9.0rc6.

Bradford1040, 2023-05-12:
been using on Rpi 3b for 3 days and I have to say all is running much smoother than 1.9.0rc5! watch now I said this it breaks lol! excellent work Dev's

pwhelan, 2023-05-15:
It works for me on a BTT CB1 connected to a KingRoon KP3S running Marlin 2.1.1.

foosel, 2023-05-23:
Thank you everyone, 1.9.0 has just been released 🥳

closed 2023-05-23 by foosel