OctoPrint/OctoPrint #4806: Software update method single_file_plugin ETag is not correctly verified

Maxinger15, 2023-05-10:
### The problem

I have implemented the software update in my [plugin](https://github.com/Maxinger15/connectandprint/blob/master/connectandprint.py).
``` python
connectandprint=dict(
                displayName=self._plugin_name,
                displayVersion=self._plugin_version,
                type="httpheader",
                header_name="ETag",
                url="https://raw.githubusercontent.com/Maxinger15/connectandprint/master/connectandprint.py",
                method="single_file_plugin"
            )
``` 

The update itself works as expected, but after every restart octoprint notifies me that a new version is available but the
etag strings are identical.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

It needs a plugin feature

### Version of OctoPrint

1.8.7

### Operating system running OctoPrint

OctoPi 0.17.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Notification:
[image]

Systeminfo Bundle:
[octoprint-systeminfo-20230510180611.zip](https://github.com/OctoPrint/OctoPrint/files/11444000/octoprint-systeminfo-20230510180611.zip)

jneilliii, 2023-05-13:
I suspect your issue is the header_name is not case matched to what is returned from GitHub. 

[image]

Try changing it to `Etag`

foosel, 2023-05-15:
Headers are case-insensitve, so no, that doesn't make a difference here. This is actually a bug. At some point I renamed the check type from `etag` to `httpheader` to make it more flexible, but forgot to adjust the part of the code that checks whether the `current` value should be stored. So it currently isn't, and that means the update check will just never have a current value to compare the latest with, and that's causing the loop. 

The fix is easy, but something that'll not make it into 1.9.0, as that's frozen already. Will schedule this for 1.10, however, should there be a bug fix release for 1.9.0 (likely at some point given my experience) this can then be backported.

closed 2023-07-18 by foosel

foosel, 2023-07-18:
The fix for this has been backported to 1.9.x and just been released as part of 1.9.2.

commit 423d3d9a (2023-05-15): 🐛 swu: fix httpheader check type

It was not storing its current value properly, thanks to an incomplete
refactoring ages ago during which etag and lastmodified types were
joined into httpheader.

Closes #4806
commit c4fab1b7 (2023-07-18): 🐛 swu: fix httpheader check type

It was not storing its current value properly, thanks to an incomplete
refactoring ages ago during which etag and lastmodified types were
joined into httpheader.

Closes #4806