OctoPrint/OctoPrint #4808: Printer progress object is initialized with incorrect `printTimeOrigin` key instead of `printTimeLeftOrigin`

ztaylor54, 2023-05-18:
### The problem

I've come across this while developing a plugin. [According to the data model](https://docs.octoprint.org/en/master/api/datamodel.html#progress-information), the print progress object should contain the key `printTimeLeftOrigin`, but it is instead initialized with the key `printTimeOrigin`:

https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/printer/standard.py#L148-L154

And looks like this shortly after OctoPrint starts:

```
{'completion': None, 'filepos': None, 'printTime': None, 'printTimeLeft': None, 'printTimeOrigin': None}
```

After loading a file from the SD card, the [`_updateProgressData()`](https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/printer/standard.py#L1089-L1105) function is called, and the incorrect key is overridden with the correct one:

```
{'completion': None, 'filepos': None, 'printTime': None, 'printTimeLeft': None, 'printTimeLeftOrigin': None}
```

This likely doesn't cause any issues for OctoPrint users, but it caused a little headache as a plugin developer when reading the progress object before it had been updated at least once.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

main branch

### Operating system running OctoPrint

n/a

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

ztaylor54, 2023-05-18:
Do I need to upload a sysinfo bundle for a code bug? It is irrelevant to this bug report -- you can reproduce it by code inspection.

cp2004, 2023-05-19:
> Do I need to upload a sysinfo bundle for a code bug? It is irrelevant to this bug report -- you can reproduce it by code inspection.

We can manually approve as well, this does look like a bug of some kind from a brief look at the links you've sent.

foosel, 2023-11-30:
Fixed by the above commit, will be in 1.10.0

commit 72bd8bf4 (2023-11-30): 🐛 Fix wrong initial field in printer state

Closes #4808
closed 2024-04-24 by foosel