OctoPrint/OctoPrint #4811: Update to 1.9 = Webcam errors and "applyCameraSettings" error logged

beepboop1234, 2023-05-25:
### The problem

Updated to 1.9.0 this morning. Now, attempting to make any changes to my camera in octolapse gives an error "Error Saving Camera Settings Status: error. Error: INTERNAL SERVER ERROR". Looking at the octoprint logs I get:

`2023-05-25 09:54:08,371 - octoprint - ERROR - Exception on /plugin/octolapse/applyCameraSettings [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2529, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 1590, in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 1613, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask_login/utils.py", line 290, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/__init__.py", line 1259, in apply_camera_settings_request
    success, error = self.apply_camera_settings([camera_profile])
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/__init__.py", line 1983, in apply_camera_settings
    camera_profiles, retries=retries, backoff_factor=backoff_factor, no_wait=no_wait
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/camera.py", line 209, in apply_camera_settings
    thread.join(requests.packages.urllib3.util.retry.Retry.BACKOFF_MAX)
AttributeError: type object 'Retry' has no attribute 'BACKOFF_MAX'
2023-05-25 09:54:15,614 - octoprint - ERROR - Exception on /plugin/octolapse/applyCameraSettings [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2529, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 1590, in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 1613, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask_login/utils.py", line 290, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/__init__.py", line 1259, in apply_camera_settings_request
    success, error = self.apply_camera_settings([camera_profile])
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/__init__.py", line 1983, in apply_camera_settings
    camera_profiles, retries=retries, backoff_factor=backoff_factor, no_wait=no_wait
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/camera.py", line 209, in apply_camera_settings
    thread.join(requests.packages.urllib3.util.retry.Retry.BACKOFF_MAX)
AttributeError: type object 'Retry' has no attribute 'BACKOFF_MAX'`

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Octolapse isn't available in safe mode

### Version of OctoPrint

1.9.0

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Prusa Mini

### Browser and version of browser, operating system running browser

Brave on Win 10

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230525124048.zip](https://github.com/OctoPrint/OctoPrint/files/11567289/octoprint-systeminfo-20230525124048.zip)

kantlivelong, 2023-05-25:
You might want to open something on the plugin's tracker.

https://github.com/FormerLurker/Octolapse/issues

foosel, 2023-05-25:
As clearly stated in the contribution guidelines and the issue form, this is the wrong bug tracker for issues with this party plugins, which this clearly is (or it would be available in safe mode.

I strongly suggest you raise this in the octolapse repository, as already suggested.

closed 2023-05-25 by foosel