OctoPrint/OctoPrint #4812: [Request] Add one line output to `octoprint systeminfo .` to report it succeeded and where to find the bundle

cp2004, 2023-05-25:
### Is your feature request related to a problem? Please describe.

This command doesn't work in the field because no one understands that it creates a systeminfo bundle, rather than giving them any output on the command line. 

### Describe the solution you'd like

One (or maybe two) lines, something like this:
```
Creating systeminfo bundle...
Created bundle at <path>
```
Similar to how the backup plugin does it.

### Describe alternatives you've considered

Crying?

jneilliii, 2023-05-25:
The docs on the forum seem to indicate that it does [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887?u=jneilliii).

> $ octoprint systeminfo .
Writing systeminfo bundle to /home/pi/octoprint-systeminfo-20210427141433.zip...
Done!
/home/pi/octoprint-systeminfo-20210427141433.zip

cp2004, 2023-05-25:
I will admit to not looking at it and just writing this based on a feeling from a lot of forum posts where we very rarely get a systeminfo bundle if OctoPrint doesn't start up properly.

foosel, 2023-05-25:
Saw this as well on the forums, might actually be a bug, needs investigation.

jneilliii, 2023-05-25:
I just tried on an instance I had access to from work and it gave me errors.

foosel, 2023-08-30:
```
> octoprint systeminfo .                                                                                                      Writing systeminfo bundle to /home/gina/devel/OctoPrint/OctoPrint/src/octoprint-systeminfo-20230830153305.zip...
Done!
/home/gina/devel/OctoPrint/OctoPrint/src/octoprint-systeminfo-20230830153305.zip
```

Fixed by the above commit.

commit 2b6e93ea (2023-08-30): 🐛 Fix octoprint systeminfo .

Was broken due to a missed call during a refactoring.

Closes #4812
closed 2024-04-24 by foosel