OctoPrint/OctoPrint #4814: SSDP multicast listener is not adding memberships for all interfaces

nastevens, 2023-05-26:
### The problem

For network isolation I have my OctoPrint setup on a RaspberryPi with a "standard" ethernet interface `eth0` and a VLAN interface `eth0.3`. Per the [Discovery Plugin documentation](https://docs.octoprint.org/en/master/bundledplugins/discovery.html) section on `interfaces` I expected SSDP discovery to be available on all interfaces by default if no option is set. To quote:

> interfaces: List of network interfaces for which to enable discovery, if unset all interfaces on the host will be used

However, I was having problems with a different system on the VLAN never receiving any response to its `M-SEARCH` SSDP query. After digging in with `tcpdump` and `netstat` I found that the SSDP multicast group `239.255.255.250` was only being used on the `eth0` interface:
```
pi@octopi:~ $ netstat -g
IPv6/IPv4 Group Memberships
Interface       RefCnt Group
--------------- ------ ---------------------
eth0            1      239.255.255.250
eth0            2      224.0.0.251
eth0            1      all-systems.mcast.net
eth0            1      ff02::fb
eth0            1      ff02::1:ffab:6072
eth0            1      ip6-allnodes
eth0            1      ff01::1
eth0.3          2      224.0.0.251
eth0.3          1      all-systems.mcast.net
eth0.3          1      ff02::fb
eth0.3          1      ff02::1:ff08:e549
eth0.3          1      ip6-allnodes
eth0.3          1      ff01::1
```

I believe the problem is in the `_ssdp_monitor` thread here: https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/plugins/discovery/__init__.py#L710. In the [ip manpage](https://www.man7.org/linux/man-pages/man7/ip.7.html) section for the socket option `IP_ADD_MEMBERSHIP` is the following statement:

> imr_address is the address of the local interface with which the system should join the multicast group; if it is equal to INADDR_ANY, an appropriate interface is chosen by the system.

Elsewhere in that document the description of `INADDR_ANY` says it is `0.0.0.0` which is the `imr_address` being given to the `setsockopt` call linked.

So to bring it all together, I believe what's happened is the code is assuming that `0.0.0.0` will cause a group membership to be added for all interfaces, when really it's only going to be added to _one_ interface, whichever one the system thinks is best. And in my case that is `eth0`, leaving `eth0.3` without an SSDP listener, and hence no replies to `M-SEARCH`. I think the fix is to iterate all interfaces and manually add multicast group memberships for each interface beyond the default one set by the system with `INADDR_ANY`. But I wanted to open the issue first to make sure there's not something I'm missing or that there's not a better way to handle the fix.

[octoprint-systeminfo-20230526140055.zip](https://github.com/OctoPrint/OctoPrint/files/11579124/octoprint-systeminfo-20230526140055.zip)

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.0

### Operating system running OctoPrint

OctoPi 0.18.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

foosel, 2023-11-28:
Thanks for this whole analysis! It took me until now to actually look at this properly, and I can tell you that yes, your diagnosis was spot on and the fix was indeed just iterating over the addresses. That's now done and it looks like it works:

``` 
$ netstat -g
IPv6/IPv4 Group Memberships
Interface       RefCnt Group
--------------- ------ ---------------------
lo              1      mdns.mcast.net
lo              1      all-systems.mcast.net
[...]
eth0            1      239.255.255.250
eth0            1      224.0.0.252
eth0            2      mdns.mcast.net
eth0            1      all-systems.mcast.net
eth0.9          1      239.255.255.250
eth0.9          1      mdns.mcast.net
lo              1      ff02::fb
lo              1      ip6-allnodes
lo              1      ff01::1
[...]
eth0            1      ff02::1:3
eth0            1      ff02::fb
eth0            1      ff02::1:ff1c:be04
eth0            1      ip6-allnodes
eth0            1      ff01::1
eth0.9          1      ip6-allnodes
eth0.9          1      ff01::1
```

Will be part of 1.10.0.

nastevens, 2023-11-28:
Thanks so much for the fix, I'm glad the analysis was helpful! Should I wait to close until the release or can this issue be closed?

foosel, 2023-11-28:
Wait until release, closing these is part of my workflow ^^

Also, it would be great if you could make sure to test this once rc1 is out 👍

nastevens, 2023-11-28:
Sounds good, will do!

nastevens, 2024-02-01:
Just updated to 1.10.0-rc1 and the fix looks good! The multicast address is bound to all interfaces now by default and I was able to discover OctoPrint from another device on the VLAN.

```
pi@octopi:~/.octoprint $ netstat -g
IPv6/IPv4 Group Memberships
Interface       RefCnt Group
--------------- ------ ---------------------
lo              1      all-systems.mcast.net
eth0            1      239.255.255.250
eth0            2      mdns.mcast.net
eth0            1      all-systems.mcast.net
eth0.3          1      239.255.255.250
eth0.3          2      mdns.mcast.net
eth0.3          1      all-systems.mcast.net
lo              1      ip6-allnodes
lo              1      ff01::1
eth0            1      ff02::fb
eth0            1      ff02::1:ffab:6072
eth0            1      ip6-allnodes
eth0            1      ff01::1
eth0.3          1      ff02::fb
eth0.3          1      ff02::1:ff08:e549
eth0.3          1      ip6-allnodes
eth0.3          1      ff01::1
```

commit 734ce079 (2023-11-28): 🐛 discovery: fix ssdp on vlans

Really listen on ALL configured addresses, not just the first one per
device.

Closes #4814

Thanks @nastevens!
closed 2024-04-24 by foosel