OctoPrint/OctoPrint #4816: Uncaught exception after Flask-Limiter upgrade

com6056, 2023-05-28:
### The problem

Getting this error:
```
2023-05-28 19:01:16,193 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/octoprint/plugins/bin/octoprint", line 8, in <module>
    sys.exit(main())
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/__init__.py", line 936, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/cli/server.py", line 317, in serve_command
    run_server(
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/cli/server.py", line 163, in run_server
    octoprint_server.run()
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/__init__.py", line 371, in run
    self._setup_app(app)
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/__init__.py", line 1471, in _setup_app
    limiter = Limiter(
TypeError: __init__() got multiple values for argument 'key_func'
```

Looks like it might be from https://github.com/OctoPrint/OctoPrint/commit/24677639f2f4e262c2ba18e4ca9288dc9f53b91e where we upgraded `Flask-Limiter` from v2 to v3 based on the hack that needed to be added in https://github.com/NixOS/nixpkgs/blob/507f9c4d85ccb3f33bfd4a595462b1bfed5ee602/pkgs/applications/misc/octoprint/default.nix#L21-L32

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.7

### Operating system running OctoPrint

Fedora Linux 36 (Server Edition)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

I can't even load the UI to get the systeminfo bundle unfortunately

com6056, 2023-05-28:
I'm running OctoPrint with `podman` via `docker-compose`:
```
  octoprint:
    image: octoprint/octoprint
    container_name: octoprint
    restart: always
    volumes:
      - /var/lib/octoprint:/octoprint:z
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
```

com6056, 2023-05-28:
> Hi @com6056!
>
> It looks like you didn't upload a [system info bundle](https://community.octoprint.org/t/29887) as requested by the template. A bundle is required to further process your issue. It [...]

Can't load the web UI and tried grabbing it via the CLI and got nothing:
```
root@7c7f54c58695:/octoprint# octoprint systeminfo
root@7c7f54c58695:/octoprint# octoprint systeminfo .
root@7c7f54c58695:/octoprint# ls -la
total 20
drwx------.  4 1014 1014  115 Jul  9  2022 .
dr-xr-xr-x.  1 root root   51 May 28 18:43 ..
-rw-------.  1 1014 1014 1392 Jun 12  2022 .bash_history
-rw-r--r--.  1 1014 1014   18 Jan 26  2021 .bash_logout
-rw-r--r--.  1 1014 1014  141 Jan 26  2021 .bash_profile
-rw-r--r--.  1 1014 1014  492 Jan 26  2021 .bashrc
drwxr-xr-x. 14 root root 4096 May 28 18:25 octoprint
drwxr-xr-x.  4 root root   28 Jul  9  2022 plugins
```

cp2004, 2023-05-28:
Could you confirm exactly what version of Flask-Limiter you have installed? 1.9.0 has a lot of installs already so I don't think this can be a general issue and must be specific to a certain combination. Could you maybe share the output of `pip list`?

com6056, 2023-05-30:
Sure thing! I don't see `Flask-Limiter` in here, so maybe I was on the wrong path or maybe I'm running `pip list` wrong:
```
root@c90ffc8b62d3:/octoprint/plugins/bin# pip list
Package                       Version
----------------------------- -------
awesome-slugify               1.6.5
backoff                       2.1.2
Bed-Visualizer                1.1.1
bson                          0.5.10
DisplayLayerProgress          1.28.0
file-read-backwards           2.0.0
Octolapse                     0.4.1
OctoPrint                     1.8.7
OctoPrint-AutoLoginConfig     0.2.0
OctoPrint-Dashboard           1.19.9
OctoPrint-FirmwareUpdater     1.13.3
OctoPrint-NavbarTemp          0.15
OctoPrint-OctoPod             0.3.14
OctoPrint-PrettyGCode         1.2.4
OctoPrint-PrintTimeGenius     2.3.0
OctoPrint-SimpleEmergencyStop 1.0.5
OpenMiko-Plugin               0.2.1
Pillow                        9.2.0
python-dateutil               2.8.2
sentry-sdk                    1.5.12
six                           1.16.0
UI-Customizer                 0.1.9.2
Unidecode                     0.4.21
```

com6056, 2023-05-30:
Oh looks like I was able to fix things up by just running `pip install octoprint` in the container:
```
root@c90ffc8b62d3:/octoprint# pip install octoprint
Requirement already satisfied: octoprint in ./plugins/lib/python3.8/site-packages (1.8.7)
Requirement already satisfied: markdown<4,>=3.2.2 in /usr/local/lib/python3.8/site-packages (from octoprint) (3.4.3)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /usr/local/lib/python3.8/site-packages (from octoprint) (2021.10.11)
Requirement already satisfied: pkginfo<2,>=1.7.1 in /usr/local/lib/python3.8/site-packages (from octoprint) (1.9.6)
Collecting emoji<2,>=1.4.2
  Downloading emoji-1.7.0.tar.gz (175 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 175.4/175.4 KB 10.2 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Requirement already satisfied: tornado<7,>=6.0.4 in /usr/local/lib/python3.8/site-packages (from octoprint) (6.3.2)
Requirement already satisfied: blinker<2,>=1.4 in /usr/local/lib/python3.8/site-packages (from octoprint) (1.6.2)
Requirement already satisfied: requests<3,>=2.26.0 in /usr/local/lib/python3.8/site-packages (from octoprint) (2.31.0)
Requirement already satisfied: passlib<2,>=1.7.4 in /usr/local/lib/python3.8/site-packages (from octoprint) (1.7.4)
Collecting Flask-Limiter<3,>=2.6
  Downloading Flask_Limiter-2.9.2-py3-none-any.whl (27 kB)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /usr/local/lib/python3.8/site-packages (from octoprint) (2021.2.23)
Requirement already satisfied: colorlog<7,>=6 in /usr/local/lib/python3.8/site-packages (from octoprint) (6.7.0)
Requirement already satisfied: frozendict<3,>=2.0 in /usr/local/lib/python3.8/site-packages (from octoprint) (2.3.8)
Collecting wrapt<1.15,>=1.14
  Downloading wrapt-1.14.1-cp38-cp38-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (81 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 81.0/81.0 KB 6.0 MB/s eta 0:00:00
Requirement already satisfied: PyYAML<6,>=5.4.1 in /usr/local/lib/python3.8/site-packages (from octoprint) (5.4.1)
Requirement already satisfied: sentry-sdk<2,>=1.5.7 in ./plugins/lib/python3.8/site-packages (from octoprint) (1.5.12)
Requirement already satisfied: future<1,>=0.18.2 in /usr/local/lib/python3.8/site-packages (from octoprint) (0.18.3)
Collecting sarge==0.1.6
  Downloading sarge-0.1.6.tar.gz (26 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: netaddr<0.9,>=0.8 in /usr/local/lib/python3.8/site-packages (from octoprint) (0.8.0)
Requirement already satisfied: pyserial<4,>=3.4 in /usr/local/lib/python3.8/site-packages (from octoprint) (3.5)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /usr/local/lib/python3.8/site-packages (from octoprint) (2.0)
Requirement already satisfied: OctoPrint-PiSupport>=2022.6.13 in /usr/local/lib/python3.8/site-packages (from octoprint) (2022.6.13)
Requirement already satisfied: argon2-cffi<22,>=21.3.0 in /usr/local/lib/python3.8/site-packages (from octoprint) (21.3.0)
Collecting Flask-Login<0.6,>=0.5
  Downloading Flask_Login-0.5.0-py2.py3-none-any.whl (16 kB)
Collecting watchdog<2,>=1
  Downloading watchdog-1.0.2-py3-none-manylinux2014_x86_64.whl (72 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 72.6/72.6 KB 6.1 MB/s eta 0:00:00
Requirement already satisfied: psutil<6,>=5.8 in /usr/local/lib/python3.8/site-packages (from octoprint) (5.9.5)
Requirement already satisfied: netifaces<1,>=0.11 in /usr/local/lib/python3.8/site-packages (from octoprint) (0.11.0)
Collecting werkzeug<2.1,>=2.0
  Downloading Werkzeug-2.0.3-py3-none-any.whl (289 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 289.2/289.2 KB 29.1 MB/s eta 0:00:00
Collecting cachelib<0.3,>=0.2
  Downloading cachelib-0.2.0-py3-none-any.whl (12 kB)
Requirement already satisfied: websocket-client<2,>=1.2.1 in /usr/local/lib/python3.8/site-packages (from octoprint) (1.5.2)
Requirement already satisfied: zipstream-ng<2.0.0,>=1.3.4 in /usr/local/lib/python3.8/site-packages (from octoprint) (1.5.0)
Requirement already satisfied: unidecode in ./plugins/lib/python3.8/site-packages (from octoprint) (0.4.21)
Requirement already satisfied: filetype<2,>=1.0.7 in /usr/local/lib/python3.8/site-packages (from octoprint) (1.2.0)
Requirement already satisfied: pylru<2,>=1.2 in /usr/local/lib/python3.8/site-packages (from octoprint) (1.2.1)
Collecting Flask-Babel<3,>=2.0
  Downloading Flask_Babel-2.0.0-py3-none-any.whl (9.3 kB)
Requirement already satisfied: feedparser<7,>=6.0.8 in /usr/local/lib/python3.8/site-packages (from octoprint) (6.0.10)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /usr/local/lib/python3.8/site-packages (from octoprint) (2.10.0)
Collecting zeroconf<0.34,>=0.33
  Downloading zeroconf-0.33.4-py3-none-any.whl (89 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 89.8/89.8 KB 5.2 MB/s eta 0:00:00
Collecting flask<2.2,>=2.1
  Downloading Flask-2.1.3-py3-none-any.whl (95 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 95.6/95.6 KB 11.4 MB/s eta 0:00:00
Requirement already satisfied: pathvalidate<3,>=2.4.1 in /usr/local/lib/python3.8/site-packages (from octoprint) (2.5.2)
Requirement already satisfied: regex in /usr/local/lib/python3.8/site-packages (from octoprint) (2023.5.5)
Requirement already satisfied: Click<9,>=8.0.3 in /usr/local/lib/python3.8/site-packages (from octoprint) (8.1.3)
[... cut ...]
Successfully installed Flask-Babel-2.0.0 Flask-Limiter-2.9.2 Flask-Login-0.5.0 cachelib-0.2.0 commonmark-0.9.1 emoji-1.7.0 flask-2.1.3 rich-12.6.0 sarge-0.1.6 watchdog-1.0.2 werkzeug-2.0.3 wrapt-1.14.1 zeroconf-0.33.4
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: You are using pip version 22.0.4; however, version 23.1.2 is available.
You should consider upgrading via the '/usr/local/bin/python -m pip install --upgrade pip' command.
```

Octoprint came up and I was able to upgrade to 1.9.0 via the web UI 🎉

cp2004, 2023-05-30:
Probably more of a docker issue then - I know it does separate the core install from plugins/updates for packages, to allow updating the container both from within OctoPrint and by just pulling a new container tag. That's the ideal situation I guess but maybe it does not always work.

wjcarpenter, 2023-07-05:
I think it's true that it's an issue with the docker image. Both the 1.9.0 and 1.9.1 octoprint/octoprint images from docker hub have this issue. I was able to upgrade 1.8.7 in place, following @com6056's lead, but it would be nice if someone fixed the docker image. Is that image part of the octoprint project, or is it provided by some 3rd party volunteer?

jneilliii, 2023-07-05:
It has it's own repo and maintainer here on GitHub.

https://github.com/OctoPrint/octoprint-docker