OctoPrint/OctoPrint #4819: Windows installation of octoprint does not update

eclarkmxtx, 2023-06-04:
### The problem

1.-I have tried to install the newest version of Octoprint on windows and it will not work
version 1.8.7 is installed and operational
[octoprint-systeminfo-20230604112739 (1).zip](https://github.com/OctoPrint/OctoPrint/files/11646131/octoprint-systeminfo-20230604112739.1.zip)

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.

### Operating system running OctoPrint

windows 10

### Printer model & used firmware incl. version

Ender 5 plus

### Browser and version of browser, operating system running browser

Chrome Version 113.0.5672.129

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

jneilliii, 2023-06-04:
This is just the nature of windows and file locking. You will more than likely need to stop OctoPrint and perform the upgrade manually via command prompt (run as administrator). These are the commands you would run.

```
C:\Octoprint\venv\Scripts\activate
pip install --upgrade octoprint
deactivate
```

foosel, 2023-06-05:
Apparently that never was implemented, fixing that for 1.10.0

foosel, 2023-06-05:
With the above commit OctoPrint will now mark itself as not updateable through the software update plugin when running under Windows, and show a corresponding message in log and UI:

[image]

The FAQ link points to https://faq.octoprint.org/update-not-possible-windows

commit cb8aa9b0 (2023-06-05): 🐛 swu: refuse OctoPrint update under Windows

Also improve error messaging in that regard (and fix some typo
issues in an existing FAQ URL).

Closes #4819