OctoPrint/OctoPrint #4822: webcam preview and settings causing the page to freeze and crash in Firefox

TheCorruptedEngineer, 2023-06-05:
### The problem

every time the control tab is opened and the webcam attempts to load it causes the site to freeze and crash. This persists across multiple devices running Firefox (even the mobile version) and isn't caused by any installed extensions, the freezing of the website also happens inside of the settings. Other browsers running on Chromium are able to load the webcam preview and run the website fine.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.0

### Operating system running OctoPrint

Android

### Printer model & used firmware incl. version

Ender3 Marlin 1.1.4

### Browser and version of browser, operating system running browser

Firefox 113.0.2 (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[Systeminfo Bundle, Browser console and issue video google drive link](https://drive.google.com/drive/folders/1w4jxoxuA8bFJwOUrGc5-spNQJ8s36m6o?usp=sharing)

cp2004, 2023-06-05:
Firefox has been fine using mjpg streams since OctoPrint existed, but this is the second report (first was a convo on discord) of it choking when loading a stream recently, but other browsers worked fine. Wonder if it's a bug with a specific version perhaps.

TheCorruptedEngineer, 2023-06-06:
> Firefox has been fine using mjpg streams since OctoPrint existed, but this is the second report (first was a convo on discord) of it choking when loading a stream recently, but other browsers worked fine. Wonder if it's a bug with a specific version perhaps.

I am having this issue on several devices running firefox, including the mobile Firefox version on my phone

foosel, 2023-06-06:
Hm... Firefox is my primary browser on both desktop and mobile. I'm currently on 113.0.1 (64-bit) on desktop. Can't say I've noticed anything out of the ordinary.

Judging by your system info bundle, this is a custom install, possibly running under root, so very much far off of anything I generally have access to and see. Could you provide more information on what kind of webcam server you are using here, what webcam, your webcam config and anything else that might be out of the ordinary here?

cp2004, 2023-06-06:
Looks like an Octo4a install to me, but I don't know anything about how webcam streaming is working there.

TheCorruptedEngineer, 2023-06-06:
> Judging by your system info bundle, this is a custom install, possibly running under root, so very much far off of anything I generally have access to and see. Could you provide more information on [...]

It was installed using https://github.com/feelfreelinux/octo4a running on a XIaomi Mi A3 running Android 11

foosel, 2023-06-07:
@TheCorruptedEngineer you wouldn't happen to have the possibility to check whether you face the same issues when you watch an mjpg stream produced by mjpg-streamer? All I do in OctoPrint is embedding the mjpg stream via an `img` tag on the page, but apparently something about the octo4a stream might be causing issues for Firefox. It feels very much like it is out of my hands.

@feelfreelinux can you chime in here?

midikeyboard, 2023-06-07:
having the same experience here in my dockerized octoprint instance.

foosel, 2023-06-07:
Then I need infos on the used webcam server. Once again, OctoPrint is embedding the webcam mjpg stream in an `img` tag. So if things in Firefox explode in some way, that's most likely due to the stream coming from the webcamserver (which is not OctoPrint). I'm happy to investigate whether it might be something in OctoPrint after all (possibly related to the pluginification that happened in 1.9.0), but for that I first need to be able to reproduce this, and currently I can't, everything is working fine here. So - we need more data.

TheCorruptedEngineer, 2023-06-07:
> @TheCorruptedEngineer you wouldn't happen to have the possibility to check whether you face the same issues when you watch an mjpg stream produced by mjpg-streamer? All I do in OctoPrint is [...]
If I open the webcam stream directly by accessing http://192.168.1.19:5001/mjpeg in a separate tab of Firefox the stream works completely fine, it's only when it attempts to load inside of Octoprint that it causes the tab to freeze

TheCorruptedEngineer, 2023-06-07:
> Can you say if this has happened under 1.8.x before and only started with 1.9.0?

I haven't used 1.8.x before

ferllings, 2023-06-26:
I can reproduce as well, since 1.9.0
OctoPrint 1.9.0 - OctoPi 0.18.0
Firefox 114

foosel, 2023-11-28:
This is still unreproduced. Since we rolled out a change that *could* potentially been related with 1.9.1, @TheCorruptedEngineer is this still reproducible for you under 1.9.3?

closed 2024-08-19 by foosel