OctoPrint/OctoPrint #4825: Not being able te resume via OctoPrint after M600

RickB188, 2023-06-06:
### The problem

1. Printer is printing normal
2. Filament runs out
3. I get a notification in OctoPrint (see below)
[image]
4. Head gets parked, filament is retracted by the M600 command
5. OctoPrint shows notification; 'Heater Timeout'; to which I press 'Reheat'
6. Hotend goes to right temperature

To this point everything is working as expected. 

7. Now I am expecting something like 'filament swapped, resume print', but I am only getting the 'Heater Timeout' multiple times one after another. Also I can't press the resume button as it is greyed out
8. Nothing I can do at this point, since I don't have a LCD. 

I am using OctoPrint version 1.9.0
Marlin version 2.1.2

In Marlin configuration_adv.h, i have enabled these lines to enable Host Actions and Host Prompt Support:

[image]

Problem looks a lot like #3699 
https://github.com/OctoPrint/OctoPrint/issues/3699

I am a beginner without a lot of experience in OctoPrint and coding. So far I have only worked with Arduino. 
Maybe it is a simple setting, but I just can't seem to figure it out. 
Also new on github, so if this is the wrong category, please let me know!

Thanks in advance!

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.0

### Operating system running OctoPrint

Windows 11

### Printer model & used firmware incl. version

Bigtreetech SKR3 running on Marlin 2.1.2

### Browser and version of browser, operating system running browser

Google Chrome 114.0.5735.90

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230606154402.zip](https://github.com/OctoPrint/OctoPrint/files/11668130/octoprint-systeminfo-20230606154402.zip)

[log file.txt](https://github.com/OctoPrint/OctoPrint/files/11668203/log.file.txt)

foosel, 2023-06-07:
First of all, please make sure that `serial.log` is (still) enabled, connect to your printer, reproduce the issue, immediately create a new system info bundle and share it here. From the looks of it, the current one didn't include the logs from the reproduction, so it's currently impossible to see what kind of communication actually happened between OctoPrint and the printer.

As far as I can see, `EMERGENCY_PARSER` is enabled, as are `HOST_ACTION_COMMANDS` and `PROMPT_SUPPORT`, so firmware *config* wise, things should be fine as far as my knowledge goes. What I'm completely unclear about yet is whether the `M600` gets triggered by you or the firmware, whether the "Reheat" button press gets through to the firmware, and what else is happening in the communication here, to be able to see if OctoPrint is doing something wrong here or the firmware. And for that, I need a `serial.log` of a reproduction.

RickB188, 2023-06-07:
Thanks for your reply. My new system info bundle of a reproduction I just did:
[octoprint-systeminfo-20230607132746.zip](https://github.com/OctoPrint/OctoPrint/files/11677027/octoprint-systeminfo-20230607132746.zip)

The M600 command gets triggered by the firmware since I am physically removing the filament from the sensor to recreate the real world scenario.

foosel, 2023-06-07:
I fear this is a firmware bug...

```
2023-06-07 13:27:21,961 - Recv: //action:prompt_begin Reheating...
2023-06-07 13:27:21,964 - Recv: //action:prompt_show
2023-06-07 13:27:23,638 - Recv:  T:207.11 /210.00 @:67
2023-06-07 13:27:23,641 - Recv: echo:busy: paused for user
2023-06-07 13:27:25,638 - Recv:  T:205.58 /210.00 @:108
2023-06-07 13:27:25,640 - Recv: echo:busy: paused for user
2023-06-07 13:27:27,638 - Recv:  T:205.05 /210.00 @:116
2023-06-07 13:27:27,640 - Recv: echo:busy: paused for user
2023-06-07 13:27:29,638 - Recv:  T:205.82 /210.00 @:93
2023-06-07 13:27:29,640 - Recv: echo:busy: paused for user
2023-06-07 13:27:31,638 - Recv:  T:207.50 /210.00 @:55
2023-06-07 13:27:31,640 - Recv: echo:busy: paused for user
2023-06-07 13:27:32,479 - Recv: echo:Insert filament and send M108
2023-06-07 13:27:32,480 - Recv: //action:prompt_end
2023-06-07 13:27:32,484 - Recv: //action:prompt_begin Reheat Done
2023-06-07 13:27:32,487 - Recv: //action:prompt_button Continue
2023-06-07 13:27:32,490 - Recv: //action:prompt_show
2023-06-07 13:27:32,639 - Recv: echo:Send M108 to heat nozzle
2023-06-07 13:27:32,640 - Recv: //action:prompt_end
2023-06-07 13:27:32,643 - Recv: //action:prompt_begin Heater Timeout
2023-06-07 13:27:32,645 - Recv: //action:prompt_button Reheat
2023-06-07 13:27:32,647 - Recv: //action:prompt_show
2023-06-07 13:27:33,638 - Recv:  T:209.55 /210.00 @:0
2023-06-07 13:27:33,640 - Recv: echo:busy: paused for user
```

Your firmware just loops through this.

It triggers a prompt "Reheating", heats up to target (almost at least), closes the prompt with a message on serial *only* to insert filament and send M108, then in rapid succession fires off two prompts with another of these serial only messages in between:

- "Reheat Done" with a "Continue" button
- "Heater Timeout" with a "Reheat" button

So something is wonky in the firmware there, I would imagine that the "Reheat Done" prompt should persist and its "Continue" button then make the print resume, but for some reason the firmware aborts it after only 200ms and switches back to the "Heater Timeout" prompt, and its "Reheat" button starts the whole process over.

I suggest you report this to the Marlin bug tracker instead, feel free to reference this ticket.

For now I'm closing this on our end here since all signs point towards it being a firmware issue.

closed 2023-06-07 by foosel

RickB188, 2023-06-08:
Okay, thanks for looking into it. Much appreciated!