OctoPrint/OctoPrint #4826: [Request]

dtibi, 2023-06-07:
### Is your feature request related to a problem? Please describe.

The current OctoPrint web SD file search feature only looks for files in the current directory view and does not search inside folders. This limitation can be frustrating when trying to locate specific files that are organized within folders.

### Describe the solution you'd like

I would like the OctoPrint web SD file search feature to be enhanced so that it includes searching inside folders. This improvement would allow users to easily find files that are stored within nested folders and provide a more comprehensive search experience.

Additionally, I believe this enhancement should be included in OctoPrint's core rather than being implemented as a plugin. Searching inside folders is a common expectation for file management systems, and it would benefit a significant number of OctoPrint users who organize their files in a hierarchical structure.

### Describe alternatives you've considered

Currently, there are no alternatives within OctoPrint's core for searching inside folders. Users may resort to manually navigating through the folder structure to locate specific files, which can be time-consuming and inconvenient.

### Additional context

The ability to search inside folders would improve the overall usability of OctoPrint's file management system. It would streamline the file access process, particularly for users who have a large number of files and utilize folders for better organization. This enhancement would align OctoPrint with common file management practices and enhance the user experience.

[image]

foosel, 2023-06-12:
> The current OctoPrint web SD file search feature only looks for files in the current directory view and does not search inside folders.

That is incorrect.

[Screencast from 2023-06-12 11-28-38.webm](https://github.com/OctoPrint/OctoPrint/assets/83657/1b23fe16-1abf-4cd6-9fa8-e412f5e763e1)

When you search for something, OctoPrint will show you only those folders in the list that contain files that contain the search phrase. So it IS limiting the results, but you still have to click into individual folders. This is done to match the UX when *not* having a search term entered. Consider it a filter, rather than a search like encountered in e.g. the Windows, also indicated through the placement of the search bar itself, which sits on top there to indicate that by entering search terms into it, you limit what you see below, but don't switch to a completely different type of view (which what you are requesting here would require).

dtibi, 2023-06-12:
😵‍💫 Oops! It seems I made a typo in the search, which led me to believe that it only searched in the current directory. In an attempt to fix it, I consulted ChatGPT, but as your comment on the pull request mentioned, it was actually unnecessary. 
My mistake! Thank you for your detailed response.

closed 2023-06-12 by dtibi