OctoPrint/OctoPrint #4831: [Request] Scroll container for settings dialog

csrubin, 2023-06-17:
### Is your feature request related to a problem? Please describe.

It's frustrating to deal with settings, especially when I have a lot of plugins installed. I scroll down (current dialog is one scroll container for sidebar and content), select a plugin that only has a few options, have to scroll back to the top to change the settings, then back down to select another plugin that is out of view.

### Describe the solution you'd like

Add a separate scroll container in the UI, similar to the way the files are shown in the main dashboard screen. This way, the sidebar in the settings dialog and be scrolled through, but content on the right side remains in place. 

### Additional context

[image]
[image]

cp2004, 2023-06-17:
Hmmm

This is how it already is:

[image]

jneilliii, 2023-06-17:
I also get both scroll bars. do you happen to be accessing the site from a mac computer? I vaguely remember something like that mentioned before.

[image]

csrubin, 2023-06-17:
Weird! Yes I’m using a Mac with Safari as the browser (I know I know, everyone hates safari but the RAM and battery usage on Mac is amazing).

jneilliii, 2023-06-17:
From the Apple menu, select System Settings.
Click Appearance. The following screen displays.
[image]
Set the Show Scroll Bars setting to Always.
Close System Settings.

foosel, 2023-06-19:
It boggles my mind that Apple decided "scrollbars are unnecessary clutter, everything has a touch screen anyhow" and broke with the basic UI contract of how scrollable panes work.

In any case, already implemented as requested, it's just Apple being special once more.