OctoPrint/OctoPrint #4832: octoprint travel area exceeds print volume

jethornton, 2023-06-19:
### The problem

Load a g code file with a move to x0.1 and you get an error 'octoprint travel area exceeds print volume' even though the g code does not exceed the travel area.

Settings > Printer XYZ is correct at 220, 220, 250 for my Ender 3.

Check off custom settings and don't change anything and the warning stops.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

New not know how

### Version of OctoPrint

lower left corner of the web interface is blank

### Operating system running OctoPrint

Debian 12

### Printer model & used firmware incl. version

Ender 3

### Browser and version of browser, operating system running browser

Firefox 102.11.0esr (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

unable to upload .gcode file
[octoprint-systeminfo-20230619165613.zip](https://github.com/OctoPrint/OctoPrint/files/11793452/octoprint-systeminfo-20230619165613.zip)

cp2004, 2023-06-19:
> unable to upload .gcode file

That's going to be fairly important for reproducing your issue. Perhaps you could zip it to upload, or rename the extension to .txt.

jethornton, 2023-06-19:
Found the version OctoPrint 1.9.0 Python 3.7.3
[xTool H1.gcode.zip](https://github.com/OctoPrint/OctoPrint/files/11793496/xTool.H1.gcode.zip)

phbasler, 2023-07-27:
I have the same message.

Octopi 0.18.0 on a raspberry pi 3 with octoprint 1.9.2.
Browser is firefox 115.2.1 on android and 115.0 on ubuntu.

The box says 

```
Travel exceeds print volume in depth.
Travel area: (0.00, 0.00, 0.00) × (120.91, 343.10, 91.27)
Object's bounding box: (0.00, 5.00, 0.00) × (120.91, 179.60, 90.77)
Print volume: (0.00, 0.00, 0.00) × (210.00, 210.00, 205.00)
```

I created the gcode file, a litophane, with cura 5.2.1 and set up for an Anycubic I3 Mega S. There were no problems in Cura and the objects bounding box is well inside.

foosel, 2023-07-27:
We cannot look into this without the GCODE file.

phbasler, 2023-07-27:
> We cannot look into this without the GCODE file.

I do not want to share the gcode for this picture as the picture is from a private event. But I could recreate the error with another picture.
The zipped gcode is attached. For this the message popup reads

~~~
Travel area in ErrorMessage.gcode exceeds the print volume of the currently selected printer profile, be careful when printing this.
More
Travel exceeds print volume in depth.
Travel area: (0.00, 0.00, 0.00) × (120.72, 274.18, 91.27)
Object's bounding box: (0.00, 5.00, 0.00) × (120.72, 179.60, 90.77)
Print volume: (0.00, 0.00, 0.00) × (210.00, 210.00, 205.00)

You can disable this check via Settings > Features > "Enable model size detection [...]"
~~~

[ErrorMessage.zip](https://github.com/OctoPrint/OctoPrint/files/12187273/ErrorMessage.zip)

jneilliii, 2023-07-27:
Is your origin configured for lower left in the printer profile?

phbasler, 2023-07-27:
> Is your origin configured for lower left in the printer profile? It seems based on the error that travel moves might be messing with you.

Yes, it is set to lower left

> 274.18 is larger than 210. 

This is correct. But I do not know why it wants a travel area this huge. 
It somehow seems to be connected to the chosen pictures I use to convert to litophanes. 
If I use pictures made by my smartphone camera then this appears, with stock images with the same resolution I could not reproduce the error.
Maybe you can see a difference in the gcode in a relevant section?
[NoErrorMessage.zip](https://github.com/OctoPrint/OctoPrint/files/12188177/NoErrorMessage.zip)

My original image printed without problems even though it had the warning box.

josephnfoley, 2023-08-08:
I'm having the same issue, has anyone found the problem. For example on my latest print the Y travel is shown to extend to 327.62 which is beyond the 300mm limit however the objects bounding box is within the limits and the part prints with no problems. It's just strange and annoying that I get this erro before every print. I have checked the G code and there doesnt appear to be any command that would send the print head to 327.62.

[image]

jneilliii, 2023-08-08:
I've helped someone through a similar issue recently in discord. You need to determine you're actual limits and adjust custom bonding box accordingly.

Start with pressing the home xy button in OctoPrint. Then on terminal tab run `M114` and record where it thinks the head is. Using your printer controls press the buttons to move your axis to their extreme and then press `M114` again. Calculate difference from reported positions and your bed dimensions.

calif94577, 2023-08-14:
So I am having the same issue with my new Octoprint server. I restored from backup so all settings are identical. I even double checked as best I could. Same exact gcode uploaded to both, new one says travel area exceeded, the other does not. Not really sure what is going on. I think I see where in the gcode it's talking about however it does not trigger this on my old server and I have never had issues printing. Thoughts? I can upload the code if needed.

foosel, 2023-08-29:
The more GCODE files (and accompanying printer profile info) we have the better chance this actually gets resolved. I cannot debug stuff that I cannot reproduce, and I don't have a chance to reproduce anything without the offending files.

josephnfoley, 2023-08-29:
Ok, so practically any file that I try causes this error when i use my new laptop with cura 5.4. The configuration should be identical to my last laptop as i copied the config file directly from one to the other and didnt need to change any settings. This is a gcode file for a calibration cube from cura, it should fit well within the machine's area and it prints fine but i still get the error. I have tried to diagnose the source of the problem by copying start and end gcode from old files that didnt have the bug and the problem persisted which leads me to believe that the problem is in the main body of the gcode and not the start or end code. Note that I changed the file name extension from .gcode to .txt just to upload it to github.

[image]

[1 Test Cube for GitHub.txt](https://github.com/OctoPrint/OctoPrint/files/12470359/1.Test.Cube.for.GitHub.txt)

foosel, 2023-08-30:
@josephnfoley OctoPrint is correct here. Your GCODE's final lines are these:

```
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
M84 ; disable motors
M107
G91 ;relative positioning
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 ;X-20 Y-20 F300 ;move Z up a bit and retract filament even more
G28 X0 ;Y0 ;move X/Y to min endstops, so the head is out of the way
G1 Y180 F2000
M84 ;steppers off
G90
M300 P300 S4000
M82 ;absolute extrusion mode
M104 S0
;End of Gcode
```

The offending line here is

    G1 Y180 F2000

At this point, the print head is currently in position (0.0, 158.7, 20.4). Absolute mode has been disabled by the `G91` a few lines prior, so this move command will interpret the `Y` value as *relative* movement of 180mm. Which means an end position of (0.0, 338.7, 20.4). And that exceeds your printer's bounds. The endstops will engage hopefully, the firmware will prevent this move from finishing, but the GCODE says it's supposed to still try it, so the travel dimensions are calculated accordingly.

That move command there should probably rather be an *absolute* move to some extreme Y position to present the printbed (e.g. y_max), but just blindly moving forward by most of the bed length is plain wrong.

@phbasler 

Your "ErrorMessage.gcode" contains the same end code:

```
;TIME_ELAPSED:22203.887838
G1 F1800 E508.29062
M204 S4000
M205 X20 Y20
M107
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
M84 ; disable motors
M107
G91 ;relative positioning
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 ;X-20 Y-20 F100 ;move Z up a bit and retract filament even more
G28 X0 ;Y0 ;move X/Y to min endstops, so the head is out of the way
G1 Y180 F2000
M84 ;steppers off
G90
M300 P300 S4000
M82 ;absolute extrusion mode
M104 S0
;End of Gcode
```

Again, `G1 Y180 F2000` after setting *relative* positioning. That's a recipe for going out of printer bounds.

@jethornton 

And the same in your end code too:

```
G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positioning

G1 X0 Y235 ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z

M82 ;absolute extrusion mode
M104 S0
;End of Gcode
```

This time it's a `G1 X0 Y235`, but still in *relative* mode, so slamming into the end stops.

I strongly suggest y'all fix your slicer end code settings. All these `G1` commands for presenting the print need to be done in *absolute* mode. Otherwise, the Y value you are setting will be added to the current Y position, which depending on the print could already be at the edge of your print bed. 

This end code - sorry to say - is simply wrong the way it is. OctoPrint is interpreting it correctly and telling you that the travel moves you've generated into your GCODE file are exceeding your printer's mechanical bounds. A malfunctioning endstop is all that's needed to turn that from working just fine to "oops, now something broke".

Anyhow, closing this, as OctoPrint did everything correct here.

closed 2023-08-30 by foosel

josephnfoley, 2023-08-30:
Thanks for your help, that seems to have resolved the problem, I will try to find better start and end code later but for now with the Y0 back in the active code the error is gone, I have no idea how that semi colon got in there, must have been a slip of the keyboard at some point. Thanks Again :)

foosel, 2023-08-30:
If you just used Cura's defaults for Anycubic Vyper, Mega or 4max, that wasn't a slip by you, that was already in there for some reason. That's also why I've opened a ticket on Cura's end ^^

calif94577, 2023-08-30:
So when I was digging I found the same thing, the relative code is what caused it, however this never caused an error in previous builds so I assumed the end code was normal and the change in Octoprint was the issue. I too am running default Cura end code. Is this normal across slicers? Would ignoring end code be reasonable here when popping this error? Will look into just changing my end code to fix this as well.

foosel, 2023-08-30:
It is not an error. It is a warning, produced by a sanity check of the print prior to print start to make sure *all* its contained commands will fit the stated printer profile build volume.

So even if it was trivial to identify end code vs actual job parts (it is not, there's no standard way to do that), it would lead this feature ad absurdum. 

Technically, your files sliced like this are wrong, even though your firmware thankfully prevents that from leading to actual damage. OctoPrint correctly identifies that problem. If you don't care about this sanity check you can disable it in the settings, but having it ignore arbitrary parts of the file simply doesn't make sense.

If anything, it not warning you before was probably a bug in that feature that has now been (unknowingly) fixed.

calif94577, 2023-08-30:
That does make sense. I think your approach to changing default cura end code is the correct fix. Hopefully they will. Would being able to provide a visual representation of the travel line with the out of bounds part painted red be possible to quickly identify where it issue is for future people who run into this?

phbasler, 2023-08-31:
@foosel Thanks for finding this! Removing the `;` in front of the `Y0` fixed the message with all files

Flachzange, 2023-09-10:
Hi all, I understood this is not an ocotprint issue. However, as someone who is new to the business I struggle a little what to change in the gcode. It happens to me since I switched to Cura 5.4.0 . 

My end code is
```

G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X0 Y220 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positionning

G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z
```

Appreciate the support. Thanks!

calif94577, 2023-09-10:
> Hi all, I understood this is not an ocotprint issue. However, as someone who is new to the business I struggle a little what to change in the gcode. It happens to me since I switched to Cura 5.4.0 [...]

I hope someone chimes in to confirm cuz I'm not 100% sure however from my understanding your end code seems right cuz it has absolute positioning not relative like ours did. Now I'm curious if my end code on Cura is G91 (relative) or if something is changing it...

Can you post the actual gcode file from a print that is causing the issue?

Flachzange, 2023-09-10:
Please see attached. Printer is obviously  a Sovol SV06 Plus.

[SSP_Pi_Camera_mount_Sovol_SV06_Plus.zip](https://github.com/OctoPrint/OctoPrint/files/12567620/SSP_Pi_Camera_mount_Sovol_SV06_Plus.zip)

calif94577, 2023-09-10:
> Please see attached. Printer is obviously a Sovol SV06 Plus.
> 
> [SSP_Pi_Camera_mount_Sovol_SV06_Plus.zip](https://github.com/OctoPrint/OctoPrint/files/12567620/SSP_Pi_Camera_mount_Sovol_SV06_Plus.zip)

Ok so I found  the issue. You missed the first line of your "End G-Code" (line 343236 of your code attached):

```
G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X0 Y220 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positionning
```

So it puts it at G91 (relative positioning) and then 3 lines down adds 220 to the Y value. This pushes it out of bounds as you only have 300mm of Y available and already at 115.703.

I would recommend (and hopefully this is where someone will chime in and confirm or correct) that you insert the following before the 4th line: 

`G90 ; Absolute positioning`

and then change the next line to:

`G1 Y{machine_depth} F3000 ;Wipe out`

This will make it go straight back. However I feel like this then makes the first line of the next section a bit wasteful:

G1 X0 Y{machine_depth} ;Present print

So I think I would just tell it to go there from the start. In short this is what I would change my end code to:

```
G91 ;Relative positioning
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G90 ; Absolute positioning
G1 X0 Y{machine_depth} F3000 ;Wipe out and Present print
G91 ;Relative positioning
G1 Z10 ;Raise Z more
G90 ; Absolute positioning

M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z
```

Or if you dont feel like you need as much wipe out (I dont see why you would) and simpler code:

```
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X0 Y2 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positioning

G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z
```

This causes just 2mm of wipe out vs the 220mm it first tried. As long as the end of your print is more than 2mm away from the far end of the print bed they you are good to go.

calif94577, 2023-09-10:
I've made a few edits and corrections so please reload the page if you read the previous comment before this one popped up.

Flachzange, 2023-09-10:
> Ok so I found the issue. You missed the first line of your "End G-Code" (line 343236 of your code attached):

Oh...sorry. You are right. Missed that during copy & paste.

Many thanks for your suggestions. For some reason I cannot reproduce it at the moment with the same model although I have not changed the end code in Cura. I will keep an eye on this and try your suggestions once it happens again.

camochu, 2023-09-28:
To everyone with an ender 3 and @foosel FYI. Cura ending gcode includes:

G90 ;Absolute positioning
G1 X0 Y{machine_depth} ;Present print

(machine_depth) becomes 235 and triggers Octoprint warning, but 235 is ok for ender 3 travel limit. It's true that you can't print al 235,235 but the nozzle can move to than limits. The same thing occurs with 0,0 you can move to this position but you can print on this place

I solved changing ender 3 profile in octoprint. I 've keep 220,220,250 as print volume for X,Y,Z and:

""
If your printer's print head may move slightly outside the print volume (e.g. for nozzle cleaning routines) you can define a custom safe bounding box for its movements below.

Custom bounding box (checked)

X Coordinates Min 0 Max 235
Y Coordinates Min 0 Max 235
Z Coordinates Min 0 Max 250
""