OctoPrint/OctoPrint #4836: [BUG?] Hitting RESET on Prusa MK3S makes file being printed vanish from NAS drive

bolsoncerrado, 2023-06-27:
### The problem

I've been running a Prusa MK3* printer farm for YEARSSS using Octoprint in its many versions, feeding the uploads folder from a single NAS drive and having ZERO issues.... until now, with the release of the 1.9.x series.

And the issue is still to be find out on why it's happening. Nothing has changed in the farm except the Octoprint version. Same plugins, same OctoPi release, all raspberry hardware, SD images, network parameters, everything is the SAME as before.

But the thing is, for example, if I have an issue in an ongoing print (nozzle clog, layer shift, whatever) and I hit RESET on the MK3 printer, the file that was being printed vanishes from the NAS.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Haven't tried yet. But I'm not using any other plugins than previous versions, in fact I run very little plugins.

### Version of OctoPrint

1.9.0

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Prusa MK3S

### Browser and version of browser, operating system running browser

Vivaldi on MacOS Ventura

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint (1).log](https://github.com/OctoPrint/OctoPrint/files/11880164/octoprint.1.log)
[octoprint-systeminfo-20230627112432.zip](https://github.com/OctoPrint/OctoPrint/files/11880165/octoprint-systeminfo-20230627112432.zip)

Let me know if you want me to test anything....

...OR if you know for a fact the newest version introduced anything that may be affecting this behaviour or present a clear issue running uploads folder in a NAS. As I said, previous versions have been running FLAWLESS for YEARS, with 0 data corruption or issues of any kind.

foosel, 2023-06-27:
What @jneilliii said. Especially since right in your log, there's what I'd call a smoking gun:

```
2023-06-27 08:15:58,656 - octoprint.plugin - ERROR - Error while calling plugin dashboard
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/shutil.py", line 563, in move
    os.rename(src, real_dst)
PermissionError: [Errno 13] Permission denied: '/home/pi/.octoprint/uploads/tmpp1n_9eb3' -> '/home/pi/.octoprint/uploads/COLT MK IV x 2.gcode'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 275, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_dashboard/__init__.py", line 495, in on_event
    self.load_from_meta(payload)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_dashboard/__init__.py", line 442, in load_from_meta
    stream.save(path)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/util.py", line 102, in save
    shutil.copyfileobj(source, dest)
  File "/usr/lib/python3.7/contextlib.py", line 119, in __exit__
    next(self.gen)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1008, in atomic_write
    shutil.move(fd.name, filename)
  File "/home/pi/oprint/lib/python3.7/shutil.py", line 577, in move
    copy_function(src, real_dst)
  File "/home/pi/oprint/lib/python3.7/shutil.py", line 263, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/home/pi/oprint/lib/python3.7/shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/.octoprint/uploads/COLT MK IV x 2.gcode'
```

In general though, nothing in 1.9.x changed that would make files vanish after cancelled. To me it sounds like it never ever gets written to the NAS, for whatever reason. The above log looks like that might be due to some weird shenanigans going on in the Dashboard plugin (I don't understand why that would even intervene with saving, but apparently it does based on that second stack trace), or it's something somewhere else. In either case: Safe mode. No exceptions. You should know that by now.

bolsoncerrado, 2023-06-27:
Now that you mention it it's true I added Dashboard recently. Let me try disabling it and/or running safe mode as well to rule out stuff.

But I don't agree on the fact it is not saved on the NAS otherwise the thumnail wouldn't show up on the screen nor the print would complete on other printers if that's the case. The "disappearance" of the file started with me as being paranoid and thinking "it can't be true" but it went further and further when I specifically started to note which exact files I uploaded to the NAS (and reportedly completed fine) just to be sure.

Not sure about the smoking gun line you refer to though.

Thanks anyway!

foosel, 2023-07-18:
Any update on this? Was it Dashboard?

bolsoncerrado, 2023-07-18:
I was about to close it yesterday when I noticed a commo printed file missing. Definetly the rate went way lower when un-installed dashboard but it's also true I have had to cancel less jobs too.

Gimme just a few more days for a final update.

Chances are high it was Dashboard.

foosel, 2023-07-18:
> Chances are high it was Dashboard.

Once again proofing why you should always test if an issue persists in safe mode **before** opening a ticket about it. Even if you have just one tiny third party plugin, if that is causing an issue, safe mode will rule it out.

bolsoncerrado, 2023-08-23:
All working as expected after uninstalling the plugin.

closed 2023-08-23 by bolsoncerrado