OctoPrint/OctoPrint #4837: [Request] use the camera's picture's aspect ratio for the <video> tag

gnurbs, 2023-06-27:
### Is your feature request related to a problem? Please describe.

currently the <video> element that displays the webcam in octopi.local/?#control seems to be fixed to 16:9 - it would be great if it would reflect the actual image of the camera, avoiding black bars around the video.

In particular with the raspberry pi camera 2, 1640x1232  is really desirable: it uses the full sensor and in particular the full field of view with 2x2 pixel binning all while being supported by the pi's hardware video encoder. (And in turn the v2 cam is nice for 3D-printing!)

### Describe the solution you'd like

the <video> element should adapt to the video resolution of the camera, e.g. 4:3 with the v2 cam

foosel, 2023-08-10:
This will be part of 1.10.0

foosel, 2024-04-25:
1.10.0 has been released.

closed 2024-04-25 by foosel