OctoPrint/OctoPrint #4838: Poor performance of `HierarchicalChainMap`'s `get_by_path`

mdziekon, 2023-06-29:
### The problem

For a long time now, I've been annoyed with poor performance of the Octoprint's interface on certain actions, eg. when starting up the web interface (as in, reloading it, due to browser being closed or "Reload needed" modal being presented), right after print has finished or after changing settings. Turns out, all of the mentioned above operations require a call to `GET /api/settings` endpoint, which on my board takes ~4300ms. This seems like a huge performance problem in this call, which I believe is quite problematic for people with low-powered devices running Octoprint (which, with still present RPi shortages, might be a big problem; throwing "better hardware" like an x86 thin client could prove viable, but that's not an option for everyone).

### Performance debugging / testing

For context, the environment I'm running my Octoprint instance is:
- OrangePi Zero 2
- Armbian 23.08 on Kernel 6.1 (self-built to include latest CPU scaling patches, which improved the situation a bit)
- ~15 plugins, with heavy ones like SpoolManager or JobHistory

As I already said, the `GET /api/settings` endpoint call takes  ~4300ms on my board (~1700ms when loading from cache). This seems quite a lot for an operation that should at best load a bunch of objects and spew them out to the browser in a JSON format. At first, I thought it was my cheap ARM board's poor IO performance, but once I eliminated that bottleneck with an attached SSD, I investigated further.

With my limited Python knowledge, I dug deep into the Octoprint's code, trying to see why does creating settings read result takes so long:
- First, I've noticed that quite a lot of the time is spent in plugins' settings generation
- Later on, I narrowed it down to `Settings._get_value`, called by parent code
- And finally, I arrived at `HierarchicalChainMap.get_by_path`, which for plugins (but probably for other stuff as well), is called twice in `Settings._get_value`, at least most of the time.

For most of the plugins, this method takes the long path to create an object of all properties belonging to the plugin. That wouldn't be a problem on its own, copying over tens of properties shouldn't be a problem. However, the problematic part lies with how this method determines which properties should be copied over, and which should not, exactly here:
https://github.com/OctoPrint/OctoPrint/blob/1d519a21702ea180c026929da2a6e17079470b9a/src/octoprint/settings/__init__.py#L286

This line iterates over every possible property stored in the settings map* (it's a flattened down map, so it's actually a list, which should already give us a hint), determining whether it belongs to the plugins path, or not - basically, even though we're supposed to be dealing with a map with nestings, it's a **linear scan** on every property. The biggest problem is with how many repeated (and wasteful) scans it performs in the entire process of settings (global) generation - since the structure holds **every** potential setting property (not just one plugin's settings - all of the instance's settings), the structure is scanned in its entirety, for each plugin, twice. This sounds like a death by a thousand cuts kind of situation.

To better visualize it on a smaller scale, and with easier to reproduce dev environment, I created a simple VM with just an octoprint instance, 4 3rd party plugins (remember, Octoprint, by default, has a few built-in plugins, so they count here as well): SpoolManager, JobHistory, Change Filament & Cost Estimator. The VM runs with 4 vcores & 1GB of ram, on my x86 machine (Threadripper 1920x, so not the newest & fastest in terms of single core performance).

With no CPU quotas applied, each call to `HierarchicalChainMap.get_by_path` takes approximately 3 miliseconds, two calls equal 6 miliseconds, which for entire one plugin settings processing gives us 6 miliseconds out of ~10miliseconds of total processing time. On each `HierarchicalChainMap.get_by_path`, it scans ~600 elements (to test whether they belong to the appropriate path). For a total of 21 plugins (internal & 3rd party), this gives us at least 25200 scans with string comparison, and ~210ms of processing time, where ~130ms was spent doing said scans. This was done with repeated calls to the same endpoint, so the caching was heavily limiting the processing time (uncached processing time oscilates between 800ms and 1200ms, but honestly I don't understand why does it differ so much). Just as a reminder - this is on x86, this only gets worse on low powered ARM boards. Please forgive me the lack of timings on the mentioned real ARM board, but running performance tests with such low-powered device is quite tedious, but I believe that even the x86 results should be enough to depict the problem (200ms of time spent to load ~30kB of data seems really bad).

### Other findings

As far as I can tell, there were already some improvements in regards of `HierarchicalChainMap` performance here:
- https://github.com/OctoPrint/OctoPrint/pull/4230
- https://github.com/OctoPrint/OctoPrint/pull/4236

But as far as I can see, none has touched the parts relevant to this problem.

I also tried to tackle this problem myself today, by trying to implement a mirror internal structure, which would potentially limit the problem by scanning only the `path[:-1]` keys, which should significanlty bring down the number of key scans (as in most cases, we wouldn't have to scan all of the map's leafs). However at some point I gave up, since I couldn't recreate exactly the structure of internal `HierarchicalChainMap._chainmap`, so ultimatelly I didn't even get to the point where I would replace the linear scan with "linear scan but for limited number of keys". However, just in case if this is any helpful, I'm leaving a link to my WIP branch:
https://github.com/mdziekon/OctoPrint/pull/1/files

To be fair, I'm not sure if this is the way to go anyway (however, this would probably be the most transparent in terms of breaking changes). Other ideas that came up to my mind was:
- creating dedicated settings objects for each plugin, and storing it separately to the global settings in memory
- improving caching by memoizing calls to plugin settings getter, which could improve performance as long as only certain plugins' settings would change

Each of these however would require significantly more Octoprint's architecture & code knowledge, which I currently don't have, so I didn't investigate further for now.

---

This turned out to be a longer description than I thought, so please forgive me if anything is unclear, and let me know if there's anything missing. As I said, my Python knowledge is quite limited, so I wouldn't be surprised if I did something stupid while testing the performance, but hopefully these informations can still prove useful for further debugging.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.1

### Operating system running OctoPrint

Linux / Armbian 23.08 (current, kernel 6.1.z)

### Browser and version of browser, operating system running browser

Firefox 114 (Nightly), Chrome 114 (Stable)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230629134647.zip](https://github.com/OctoPrint/OctoPrint/files/11909219/octoprint-systeminfo-20230629134647.zip)

foosel, 2023-07-03:
First of all, thank you for this very very detailed and well researched post. I have seen it, I'm adding it to the backlog, but it might be some time until I get around to analysis and responding properly, because I have some other stuff still on the todo list. Just so you know, this HAS been seen, this WILL get looked at properly!

foosel, 2023-09-06:
So I finally found the time to properly look into this, and I think I've found a solution. I now create a cache of prefixes mapping to contained keys, that gets invalidated when the structure changes, and is used to generated the returned maps.

That's something that will require some more field testing for sure, and I'll also look if I can't add some dedicated unit testing around the caching and invalidation as well, but it certainly sped things up measurably even on my development machine (11th Gen i7-1165G7). Tons of `get_by_path` calls that previously needed 2ms and more now take less than 0.1ms, the initial settings fetch right after server start went down from 2s+ to 400-700ms. I also added a timing logger for `get_by_path` to be able to monitor this a bit in the future, should not have impact in prod though.

So thanks again for these nicely done research and heads-up, I think that caused a big jump in performance on the settings 🤝

commit cf131319 (2023-09-06): ⚡️ Improve settings performance

Closes #4838