OctoPrint/OctoPrint #4841: Spelling? mistake.

zaprodk, 2023-07-01:
### The problem

When activating Gcode viewer printing a large gcode, i get this warning:
**"You've selected for printing which has a size of XX.XMB."**

I believe there's missing a word or two in that sentence.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.1

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Original Prusa MK3

### Browser and version of browser, operating system running browser

Chrome Version 114.0.5735.134 (Official Build) (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230701224128.zip](https://github.com/OctoPrint/OctoPrint/files/11927198/octoprint-systeminfo-20230701224128.zip)

jneilliii, 2023-07-01:
Can confirm this is actually a bug I think because he element is data-bound to a knockout observable `selectedFile.name` in a strong HTML tag.

`<strong data-bind="text: selectedFile.name"></strong>`

I suspect something has broke that reference as the file selected name in the state panel is bound to `filedisplay`.  If it's not that then it shouldn't be difficult to find the right reference.

jneilliii, 2023-07-01:
seems name doesn't exist as an attributed to selectFile and never gets assigned. 

https://github.com/OctoPrint/OctoPrint/blob/1d519a21702ea180c026929da2a6e17079470b9a/src/octoprint/plugins/gcodeviewer/static/js/gcodeviewer.js#L50-L54

jneilliii, 2023-07-01:
fixed in above commit/PR and once merged should resolve the issue. switched the binding to the path attribute instead of name.

[image]

foosel, 2023-07-03:
Thanks for the PR, merged, fix will be in 1.10.0!

commit 731fa830 (2023-07-03): 🐛 GCV: Fix filename in rendering warning dialog (#4842)

Closes #4841
closed 2024-04-24 by foosel