OctoPrint/OctoPrint #4843: [Request] Enable/Disable Drag&Drop Upload UI

InoSiX, 2023-07-05:
### Is your feature request related to a problem? Please describe.

Accidental long clicks on Octoprint UI interface while moving mouse presents the user with the Upload dialog (obscuring the window). Most UI elements, if a click is held and the mouse is moved, this issue will occur.

### Describe the solution you'd like

I am calling it an issue, but reality is a lot of this is expected behavior. Over my years of using Octoprint, I have never used this feature. Anything I upload either happens directly from Cura or via the "upload files" feature on the interface.

I have however triggered this upload overlay many, many times. It doesn't cause any workflow issues as letting the click go over either option causes no action, but I still find this annoying.

Consider allowing the user to enable/disable this function.

In the example picture I provided, I selected "ABL Expert Plugin" in the Settings UI. I purposefully dragged it really far, but this behavior happens regularly for me in day to day usuage.

### Describe alternatives you've considered

I honestly expected this to already be a configuration item, so I went through literally several times. Although it is possible I missed the option, it's more likely it doesn't exist.

### Additional context

[image]

foosel, 2023-07-06:
Actually a really good point, thanks for bringing that up!

I created the drag-n-drop functionality at a time when there wasn't yet any slicer integrations, so it featured heavily in my own workflow. These days I can't say when I last even used it myself. So yes, allowing to switch this off to avoid shenanigans caused by it is a no-brainer (although a point could be made that dragging page elements shouldn't even trigger it, but I frankly so far never got around to investigating that)

jneilliii, 2023-07-16:
I took a stab at this one and submitted the above PR.

sdmtt, 2023-07-31:
Just ran into this issue a few moments ago, it felt weird that the overlay showed up when dragging only text. 

> (although a point could be made that dragging page elements shouldn't even trigger it, but I frankly so far never got around to investigating that)

Wouldn't this be a bit cleaner? You can keep the feature enabled even if you never use it, and you don't have to deal with yet another user setting.

Here's some code I tried in the devtools while I waited for a print. It seems to work OK, but I haven't tested it extensively.

[src/octoprint/static/js/app/viewmodels/files.js#L1925](https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/static/js/app/viewmodels/files.js#L1925)
````js
self._handleDragEnter = function (e) {
    let draggedFiles = Array.from(e.originalEvent.dataTransfer.items).filter(item => item.kind === 'file');

    if (! draggedFiles.length) {
        return;
    }

    // .....
}
````

foosel, 2023-08-07:
@sdmtt Thank you for that solution, that looks great and is now pushed!

foosel, 2023-08-07:
Since a case can certainly made for not wanting to have drag and drop enabled at all (beyond the now fixed bug where it triggers on just dragged text), I've also merged the PR by @jneilliii (thanks!)

Both the fix and the feature flag will be part of 1.10.0

fallingcats, 2023-10-25:
I just want to throw in that octoprint in firefox on wayland octoprint seems to trigger the upload overlay quite often when switching from another window/tab and can't easily be dismissed because I'm not actually dragging anything at all.
I hope that only triggering on a file will fix that, otherwise I'll need to file another bug (which is odd because it doesn't happen with any other website.)

manfredg, 2023-11-15:
Just wanted to add that I am encountering the exact same issue with Firefox on Wayland. Just focusing another window and back with Alt-Tab for instance results in an unusable Octoprint Tab (ie the drag and drop overlay being activated). The imperfect fix I found is to drag-and-drop the Firefox address bar, which doesn't make any sense, but it seems to give it what it wants and the overlay goes away.

> I just want to throw in that octoprint in firefox on wayland seems to trigger the upload overlay quite often when switching from another window/tab, and it can't easily be dismissed because I'm not actually dragging anything at all. I hope that only triggering on a file will fix that, otherwise I'll need to file another bug (which is odd because it doesn't happen with any other website.)
> 
> Edit: [image]

commit ac5b3206 (2023-08-07): 🐛 Don't trigger drop zone w/o files

Closes #4843

Thanks to @sdmtt for the suggestion!
closed 2024-04-24 by foosel