OctoPrint/OctoPrint #4844: [Request] HTTPS Certificate Management

kaysond, 2023-07-06:
### Is your feature request related to a problem? Please describe.

I would like to use a certificate signed by my own PKI for accessing octoprint. While I know I can ssh in, generate the CSR, sign it, etc. via command line, it would be so much more convenient to do these operations via the web interface, as is common for many web interfaces.

### Describe the solution you'd like

Implementing this needs the following functions:

- Generate an internal private key and Certificate Signing Request (generally for CN=hostname) and either display PEM-encoded CSR or allow download
- Upload signed public key
- Upload a custom private key + public key combo (instead of using a CSR)

This could be a plugin, but it is also somewhat core to running a web GUI. The big reason I'd want this as a part of the core instead of a plugin is because its security related, and I'd feel comfortable using a core feature in OctoPrint proper. I'd be fairly hesitant to use a plugin that's not in the OctoPrint repo.

### Describe alternatives you've considered

You can already do this via command line, but its very clunky!

cp2004, 2023-07-06:
OctoPrint doesn't handle https at all, this is left to a reverse proxy to sort out that sits in front of it. I think it would be strange for OctoPrint to manage certificates it doesn't use.

foosel, 2023-07-06:
Yes, this is out of scope for OctoPrint indeed, as mentioned.

closed 2023-07-06 by foosel

kaysond, 2023-07-06:
Ah my bad. I used the octopi image and didn't realize it includes haproxy in front of it since https just worked!

kaysond, 2023-07-06:
I guess maybe the feature request should've been for https support in octoprint itself... Which begs the question - why is this not already supported?

I believe flask supports an ssl context, so it would be pretty easy, right?

foosel, 2023-07-06:
Please never just go "it would pretty easy right". Just implementation: yes. But then there's a whole rat's tail of support, configuration, docs etc... When a reverse proxy is already an option (and for webcam stuff needed anyhow), why spend resources on something which can be achieved with existing tools? It simply doesn't make sense with established tools in the place that will do it already, and much better too (think integration with things like LetsEncrypt etc).

kaysond, 2023-07-06:
Because integration can make things much more convenient. I can understand not wanting to add support burden, though. Thanks anyways!