OctoPrint/OctoPrint #4845: Octoprint process using 100-140% CPU for long periods of time

Jinra, 2023-07-06:
### The problem

Since version 1.9.1 Octoprint has been taking up 100%+ cpu usage for long periods of time. CPU temperatures are hitting about 83C as well. This happens in both normal and safe mode, and after restarting my Raspberry pi 3b+. This is causing very slow performance in the web UI, and sometimes it will not connect at all. The problem did not occur on 1.9.0 and prior versions

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.1

### Operating system running OctoPrint

OctoPi 0.18.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230706105030.zip](https://github.com/OctoPrint/OctoPrint/files/11970819/octoprint-systeminfo-20230706105030.zip)

Jinra, 2023-07-19:
thanks @foosel! Since I opened this issue, it's only happened maybe once or twice more, and hasn't happened again. The over temp alerts were only triggering when it had the high CPU usage and not since then.

I think this issue might be resolved for me personally (fingers crossed anyway 🤞), but if you want to investigate further, here is a link to the gcode file. https://www.dropbox.com/scl/fi/ubcwlnm9pezxnxml5mj7a/colored_parts_0.2mm_ASA_22h52m.gcode?rlkey=gnijw6ua4z7ty7mmx1co3ft19&dl=0

foosel, 2023-07-19:
Thanks... The file analysed just fine over here within 80s on my development machine, so it would most likely take longer on the Pi. While analysis was running I saw a CPU load spike, as expected (as I said, analysis is a ton of math).

As you are currently no longer seeing it, I'll close this. If it happens to show up again we can always reopen to analyse further.

closed 2023-07-19 by foosel