OctoPrint/OctoPrint #4846: [Request] "Moveable" PROGRESS BAR position

bolsoncerrado, 2023-07-07:
### Is your feature request related to a problem? Please describe.

I run a farm where I open multiple Octoprint tabs in a single browser.

With the recent addition of slicer thumbnails I have to scrolldown past the thumbnails to find out a "quick way" to see the print progress of each printer.

### Describe the solution you'd like

It would be great a setting to determine the position of the PROGRESS BAR, for example BEFORE the thumbnails would be great!

### Describe alternatives you've considered

Tried to look for plugins but no joy.

jneilliii, 2023-07-07:
I need to see a screenshot of this. Sounds like you've moved your state panel in the sidebar? You should consider UI Customizer that allows you to position things around there UI.

bolsoncerrado, 2023-07-07:
[image]

bolsoncerrado, 2023-07-07:
UI Customizer looks a bit overkill for this "simple" change?

jneilliii, 2023-07-07:
Oh, I see. You haven't moved the state panel, you're just using a super tiny window for each instance. Why don't you scale the state panel image in soccer thumbnails settings?

bolsoncerrado, 2023-07-07:
Tried but it looks "weird"...

[image]

jneilliii, 2023-07-07:
As this is an issue related to a plugin I don't think core OctoPrint is the right place to make this request. Without the customization of my plugin you wouldn't have this issue and wouldn't need to move the progress bar. Would be happy to make it an option in Slicer Thumbnails though if you make the request over there.

closed 2023-07-07 by jneilliii

bolsoncerrado, 2023-07-07:
Got it, makes sense, thanks.