OctoPrint/OctoPrint #4848: 1.9.1 upgrade broke GCode Viewer

davidmcnabnz, 2023-07-10:
### The problem

Up until upgrading OctoPrint to 1.9.1, the inbuilt GCode Viewer was working well.
But following the upgrade, the viewer has stopped working.

When I click to the Viewer tab (and, if the GCode file is large, click the orange confirm button), there seems to be no attempt to load/render the GCode.

I made no other settings change to Octoprint.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Got a 3-day print in progress, so cannot yet try in safe mode

### Version of OctoPrint

1.9.1

### Operating system running OctoPrint

Ubuntu 22.04 LTS

### Printer model & used firmware incl. version

Creality CR-10 Max, with factory firmware

### Browser and version of browser, operating system running browser

Firefox 115.0.1 (issue also present with Chrome)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230711092624.zip](https://github.com/OctoPrint/OctoPrint/files/12008628/octoprint-systeminfo-20230711092624.zip)

foosel, 2023-07-11:
Please update the report once you've had a chance to test in safe mode. A 3-day print in progress or not, we need the info whether this happens in safe mode or not.

davidmcnabnz, 2023-07-12:
ok the big print just completed.

I restarted the octo 1.9.1 daemon in safe mode, but there's still a problem with the gcode viewer tab in both Firefox and Chrome. Please refer attached screenshots of both the browser render, and the JS console.

Nasty JS traceback on the gcode console.

[image]
[image]
[image]
[image]

cp2004, 2023-07-12:
I notice that you host OctoPrint accessible to the internet on your domain. Perhaps you need to refresh some caching?

The mentioned function that is causing an error on your instance does exist in the source, so something about how it's loading in the browser is not right.

Also, get some extra security for your OctoPrint install, since it's directly accessible to the internet for everyone to access.

davidmcnabnz, 2023-07-12:
@cp2004 I think you're chasing after red herrings.

I disabled the internet access, so any further mention of 'security' in this ticket is now off-topic.

Anyway, I since full-refreshed both Chrome and Firefox browsers (Ctrl-F5) on the direct LAN address for the Octoprint daemon, but the problem persists. This clearly nothing to do with 'caching'. 

Up until the 1.9.1 update, the GCode viewer worked perfectly 'as advertised'. Since 1.9.1 update, it's been broken.

(I've often seen issues like this, where migration logic misses an unforeseen edge case. It's especially prevalent in the WordPress plugins ecosystem.)

Maybe I need to hide my config directory, uninstall and clean-install OctoPrint from scratch?

davidmcnabnz, 2023-07-12:
IIRC the GCode viewer was also broken in 1.9.0. Now that you mention it, I think I further upgraded to 1.9.1 in an attempt to resolve it.

davidmcnabnz, 2023-07-12:
Anyway, I'll clean-install and see if that helps.

foosel, 2023-07-12:
Then it sounds like something somewhere in your environment didn't properly update said file or the caches of said file. Which - yes I know - is extremely strange, but that's the only explanation I can offer right now, given that the function in question is defined in the sources of both 1.9.0 and 1.9.1. 

You could check if your current install has it, a simple `grep -r donwloadPath /home/david/.local/lib/python3.10/site-packages/octoprint` should already tell you that.

As a side note, I notice you didn't install OctoPrint into a virtual environment as recommended. It could also possibly be that you have some old version interfering with the fresh install and files getting mixed up, so maybe installing into a clean virtual environment as stated in the installation guides is a good idea to rule this out (and get back to a recommended installation scenario).

davidmcnabnz, 2023-07-12:
```
$ grep -r downloadPath /home/david/.local/lib/python3.10/site-packages/octoprint
/home/david/.local/lib/python3.10/site-packages/octoprint/plugins/gcodeviewer/static/js/gcodeviewer.js:                    url: OctoPrint.files.downloadPath("local", path),
/home/david/.local/lib/python3.10/site-packages/octoprint/static/js/app/client/files.js:    OctoPrintFilesClient.prototype.downloadPath = function (location, path) {
```

foosel, 2023-07-12:
So it even *is* defined in your installation, it's the second hit.

foosel, 2023-07-12:
Out of curiousity, what does

    pip freeze | grep -i octoprint

produce?

foosel, 2023-07-12:
No left over installs then either that could explain that. I'm sorry but now I'm out of ideas for explanations. Based on the above it should all just work and be found by the browser.

cp2004, 2023-07-12:
I'd probably recommend trying to use a virtual environment install rather than system wide. I do have a feeling that we've seen these kinds of issues with system-wide installs rather than virtual environments before, they just don't seem to always upgrade cleanly. You should be able to switch out your install without losing any of the config data, since this is stored separately.

davidmcnabnz, 2023-07-12:
`pip uninstall` followed by `pip install` has resolved the issue. GCode viewer is now again working well.

The issue was that the earlier `pip install --upgrade octoprint` seems to have left an earlier version in place. When I ran pip uninstall octoprint I noticed:
```
Found existing installation: OctoPrint 1.8.6
Uninstalling OctoPrint-1.8.6:
  Would remove:
    /usr/local/bin/octoprint
    /usr/local/lib/python3.10/dist-packages/OctoPrint-1.8.6.dist-info/*
    /usr/local/lib/python3.10/dist-packages/octoprint/*
    /usr/local/lib/python3.10/dist-packages/octoprint_client/*
    /usr/local/lib/python3.10/dist-packages/octoprint_setuptools/*
Proceed (Y/n)? y
  Successfully uninstalled OctoPrint-1.8.6
```

Thanks for your help and feedback. This can close. It may be worth an FAQ and/or knowledge base mention that some problems may be resolved via pip uninstall then install.

davidmcnabnz, 2023-07-12:
Come to think of it, I'm now recalling that I did the 1.9.1 and 1.9.0 upgrades via the web UI, not via command-line pip.

jneilliii, 2023-07-12:
Thanks for the follow-up.

closed 2023-07-12 by jneilliii

foosel, 2023-07-13:
> `pip uninstall` followed by `pip install` has resolved the issue. GCode viewer is now again working well.
>
> The issue was that the earlier `pip install --upgrade octoprint` seems to have left an [...]

And that is *precisely* why the installation guide says to use a virtual env. Your updates where installed into your user directory, but at some point (possibly during the initial install?) you put it globally under the unwritable `/usr/local/bin`. Hilarity ensued. What I don't get is why `pip freeze` didn't make that show up, but possibly even `pip` got confused by this. 

Please, in the future, if an install guide tells you to set something up in a certain way, follow it precisely. There's a reason for every single step and recommendation.