OctoPrint/OctoPrint #4849: Extrude / Retract from UI does not work after edit Printer Profile.

srLinux, 2023-07-10:
### The problem

After I edited the Printer Profile in the UI, i can't use Extrude / Retract from the Controls. The system console output shows me following exception:
```
2023-07-10 21:41:17,508 - octoprint - ERROR - Exception on /api/printer/tool [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 2529, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 1613, in decorated_view
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/octoprint/vendor/flask_principal.py", line 196, in _decorated
    rv = f(*args, **kw)
  File "/usr/local/lib/python3.8/site-packages/octoprint/server/api/printer.py", line 141, in printerToolCommand
    printer.extrude(amount, speed=speed, tags=tags)
  File "/usr/local/lib/python3.8/site-packages/octoprint/printer/standard.py", line 564, in extrude
    ["G91", "M83", "G1 E%s F%d" % (amount, extrusion_speed), "M82", "G90"],
TypeError: %d format: a number is required, not str
```
i checked the profile file ( ../printerProfliles/ ) and it shows:

```
root@2ac12d7a5fd3:/octoprint/octoprint/printerProfiles# cat test.profile 
axes:
  e:
    inverted: false
    speed: '4323'
  x:
    inverted: false
    speed: 2340
  y:
    inverted: false
    speed: 234324
  z:
    inverted: false
    speed: 2343
```
So i think cause this singlequotes, octoprint will interpreted it as string instead of int.
After i removed the singlequotes, it is working as expected.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.1

### Operating system running OctoPrint

docker (octoprint/octoprint)

### Printer model & used firmware incl. version

Creality Ender5 - Marilin 2.1.2.1

### Browser and version of browser, operating system running browser

firefox 102.10.0esr (64-bit), Linux 2ac12d7a5fd3 6.1.0-9-amd64

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230710222929.zip](https://github.com/OctoPrint/OctoPrint/files/12008996/octoprint-systeminfo-20230710222929.zip)

Verustus, 2023-07-22:
You can easily fix that by going to line 564 in /usr/local/lib/python3.8/site-packages/octoprint/printer/standard.py and replacing it with:             ["G91", "M83", "G1 E%s F%d" % (amount, int(extrusion_speed)), "M82", "G90"],

srLinux, 2023-07-22:
@Verustus makes sense, would you please create a PR?

foosel, 2023-07-23:
That would be fixing the symptom, not the bug. A string shouldn't even be written to the profile in the first place, that needs to be fixed. If anyone wants to look into that, that would be lovely and speed things up.

Verustus, 2023-07-23:
@foosel I know but it still works for now. I mean it's just a quick fix. I just needed it to work as fast as possible and because I thougth someone might have the same problem they can fix for now.

zackumar, 2023-07-25:
Hi @foosel! I took a quick look at it, and in octoprint/printer/profile.py, it's not converting the speed for axes e to an int along with the others.

Line 664:

```
# convert ints
for path in (
    ("extruder", "count"),
    ("axes", "x", "speed"),
    ("axes", "y", "speed"),
    ("axes", "z", "speed"),
):
```

Adding `("axes", "e", "speed")` seems to fix the problem. It's also not converting `axes.e.inverted` to a boolean either, but there is no option for that in the UI that I could find but we might run into the same problem if it does get added in.

I don't have time to make a PR today, but if no one else can, I can get to it sometime this week.

srLinux, 2023-07-25:
ok i can't! cause i am not a developer and just knowing git basic commands... PR #4859

foosel, 2023-08-28:
Fix kindly provided in #4860, will be part either of the next 1.9.x bugfix release or of 1.10.

closed 2023-10-09 by foosel

sheminasalam, 2023-10-14:
I am still not able to extrude/retract even after the latest update. I am running octoprint in my orangepi having armbian.

foosel, 2023-10-16:
Try setting the default extrude/retract amount again in your printer profile, save, and try if that solves it.

If it doesn't, please enable [serial.log](https://faq.octoprint.org/serial-log), reproduce the issue, then share a [systeminfo bundle](https://faq.octoprint.org/bundle).

commit 87bea3c7 (2023-08-28): 🐛 Fix extrusion rate not being converted to int (#4860)

Closes  #4849